jsk-ros-pkg/jsk_robot #229: About PR2 tilt self filter behavior

aginika, 2015-01-04:
/tilt_scan_shadow_filter (topic) -> /tilt_laser_self_filter (node) ->/tilt_scan_filtered (topic)

In this flow, I expected that the /tilt_laser_sefl_filter just filter the self filter, it actually works for self filter, but it seems to filter remote or higher point clouds  too. How can I change this behavior? from parmas?

Odded-numbered image is tilt_scan_shadow_filter and 
evened-numbered image is tilt_scan_filtered.

[image]
[image]
[image]
[image]

garaemon, 2015-01-04:
The parameters are written in yaml file

aginika, 2015-01-05:
We couldn't find out the parameter which seems to change the odd output of the self_filter.

garaemon, 2015-01-05:
Please use original parameter with new self filter setting.

✉︎ Ryohei

aginika, 2015-01-05:
Thank you very much!

aginika, 2015-01-05:
Solved with #233

closed 2015-01-05 by aginika