jsk-ros-pkg/jsk_robot #241: Make the tilt_scan processing chain cleaner.

aginika, 2015-01-12:
Now, as far as I searched, I think the tilt_scan processing is below.
[image]

I think we can change below things?

- delete /scan_to_cloud_filter_chain_tilt_laser (What is /tilt_scan_shadow2_filter?)
- delete /tilt_scan_assembler
- remove /tilt_scan from /move_base_node input or remove /ground_objectcloud_mux from /move_base_node input
- remove /sac_ground_removal because move_base_node will automatically remove with min_obstacle_height or max_obstacle_height

I know the best way to decide what to do is to try on real robot, but before that, are there any idea?

garaemon, 2015-01-13:
> delete /scan_to_cloud_filter_chain_tilt_laser (What is /tilt_scan_shadow2_filter?)

+1
> delete /tilt_scan_assembler

-1 It provide service interface to collect scans into pointcloud.
> remove /tilt_scan from /move_base_node input or remove /ground_objectcloud_mux from /move_base_node input

why?

> remove /sac_ground_removal because move_base_node will automatically remove with min_obstacle_height or max_obstacle_height

Did you confirm that robots avoid obstacles as we expected?

aginika, 2015-02-05:
I removed the ground_object from the costmap_common_params.yaml setting fot testing.
Then the local_costmap seems to only use base_scan... I couldn't understand why it doesn't use til_scan as costmap even though move_base_node is subscribing /tilt_scan.
[image]

```
observation_sources: base_scan_marking base_scan tilt_scan #ground_object_cloud

base_scan_marking: {sensor_frame: base_laser_link, topic: /base_scan_marking, data_type: PointCloud2, expected_update_rate: 0.2,
  observation_persistence: 0.0, marking: true, clearing: false, min_obstacle_height: 0.08, max_obstacle_height: 2.0}

base_scan: {sensor_frame: base_laser_link, topic: /base_scan, data_type: LaserScan, expected_update_rate: 0.2,
             observation_persistence: 0.0, marking: false, clearing: true, min_obstacle_height: -0.10, max_obstacle_height: 2.0, raytrace_range: 10.0}
# Modified : extend raytrace area (10.0 m)

tilt_scan: {sensor_frame: laser_tilt_link, topic: /tilt_scan, data_type: LaserScan, expected_update_rate: 0.2,
             observation_persistence: 0.2, marking: false, clearing: true, min_obstacle_height: -20.00, max_obstacle_height: 40.0}

```

aginika, 2015-02-05:
pr2_navigation_config's costmap_common_params,yaml's tilt_scan is subscribing /tilt_scan_interpolated.

/tilt_scan_interpolated is published from laser_and_filters.xml
```
<node pkg="laser_filters" type="scan_to_scan_filter_chain" name="interpolate_missing_tilt_laser_data_filter">
  <rosparam command="load" file="$(find pr2_navigation_perception)/config/laser_interpolation.yaml" />
  <remap from="scan" to="tilt_scan" />
  <remap from="scan_filtered" to="tilt_scan_interpolated" />
</node>
```

But our laser_and_filter.xml doesn't have this node...

garaemon, 2015-02-05:
let's add it

aginika, 2015-02-05:
>> delete /scan_to_cloud_filter_chain_tilt_laser (What is /tilt_scan_shadow2_filter?)
> +1
tilt_scan_shadow2_filter seemed to be used for collider input.
The collider is now set as if="false" in tilt_cloud_scan.launch.
How should I do?

If collider is not needed anymore, I think I can remove tilt_cloud_scan.launch.

aginika, 2015-02-05:
>> remove /tilt_scan from /move_base_node input or remove /ground_objectcloud_mux from /move_base_node input

> why?

>> remove /sac_ground_removal because move_base_node will automatically remove with min_obstacle_height or max_obstacle_height

> Did you confirm that robots avoid obstacles as we expected

I misunderstood around this move_base input.
In default setting, move_base will use ground_object_removal as marker and /tilt_scan as cleaner.
So we need to use both input and we can't clean around move_base inputs.

Only cleaning point is above collider around.

garaemon, 2015-02-05:
Is marker an obstacle?

✉︎ Ryohei Ueda

aginika, 2015-02-05:
yes , I think
```
~<name>/<source_name>/clearing (bool, default: false)
Whether or not this observation should be used to clear out freespace.
~<name>/<source_name>/marking (bool, default: true)
Whether or not this observation should be used to mark obstacles.
```

garaemon, 2015-02-06:
Could you update the picture of move_base pipilene? and add it to README.md

✉︎ Ryohei Ueda

aginika, 2015-02-06:
Sure, I will.
I will remove collider related topics and send Pull Request.
After that I will send PR about updating README

closed 2015-02-06 by aginika