jsk-ros-pkg/jsk_robot #244: jsk_pepper_startup.launch のnao_speech.py が落ちる

kochigami, 2015-01-21:
以前から、roslaunch jsk_pepper_startup jsk_pepper_startup.launch を立ち上げるときに、
一度nao_speech.pyが落ちると、何度このコマンドを打っても、しばらく落ちたままになってしまいます。

パソコンの端末を消してもだいたい駄目でした。
Pepperをリブートしても駄目でした。
自分のパソコンをリブートした後は、jsk_pepper_startup.launchは一度だけ起動したけれど、
一度消してもう一度上げるとnao_speech.pyは落ちてしまいました。

```ps auxwww```の結果を以下に示します。nao_speech.pyは無いように思います。
どうすれば良いか分からないのですが、何か解決方法が分かればここに書かせていただきます。

```
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root         1  0.0  0.0  24564  2520 ?        Ss   22:27   0:00 /sbin/init
root         2  0.0  0.0      0     0 ?        S    22:27   0:00 [kthreadd]
root         3  0.0  0.0      0     0 ?        S    22:27   0:00 [ksoftirqd/0]
root         4  0.0  0.0      0     0 ?        S    22:27   0:00 [kworker/0:0]
root         5  0.0  0.0      0     0 ?        S<   22:27   0:00 [kworker/0:0H]
root         7  0.0  0.0      0     0 ?        S    22:27   0:00 [migration/0]
root         8  0.0  0.0      0     0 ?        S    22:27   0:00 [rcu_bh]
root         9  0.0  0.0      0     0 ?        S    22:27   0:00 [rcuob/0]
root        10  0.0  0.0      0     0 ?        S    22:27   0:00 [rcuob/1]
root        11  0.0  0.0      0     0 ?        S    22:27   0:00 [rcuob/2]
root        12  0.0  0.0      0     0 ?        S    22:27   0:00 [rcuob/3]
root        13  0.0  0.0      0     0 ?        S    22:27   0:00 [rcuob/4]
root        14  0.0  0.0      0     0 ?        S    22:27   0:00 [rcuob/5]
root        15  0.0  0.0      0     0 ?        S    22:27   0:00 [rcuob/6]
root        16  0.0  0.0      0     0 ?        S    22:27   0:00 [rcuob/7]
root        17  0.1  0.0      0     0 ?        S    22:27   0:03 [rcu_sched]
root        18  0.0  0.0      0     0 ?        S    22:27   0:00 [rcuos/0]
root        19  0.0  0.0      0     0 ?        S    22:27   0:00 [rcuos/1]
root        20  0.0  0.0      0     0 ?        S    22:27   0:00 [rcuos/2]
root        21  0.0  0.0      0     0 ?        S    22:27   0:00 [rcuos/3]
root        22  0.0  0.0      0     0 ?        S    22:27   0:00 [rcuos/4]
root        23  0.0  0.0      0     0 ?        S    22:27   0:00 [rcuos/5]
root        24  0.0  0.0      0     0 ?        S    22:27   0:00 [rcuos/6]
root        25  0.0  0.0      0     0 ?        S    22:27   0:00 [rcuos/7]
root        26  0.0  0.0      0     0 ?        S    22:27   0:00 [watchdog/0]
root        27  0.0  0.0      0     0 ?        S    22:27   0:00 [watchdog/1]
root        28  0.0  0.0      0     0 ?        S    22:27   0:00 [migration/1]
root        29  0.0  0.0      0     0 ?        S    22:27   0:00 [ksoftirqd/1]
root        30  0.0  0.0      0     0 ?        S    22:27   0:00 [kworker/1:0]
root        31  0.0  0.0      0     0 ?        S<   22:27   0:00 [kworker/1:0H]
root        32  0.0  0.0      0     0 ?        S    22:27   0:00 [watchdog/2]
root        33  0.0  0.0      0     0 ?        S    22:27   0:00 [migration/2]
root        34  0.0  0.0      0     0 ?        S    22:27   0:00 [ksoftirqd/2]
root        36  0.0  0.0      0     0 ?        S<   22:27   0:00 [kworker/2:0H]
root        37  0.0  0.0      0     0 ?        S    22:27   0:00 [watchdog/3]
root        38  0.0  0.0      0     0 ?        S    22:27   0:00 [migration/3]
root        39  0.0  0.0      0     0 ?        S    22:27   0:00 [ksoftirqd/3]
root        40  0.0  0.0      0     0 ?        S    22:27   0:00 [kworker/3:0]
root        41  0.0  0.0      0     0 ?        S<   22:27   0:00 [kworker/3:0H]
root        42  0.0  0.0      0     0 ?        S    22:27   0:00 [watchdog/4]
root        43  0.0  0.0      0     0 ?        S    22:27   0:00 [migration/4]
root        44  0.0  0.0      0     0 ?        S    22:27   0:00 [ksoftirqd/4]
root        46  0.0  0.0      0     0 ?        S<   22:27   0:00 [kworker/4:0H]
root        47  0.0  0.0      0     0 ?        S    22:27   0:00 [watchdog/5]
root        48  0.0  0.0      0     0 ?        S    22:27   0:00 [migration/5]
root        49  0.0  0.0      0     0 ?        S    22:27   0:00 [ksoftirqd/5]
root        50  0.0  0.0      0     0 ?        S    22:27   0:00 [kworker/5:0]
root        51  0.0  0.0      0     0 ?        S<   22:27   0:00 [kworker/5:0H]
root        52  0.0  0.0      0     0 ?        S    22:27   0:00 [watchdog/6]
root        53  0.0  0.0      0     0 ?        S    22:27   0:00 [migration/6]
root        54  0.0  0.0      0     0 ?        S    22:27   0:00 [ksoftirqd/6]
root        56  0.0  0.0      0     0 ?        S<   22:27   0:00 [kworker/6:0H]
root        57  0.0  0.0      0     0 ?        S    22:27   0:00 [watchdog/7]
root        58  0.0  0.0      0     0 ?        S    22:27   0:00 [migration/7]
root        59  0.0  0.0      0     0 ?        S    22:27   0:00 [ksoftirqd/7]
root        60  0.0  0.0      0     0 ?        S    22:27   0:00 [kworker/7:0]
root        61  0.0  0.0      0     0 ?        S<   22:27   0:00 [kworker/7:0H]
[... 164 lines cut ...]
1000      4594  0.0  0.0  30248  6088 pts/3    Ss+  22:35   0:00 /bin/bash --noediting -i
1000      5345  4.6  1.8 1281856 302232 ?      Sl   22:38   1:05 /usr/lib/firefox/firefox
1000      5375  0.0  0.0 293104  4132 ?        Sl   22:38   0:00 /usr/lib/at-spi2-core/at-spi-bus-launcher
1000      5421  1.1  0.2 524156 45868 ?        Sl   22:38   0:16 /usr/lib/firefox/plugin-container /usr/lib/flashplugin-installer/libflashplayer.so -greomni /usr/lib/firefox/omni.ja -appomni /usr/lib/firefox/browser/omni.ja -appdir /usr/lib/firefox/browser 5345 true plugin
root      5465  0.0  0.0      0     0 ?        S    22:38   0:00 [kworker/u16:0]
1000      5466  0.0  0.1 177144 21168 ?        SLl  22:38   0:00 /usr/lib/mozc/mozc_server
1000      5550  0.0  0.0   4404   612 ?        Ss   22:41   0:00 /bin/sh -c gnome-terminal
1000      5551  0.9  0.1 619004 26452 ?        Sl   22:41   0:10 gnome-terminal
1000      5556  0.0  0.0  14792   848 ?        S    22:41   0:00 gnome-pty-helper
1000      5558  0.0  0.0  30424  6480 pts/2    Ss+  22:41   0:00 bash
1000      6153  0.0  0.0  30556  6508 pts/4    Ss+  22:49   0:00 bash
root      6404  0.0  0.0      0     0 ?        S    22:50   0:00 [kworker/u16:1]
1000      6921  0.0  0.0  30792  6752 pts/5    Ss+  22:53   0:00 bash
root      7539  0.0  0.0      0     0 ?        S    22:56   0:00 [kworker/u16:2]
1000      7756  0.2  0.1 455284 18848 ?        Ssl  22:57   0:00 /opt/ros/hydro/lib/diagnostic_aggregator/aggregator_node __name:=diag_agg __log:=/home/kochigami/.ros/log/6b2c1b9a-a135-11e4-aed6-c03fd5ac4304/diag_agg-1.log
1000      7761  2.2  0.0 540736 10860 ?        Ssl  22:57   0:05 /opt/ros/hydro/lib/robot_state_publisher/state_publisher __name:=robot_state_publisher __log:=/home/kochigami/.ros/log/6b2c1b9a-a135-11e4-aed6-c03fd5ac4304/robot_state_publisher-2.log
1000      7767 11.1  0.3 1404212 54368 ?       Ssl  22:57   0:26 python /home/kochigami/ros/hydro/src/ros-naoqi/naoqi_bridge/naoqi_driver/nodes/naoqi_joint_states.py --pip=133.11.216.224 --pport=9559 __name:=naoqi_joint_states __log:=/home/kochigami/.ros/log/6b2c1b9a-a135-11e4-aed6-c03fd5ac4304/naoqi_joint_states-3.log
root      8192  0.0  0.0      0     0 ?        S    22:58   0:00 [kworker/4:0]
1000      8382  0.0  0.0  30340  6284 pts/6    Ss   22:59   0:00 bash
root      8555  0.0  0.0      0     0 ?        S    23:00   0:00 [kworker/4:3]
root      8576  0.0  0.0      0     0 ?        S    23:01   0:00 [kworker/0:2]
root      8577  1.0  0.0  57588  3184 ?        Ss   23:01   0:00 sshd: [accepted]    
sshd      8578  0.0  0.0  51384  1292 ?        S    23:01   0:00 sshd: [net]         
1000      8579  0.0  0.0  23320  1276 pts/6    R+   23:01   0:00 ps auxwww
```

k-okada, 2015-01-22:
これってnaoでも起きるかな？

kochigami, 2015-01-22:
naoでは起きたことがありません。

naoとpepperで、カメラの所以外は同じプログラムを上げることもあるのですが、pepperの方だけ落ちます。
以前おっしゃっていたように、一度rosのノードが落ちても、pepperの音声認識に関する機能は落ちていないため起きているのだと思います。しかし、今回は生きているプログラムを見つけられませんでした。

kochigami, 2015-01-27:
naoでも同じことが起きるのを確認しました。
```ps auxwww```の結果を添付致します。
```
[http://localhost:11311] kochigami@kochigami-ThinkPad-T440p:~ $ ps auxwww
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root         1  0.0  0.0  24572  2524 ?        Ss   11:13   0:00 /sbin/init
root         2  0.0  0.0      0     0 ?        S    11:13   0:00 [kthreadd]
root         3  0.0  0.0      0     0 ?        S    11:13   0:00 [ksoftirqd/0]
root         5  0.0  0.0      0     0 ?        S<   11:13   0:00 [kworker/0:0H]
root         7  0.0  0.0      0     0 ?        S    11:13   0:00 [migration/0]
root         8  0.0  0.0      0     0 ?        S    11:13   0:00 [rcu_bh]
root         9  0.0  0.0      0     0 ?        S    11:13   0:00 [rcuob/0]
root        10  0.0  0.0      0     0 ?        S    11:13   0:00 [rcuob/1]
root        11  0.0  0.0      0     0 ?        S    11:13   0:00 [rcuob/2]
root        12  0.0  0.0      0     0 ?        S    11:13   0:00 [rcuob/3]
root        13  0.0  0.0      0     0 ?        S    11:13   0:00 [rcuob/4]
root        14  0.0  0.0      0     0 ?        S    11:13   0:00 [rcuob/5]
root        15  0.0  0.0      0     0 ?        S    11:13   0:00 [rcuob/6]
root        16  0.0  0.0      0     0 ?        S    11:13   0:00 [rcuob/7]
root        17  0.1  0.0      0     0 ?        S    11:13   0:06 [rcu_sched]
root        18  0.0  0.0      0     0 ?        S    11:13   0:01 [rcuos/0]
root        19  0.0  0.0      0     0 ?        S    11:13   0:00 [rcuos/1]
root        20  0.0  0.0      0     0 ?        S    11:13   0:00 [rcuos/2]
root        21  0.0  0.0      0     0 ?        S    11:13   0:00 [rcuos/3]
root        22  0.0  0.0      0     0 ?        S    11:13   0:00 [rcuos/4]
root        23  0.0  0.0      0     0 ?        S    11:13   0:00 [rcuos/5]
root        24  0.0  0.0      0     0 ?        S    11:13   0:00 [rcuos/6]
root        25  0.0  0.0      0     0 ?        S    11:13   0:00 [rcuos/7]
root        26  0.0  0.0      0     0 ?        S    11:13   0:00 [watchdog/0]
root        27  0.0  0.0      0     0 ?        S    11:13   0:00 [watchdog/1]
root        28  0.0  0.0      0     0 ?        S    11:13   0:00 [migration/1]
root        29  0.0  0.0      0     0 ?        S    11:13   0:00 [ksoftirqd/1]
root        30  0.0  0.0      0     0 ?        S    11:13   0:00 [kworker/1:0]
root        31  0.0  0.0      0     0 ?        S<   11:13   0:00 [kworker/1:0H]
root        32  0.0  0.0      0     0 ?        S    11:13   0:00 [watchdog/2]
root        33  0.0  0.0      0     0 ?        S    11:13   0:00 [migration/2]
root        34  0.0  0.0      0     0 ?        S    11:13   0:00 [ksoftirqd/2]
root        36  0.0  0.0      0     0 ?        S<   11:13   0:00 [kworker/2:0H]
root        37  0.0  0.0      0     0 ?        S    11:13   0:00 [watchdog/3]
root        38  0.0  0.0      0     0 ?        S    11:13   0:00 [migration/3]
root        39  0.0  0.0      0     0 ?        S    11:13   0:00 [ksoftirqd/3]
root        41  0.0  0.0      0     0 ?        S<   11:13   0:00 [kworker/3:0H]
root        42  0.0  0.0      0     0 ?        S    11:13   0:00 [watchdog/4]
root        43  0.0  0.0      0     0 ?        S    11:13   0:00 [migration/4]
root        44  0.0  0.0      0     0 ?        S    11:13   0:00 [ksoftirqd/4]
root        45  0.0  0.0      0     0 ?        S    11:13   0:00 [kworker/4:0]
root        46  0.0  0.0      0     0 ?        S<   11:13   0:00 [kworker/4:0H]
root        47  0.0  0.0      0     0 ?        S    11:13   0:00 [watchdog/5]
root        48  0.0  0.0      0     0 ?        S    11:13   0:00 [migration/5]
root        49  0.0  0.0      0     0 ?        S    11:13   0:00 [ksoftirqd/5]
root        51  0.0  0.0      0     0 ?        S<   11:13   0:00 [kworker/5:0H]
root        52  0.0  0.0      0     0 ?        S    11:13   0:00 [watchdog/6]
root        53  0.0  0.0      0     0 ?        S    11:13   0:00 [migration/6]
root        54  0.0  0.0      0     0 ?        S    11:13   0:00 [ksoftirqd/6]
root        56  0.0  0.0      0     0 ?        S<   11:13   0:00 [kworker/6:0H]
root        57  0.0  0.0      0     0 ?        S    11:13   0:00 [watchdog/7]
root        58  0.0  0.0      0     0 ?        S    11:13   0:00 [migration/7]
root        59  0.0  0.0      0     0 ?        S    11:13   0:00 [ksoftirqd/7]
root        60  0.0  0.0      0     0 ?        S    11:13   0:00 [kworker/7:0]
root        61  0.0  0.0      0     0 ?        S<   11:13   0:00 [kworker/7:0H]
root        62  0.0  0.0      0     0 ?        S<   11:13   0:00 [khelper]
root        63  0.0  0.0      0     0 ?        S    11:13   0:00 [kdevtmpfs]
root        64  0.0  0.0      0     0 ?        S<   11:13   0:00 [netns]
root        65  0.0  0.0      0     0 ?        S<   11:13   0:00 [writeback]
root        66  0.0  0.0      0     0 ?        S<   11:13   0:00 [kintegrityd]
root        67  0.0  0.0      0     0 ?        S<   11:13   0:00 [bioset]
root        68  0.0  0.0      0     0 ?        S<   11:13   0:00 [crypto]
root        69  0.0  0.0      0     0 ?        S<   11:13   0:00 [kworker/u17:0]
root        70  0.0  0.0      0     0 ?        S<   11:13   0:00 [kblockd]
root        71  0.0  0.0      0     0 ?        S<   11:13   0:00 [ata_sff]
[... 184 lines cut ...]
1000      6797  0.6  0.1 456264 21252 ?        Ssl  12:30   0:04 python /opt/ros/hydro/lib/rocon_gateway/gateway.py __name:=gateway __log:=/home/kochigami/.ros/log/c6db9d8c-a5d4-11e4-81f7-28d2444a1522/gateway-3.log
[... cut ...]
1000     12082  0.1  0.0 433532 14304 pts/8    Sl+  12:37   0:00 /home/kochigami/ros/hydro/devel/lib/aldebaran_robot_ws/pepper_test2
1000     15816  0.1  0.0  30408  6452 pts/3    Ss   12:39   0:00 bash
1000     16554  1.1  0.2 515956 40508 ?        Sl   12:40   0:00 /usr/lib/firefox/plugin-container /usr/lib/flashplugin-installer/libflashplayer.so -greomni /usr/lib/firefox/omni.ja -appomni /usr/lib/firefox/browser/omni.ja -appdir /usr/lib/firefox/browser 4479 true plugin
1000     16857  0.0  0.0  23320  1284 pts/3    R+   12:41   0:00 ps auxwww
```

k-okada, 2015-01-27:
このとき，
１）すべてのROSのプログラムを落として（roscoreとしてエラーが出なくてたちあがればいい）nao_speech.pyを立ち上げた結果（エラーで落ちるんだよね）を教えてください．
２）一回PCをリブートしてnao_speech.pyを立ち上げた時の結果（エラーで落ちる気がする）を教えてください．
３）NAOをリブートしたら治るよね．

kochigami, 2015-01-27:
もう一度試してみたら、自分のプログラムのせいなのかな、と思いました。お騒がせし、申し訳ございません。
自分のパソコンでは、今までは
https://github.com/ros-naoqi/nao_robot/pull/2
のエンコーディングの部分でエラーが出ていたので、この部分を外して使っていました。
自分のプログラム（エンコーディングなし）・nao_appsのプログラム（エンコーディングあり）の２つで試しました。

3)NAOをリブートして、自分のプログラム（エンコーディングなし）を使うと、
```
Traceback (most recent call last):
  File "/home/kochigami/ros/hydro/src/aldebaran_robot_ws/scripts/nao_speech.py", line 489, in <module>
    ROSNaoSpeechModule = NaoSpeech("ROSNaoSpeechModule")
  File "/home/kochigami/ros/hydro/src/aldebaran_robot_ws/scripts/nao_speech.py", line 115, in __init__
    self.reconf_server = ReConfServer(NodeConfig, self.reconfigure)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/dynamic_reconfigure/server.py", line 69, in __init__
    self.config = initial_config(encode_config(self.config), type.config_description)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/dynamic_reconfigure/encoding.py", line 272, in initial_config
    add_params(d['groups'], description)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/dynamic_reconfigure/encoding.py", line 266, in add_params
    group['parameters'][param['name']] = d[param['name']]
  File "/opt/ros/hydro/lib/python2.7/dist-packages/dynamic_reconfigure/encoding.py", line 65, in __getitem__
    return super(Config, self).__getitem__(name)
KeyError: 'vocabulary'
```
というエラーが出ました。
1)のもとで、nao_appsのプログラム（エンコーディングあり）を立ち上げると動きました。

kochigami, 2015-01-27:
度々すみません。立ち上がったもとで、音声認識を行うとこのようなエラーが出てしまいます。
pythonのデフォルトのエンコーディングを変えようとしたのですが、なぜか変わらなくて、
エンコーディングを外した自分のプログラムを使っていた、という経緯でした。
自分の環境の問題だと思います。
```
[ERROR] [WallTime: 1422332670.194427] Error processing request: 'ascii' codec can't decode byte 0xe3 in position 0: ordinal not in range(128)
['Traceback (most recent call last):\n', '  File "/opt/ros/hydro/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 623, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/kochigami/ros/hydro/src/ros-naoqi/nao_robot/nao_apps/nodes/nao_speech.py", line 370, in start\n    self.conf )\n', '  File "/home/kochigami/ros/hydro/src/ros-naoqi/nao_robot/nao_apps/nodes/nao_speech.py", line 434, in __init__\n    self.reconfigure(config)\n', '  File "/home/kochigami/ros/hydro/src/ros-naoqi/nao_robot/nao_apps/nodes/nao_speech.py", line 456, in reconfigure\n    Util.parse_vocabulary( config["vocabulary"].encode(\'utf-8\') ),\n', "UnicodeDecodeError: 'ascii' codec can't decode byte 0xe3 in position 0: ordinal not in range(128)\n"]
[ERROR] [WallTime: 1422332685.198621] Speech recognition was not started
[ERROR] [WallTime: 1422332687.274093] Error processing request: 'ascii' codec can't decode byte 0xe3 in position 0: ordinal not in range(128)
['Traceback (most recent call last):\n', '  File "/opt/ros/hydro/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 623, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/kochigami/ros/hydro/src/ros-naoqi/nao_robot/nao_apps/nodes/nao_speech.py", line 370, in start\n    self.conf )\n', '  File "/home/kochigami/ros/hydro/src/ros-naoqi/nao_robot/nao_apps/nodes/nao_speech.py", line 434, in __init__\n    self.reconfigure(config)\n', '  File "/home/kochigami/ros/hydro/src/ros-naoqi/nao_robot/nao_apps/nodes/nao_speech.py", line 456, in reconfigure\n    Util.parse_vocabulary( config["vocabulary"].encode(\'utf-8\') ),\n', "UnicodeDecodeError: 'ascii' codec can't decode byte 0xe3 in position 0: ordinal not in range(128)\n"]
```

k-okada, 2015-01-27:
エンコーディングに関してはhttps://github.com/ros/dynamic_reconfigure/commits/master
で僕がコミットしたところなので，
rosversion dynamic_reconfigure
で，1.5.38でなけれ自分catkin workspaceにdynamic reconfigureを入れておけばいいともいます．
cd ~/catkin_ws/src
wstool set dynamic_reconfigure http://github.com/ros/dynamic_reconfigure -v
1.5.38 --git
で伝わるかな．

kochigami, 2015-01-27:
エンコーディングについてご教示ありがとうございました。
先程までナオのstartup用のlaunchを立ち上げぱなしにしていたら、次のようなエラーが出てlaunchが落ちていました。

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 551, in __bootstrap_inner
    self.run()
  File "/home/kochigami/ros/hydro/src/ros-naoqi/naoqi_bridge/naoqi_sensors/src/naoqi_sensors/naoqi_camera.py", line 252, in run
    image = self.camProxy.getImageRemote(self.nameId)
  File "/home/kochigami/naoqi/pynaoqi-python2.7-2.1.0.19-linux64/naoqi.py", line 301, in __call__
    return self.__wrapped__.method_missing(self.__method__, *args, **kwargs)
  File "/home/kochigami/naoqi/pynaoqi-python2.7-2.1.0.19-linux64/naoqi.py", line 371, in method_missing
    raise e
RuntimeError: 	ALVideoDevice::getImageRemote
	Socket disconnected

[ERROR] [WallTime: 1422340394.009089] Exception caught:
	ALMotion::setStiffnesses
	Socket disconnected
Traceback (most recent call last):
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rospy/core.py", line 399, in signal_shutdown
    h()
  File "/home/kochigami/ros/hydro/src/ros-naoqi/naoqi_bridge/naoqi_driver/src/naoqi_driver/naoqi_node.py", line 104, in __on_ros_shutdown
    self.join()
  File "/usr/lib/python2.7/threading.py", line 658, in join
    raise RuntimeError("cannot join current thread")
RuntimeError: cannot join current thread
================================================================================REQUIRED process [naoqi_joint_states-1] has died!
process has finished cleanly
log file: /home/kochigami/.ros/log/cd6a6da6-a5e1-11e4-a131-28d2444a1522/naoqi_joint_states-1*.log
Initiating shutdown!
================================================================================
[nao_audio-11] killing on exit
 [nao_tactile-10] killing on exit
[nao_walker-9] killing on exit
[camera-8] killing on exit
[pose_manager-7] killing on exit
[pose_controller-6] killing on exit
[nao_vision-5] killing on exit
 [base_footprint-3] killing on exit
[nao_speech-4] killing on exit
[robot_state_publisher-2] killing on exit
[INFO] [WallTime: 1422340397.503566] Stopping naoqi_camera
[INFO] [WallTime: 1422340397.504198] naoqi_camera stopped
[INFO] [WallTime: 1422340397.507587] Stopping pose_controller
[INFO] [WallTime: 1422340397.508386] pose_controller stopped
[INFO] [WallTime: 1422340397.509097] Stopping nao_vision_interface
[INFO] [WallTime: 1422340397.509984] nao_vision_interface stopped
[naoqi_joint_states-1] killing on exit
[INFO] [WallTime: 1422340398.127958] nao pose_controller stopped.
[INFO] [WallTime: 1422340398.170745] Stopping ROSNaoVisionModule ...
[nao_audio-11] escalating to SIGTERM
 [nao_tactile-10] escalating to SIGTERM
[nao_speech-4] escalating to SIGTERM
[nao_vision-5] escalating to SIGTERM
[nao_audio-11] escalating to SIGKILL
 [nao_tactile-10] escalating to SIGKILL
[nao_vision-5] escalating to SIGKILL
[nao_speech-4] escalating to SIGKILL
Shutdown errors:
 * process[nao_audio-11, pid 30495]: required SIGKILL. May still be running.
 * process[nao_tactile-10, pid 30453]: required SIGKILL. May still be running.
 * process[nao_vision-5, pid 30123]: required SIGKILL. May still be running.
 * process[nao_speech-4, pid 30085]: required SIGKILL. May still be running.
shutting down processing monitor...
... shutting down processing monitor complete
done
```
これが出た後、もう一度launchを上げると、nao_speech.pyが落ちる今までと同じ状態になりました。
naoをリブートしたり、パソコンをリブートしたりして、引き続き確認していきます。

furushchev, 2015-12-16:
@kochigami  Any update? If this problem was already solved, please close this issue.

closed 2015-12-16 by kochigami