jsk-ros-pkg/jsk_robot #270: baxter_tweet.launch could not launch

k-okada, 2015-02-18:
baxter need to load machine setup
```
[http://baxter:11311][133.11.216.63] baxter@mouro:~/ros/indigo/src/jsk-ros-pkg/jsk_robot/jsk_baxter_robot/jsk_baxter_startup$ roslaunch jsk_baxter_lifelog/baxter_tweet.launch 
... logging to /home/baxter/.ros/log/32ef160c-b73c-11e4-88b2-10c37b92512a/roslaunch-mouro-22657.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

ERROR: unknown machine [localhost]
The traceback for the exception was written to the log file
[http://baxter:11311][133.11.216.63] 
```

aginika, 2015-02-23:
I hope #273 has solved this problem.

aginika, 2015-08-10:
I think this now working on real robot, so please close.
The error has never been seen

closed 2015-08-10 by k-okada