jsk-ros-pkg/jsk_robot #292: can't connect mongodb erorr about baxter.launch

wkentaro, 2015-04-09:
I launched baxter.launch and had such error::
```sh
$ roslaunch jsk_baxter_startup baxter.launch
[ERROR] [WallTime: 1428562261.824058] MongoDB process stopped!
[ERROR] [WallTime: 1428562261.824136] Mongo process error! Exit code=100
```
What's the problem or where can I find it by reading the code?

furushchev, 2015-04-09:
this may be because zombie process still exists.
please check

```bash
$ lsof -i:62345 # assume that mongod uses port 62345
```

And if zombie `mongod` is launched unless launching `baxter.launch`, please kill the process.:

```bash
$ sudo pkill mongod
```

then restart `baxter.launch`

wkentaro, 2015-04-09:
It is solved. thank you!

closed 2015-04-09 by wkentaro