jsk-ros-pkg/jsk_robot #293: Pepper can't lift its right hand over its chest.

jiangjun0105, 2015-04-10:
I want Pepper to lift its hand over its head, but it can only lift its right hand over its hip. 
The lift hand is OK.
Does it because the servo is too hot?
But I didn't let it move its arm a lot.

k-okada, 2015-04-10:
please paste your program here or gist https://gist.github.com/

jiangjun0105, 2015-04-10:
I ran the follow command in pepper-interface.l
```
send *pepper* :angle-vector #f(2.0 -2.0 -5.0 85.0 10.0 -70.0 -20.0 -40.0 -30.0 -20.0 50.0 50.0 -50.0 0.0 0.0)
```
Maybe it is hard to understand only by code and numbers.
Can I paste the screenshot of pepper's model here?

k-okada, 2015-04-10:
Sure,

jiangjun0105, 2015-04-10:
Pepper couldn't do this.
It only could lift its right hand up to the height of its waist.
[image]

jiangjun0105, 2015-04-10:
By the way, I think there are some problems with pepper model's right hand, which is completely same with its left hand. I think the right hand is supposed to be symmetrical with the left hand.

k-okada, 2015-04-10:
Ok

So
Can you 
send *ri* :angle-vector (send *pepper* :angle-vector) ?
If yes, the move right arm slightly, may be tilt shoulder pitch angle for 10 degree and send to the robot again. If it works, then move arm and send again

kochigami, 2015-04-11:
Please tell me how to use :angle-vector-sequence method without program being stopped.

If Pepper's program and her movement is stopped, do I also have to separate the program and reduce the number of angle vector sequences like this?
before
```
(send *ri* :angle-vector-sequence (list "angle-vector-sequence1" "angle-vector-sequence2" "angle-vector-sequence3" ))
```
after
```
(send *ri* :angle-vector-sequence (list "angle-vector-sequence1"  "angle-vector-sequence2")) 
(send *ri* :angle-vector-sequence (list "angle-vector-sequence3")) 
```
Thank you!

k-okada, 2015-04-11:
if you call ` (send send *ri* :wait-interpolation)`, then the program
pauses until the motion finishes.
if it does not works, please use `(unix:sleep 10)` to pause the program.

k-okada, 2015-04-11:
oh, are you saying when you call :angle-vector-sequence , then the program
stops (the :anlge-vector-sequence method did not returns) until the motion
of the robot stops?

kochigami, 2015-04-11:
Sorry for my poor English. That is what I wanted to say.
I wondered the program sometimes fails to execute next function because there are many angle vector sequences or it sometimes takes long time to return :angle-vector-sequence method. 

(今、音声ファイル再生と:angle-vector-sequence を同時に使う関数を書いて、それを8個並べて使っているのですが、たまに次の関数の処理に行かず、処理が止まることがありました。
その時は、音声ファイル再生は終わり、ロボットの動きも止まっていて、次の関数の内容は何も実行されていません。
音声ファイル再生だけだと止まらないようなので、動く部分が原因なのだと思います。
角度列 (10~15個くらい) を書きすぎて、処理が止まるのでしょうか、それとも、:angle-vector-sequence はたまに返り値を返すのに時間がかかってしまうのでしょうか。日本語も分かりづらくすみません。)