jsk-ros-pkg/jsk_robot #297: Maybe we need to execute "sudo apt-get install -y ros-indigo-pr2-mannequin-mode" when building Pepper environment on indigo

kochigami, 2015-04-17:
According to https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_pepper_robot, I tried to create Pepper environment.
I used my ubuntu 14.04 PC, in which nothing related to ROS was installed.

When I ran ```rosdep install -y -r --from-paths src --ignore-src```, there were some erros as follows:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved to system dependencies:
jsk_baxter_web: Cannot locate rosdep definition for [ros3djs]
jsk_pr2_calibration: Cannot locate rosdep definition for [pr2_calibration_launch]
jsk_baxter_startup: Missing resource baxter_interface
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/opt/ros/indigo/share
ROS path [2]=/opt/ros/indigo/stacks
ROS path [3]=/home/kochigami/catkin_ws/src
pr2_base_trajectory_action: Missing resource pr2_controllers_msgs
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/opt/ros/indigo/share
ROS path [2]=/opt/ros/indigo/stacks
ROS path [3]=/home/kochigami/catkin_ws/src
Continuing to install resolvable dependencies...
executing command [sudo apt-get install -y ros-indigo-pr2-mannequin-mode]
```
I executed ```sudo apt-get install -y ros-indigo-pr2-mannequin-mode```, and finally completed steps for building Pepper environment. 
Maybe ```ros-indigo-jsk-pcl-ros``` was also installed while executing that command. (I forgot to record log...) 
I don't know whether this occurred only my environment or not, but I reported it just in case.

k-okada, 2015-04-18:
I think this is ok, you have mixed two things here,

1) getting 
```
jsk_baxter_web: Cannot locate rosdep definition for [ros3djs]
jsk_pr2_calibration: Cannot locate rosdep definition for [pr2_calibration_launch]
jsk_baxter_startup: Missing resource baxter_interface
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/opt/ros/indigo/share
ROS path [2]=/opt/ros/indigo/stacks
ROS path [3]=/home/kochigami/catkin_ws/src
pr2_base_trajectory_action: Missing resource pr2_controllers_msgs
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/opt/ros/indigo/share
ROS path [2]=/opt/ros/indigo/stacks
ROS path [3]=/home/kochigami/catkin_ws/src
Continuing to install resolvable dependencies...
```
means they are not released as deb package yet, and it shows error, but in order to ignore this , we put `-r` option in rosdep. Looking at the warning message, it seems to related to pr2 and baxter, so you can skip that.

2) Next thing is 
```
executing command [sudo apt-get install -y ros-indigo-pr2-mannequin-mode]
[sudo] password for k-okada: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done

```
and you're asked to input your password for sudo, that what `rosdep install`  do. It installed depended packages for source directory.

kochigami, 2015-04-18:
Oh, I see !
I really appreciate your explanation.

closed 2015-04-18 by kochigami