jsk-ros-pkg/jsk_robot #302: BaxterのIKが解けない

aginika, 2015-04-28:
以下のような場合に前方１６ｃｍ移動させようとするとikが解けないのですが、
解けるようにすることはできますでしょうか？
以下はinverse-kinematicsの失敗ログです。

```
;; ik success log at Tue Apr 28 18:35:10 2015 on (9.12 host14.storm.ros.org Thu Apr 23 17:11:49 UTC 2015  1.0.6)
;;
;; link-list (#<bodyset-link #X7864e88 left_upper_shoulder  64.027 259.027 1079.626 / 2.479 0.0 0.0> #<bodyset-link #X78076b0 left_lower_shoulder  9.648 301.499 1349.976 / -0.663 -1.152 1.571> #<bodyset-link #X77ea7e0 left_upper_elbow  42.337 275.968 1443.162 / -0.008 0.339 0.25> #<bodyset-link #X77eb560 left_lower_elbow  191.501 209.77 1659.941 / 0.465 1.144 -0.97> #<bodyset-link #X6ad2920 left_upper_forearm  229.799 228.967 1565.624 / 3.063 0.371 2.926> #<bodyset-link #X6ad2e48 left_lower_forearm  320.585 279.866 1315.533 / 0.0 -0.001 -1.37> #<bodyset-link #X6ad3178 left_wrist  436.56 279.882 1315.602 / 0.2 1.57 0.2>)
;; move-target #<cascaded-coords #Xc611350 :larm-end-coords  650.11 279.912 1315.731 / 0.0 -0.001 0.0>
;; rotatoin-axis t, translation-axis t
;; thre 1, rthre 0.017453, stop 50
(load "package://baxtereus/baxter.l")
(setq c0 '(#s(coordinates plist nil rot #2f((1.0 -0.000138 -0.000605) (0.000138 1.0 -0.000122) (0.000605 0.000122 1.0)) pos #f(810.11 279.912 1315.73))))
(setq av0 #f(-44.9561 97.0093 -113.994 -144.932 139.438 -5.58105 -68.335 -11.4917 -97.0752 -2.39502 94.4824 134.517 -91.5601 8.74512 0.021973))
(defun baxter-robot-2015-04-28-18-35-10-setup () (let ((r (instance baxter-robot :init))) (setq *robot* r) (progn (send r :newcoords (make-coords :4x4 #2f((1.0 0.0 0.0 0.0) (0.0 1.0 0.0 0.0) (0.0 0.0 1.0 950.0) (0.0 0.0 0.0 1.0)))) (mapc #'(lambda (j a) (send* j :joint-angle a nil)) (list (send r :head_pan) (send r :left_s0) (send r :left_s1) (send r :left_e0) (send r :left_e1) (send r :left_w0) (send r :left_w1) (send r :left_w2) (send r :right_s0) (send r :right_s1) (send r :right_e0) (send r :right_e1) (send r :right_w0) (send r :right_w1) (send r :right_w2)) '(-44.9561 97.0093 -113.994 -144.932 139.438 -5.58105 -68.335 -11.4917 -97.0752 -2.39502 94.4824 134.517 -91.5601 8.74512 0.021973)) (objects (list r))) (objects (list *robot*))))
(defun baxter-robot-2015-04-28-18-35-10-check () (let ((r *robot*)) (send* r :inverse-kinematics (list (list (make-coords :pos #f(810.11 279.912 1315.73) :rot #2f((1.0 -0.000138 -0.000605) (0.000138 1.0 -0.000122) (0.000605 0.000122 1.0)))) :dump-command nil :debug-view t :move-target (let* ((p (send r :link "left_wrist")) (mt (make-cascoords :coords (send (send p :copy-worldcoords) :transform (make-coords :4x4 #2f((-3.673205e-06 -9.281176e-17 -1.0 0.0) (2.212062e-16 1.0 -9.281257e-17 -1.136868e-13) (1.0 -2.212065e-16 -3.673205e-06 213.55) (0.0 0.0 0.0 1.0))))))) (send p :assoc mt) mt) :link-list (list (send r :link "left_upper_shoulder") (send r :link "left_lower_shoulder") (send r :link "left_upper_elbow") (send r :link "left_lower_elbow") (send r :link "left_upper_forearm") (send r :link "left_lower_forearm") (send r :link "left_wrist")) :move-target (let* ((p (send r :link "left_wrist")) (mt (make-cascoords :coords (send (send p :copy-worldcoords) :transform (make-coords :4x4 #2f((-3.673205e-06 -9.281176e-17 -1.0 0.0) (2.212062e-16 1.0 -9.281257e-17 -1.136868e-13) (1.0 -2.212065e-16 -3.673205e-06 213.55) (0.0 0.0 0.0 1.0))))))) (send p :assoc mt) mt) :collision-avoidance-link-pair nil :link-list (list (send r :link "left_upper_shoulder") (send r :link "left_lower_shoulder") (send r :link "left_upper_elbow") (send r :link "left_lower_elbow") (send r :link "left_upper_forearm") (send r :link "left_lower_forearm") (send r :link "left_wrist"))))))
(defun ik-setup () (baxter-robot-2015-04-28-18-35-10-setup))
(defun ik-check () (baxter-robot-2015-04-28-18-35-10-check))
(setq ik-failed '((:dif-pos . (#f(18.2789 9.8726 -31.3867))) (:dif-rot . (#f(0.008687 0.003955 0.001696))
)))

```

k-okada, 2015-04-28:
サンプルプログラムを贈ってもらえますでしょうか？

wkentaro, 2015-04-28:
```lisp
(setq av0 #f(0.0 97.3149 -121.216 -143.873 138.766 -5.67992 -73.1565 -16.2255 -20.0 -25.0 40.0 60.0 20.0 80.0 0.0))
(send *baxter* :angle-vector av0)
(send *baxter* :larm :move-end-pos #f(160 0 0) :world)
```

aginika, 2015-04-28:
@wkentaro ありがとうございます。

k-okada, 2015-04-28:
これは大変面白い問題です．

euslispにはavoid-nspace-gain, avoid-weight-gain, （に加えてavoid-collison-gain）があるのは，講義，演習，jskeusマニュアル，@snozawaさんの博士論文を通じて皆さんよくご存知:stuck_out_tongue_winking_eye:だと思いますが，　前者は関節のリミットに近ければそから避けるようにnull spaceを使って避ける方向に動く，後者は関節のリミットに向かう方向に動いている場合はその部分のWeightを小さくして（実際は大きくだっけ？）その関節を余り使わずに，他の関節を使って動いていこうとしています．

後もうひとつIK関連のプログラムを呼び出すときは:debug-view t とすると，いろいろな表示がでて，:debug-view :no-messageなどのオプションもあることは普段からソースコードを見ていれば常識ですね！

さて，今回は初期姿勢がだいぶリミットにかかっているのが問題で，第一，第二軸がアウトで，それが，全く動いてくれないのがアレなきがします．ということで，それを何とかして回避する，というのが解決策になります．

```
(send *baxter* :larm :move-end-pos #f(160 0 0) :world :avoid-weight-gain 0.5 :avoid-nspace-gain 0.1 :debug-view t :dif-pos-ratio 0.1 :dif-rot-ratio 0.1 :stop 200)
```
は色々パラメータを変えてみましたが，どうもうまく行かないね，という残骸です．

try 1は最初にnull spaceを使って，少しリミットから離そうという作戦ですが，普通だとうまくいかず，`:rotateion-aixs nil`で一回，手先の向きを変えてもいいから，リミットから離して，それを初期姿勢にして動かしました．

try 2 は1mmづつ動かしたら，うまく行った，という例です．2mmだとダメなので微妙です．

try 3はreset-manip-pose にしてから動かすものです．これってuntuck-pose だっけ？なかなかいい姿勢です．

try 3かtry 1だとおもいますが，実機がav0のときに，try3で出てきた姿勢をおくったときに，どういう挙動に成るかが知りたいです．今回eus上ではヤコビアンで速度的に連続な領域で動作をつくっていますが，結局その最終姿勢をロボットに送って，補間はロボットにやらせています．このとき，たとえあ腕が１８０度グルンと回るようなことがあるとあまり良くないので，こんどはtry1の結果を送りたいです．あるいはtry1で腕の向きがかわった姿勢を一回送ってから，最終結果を送ります．それでもだめなら，こちらで連続的な運動を生成してそれを送るのが良いです．という風になります．

いずれにせよ，方針をきめたらbaxter-interface.lなどで対応して，みんなが書いているプログラムの変更は内容にするので，上のことを調べてくれレバ，それだけでよいです．

```
(setq av0 #f(0.0 97.3149 -121.216 -143.873 138.766 -5.67992 -73.1565 -16.2255 -20.0 -25.0 40.0 60.0 20.0 80.0 0.0))
(send *baxter* :angle-vector av0)
(send *viewer* :draw-objects)
(send *baxter* :larm :move-end-pos #f(160 0 0) :world :debug-view t) ;; ng
;;(send *baxter* :larm :move-end-pos #f(160 0 0) :world :avoid-weight-gain 0.1 :debug-view t) ;; ng
;;(send *baxter* :larm :move-end-pos #f(160 0 0) :world :avoid-weight-gain 0.5 :avoid-nspace-gain 0.1 :debug-view t :dif-pos-ratio 0.1 :dif-rot-ratio 0.0 :stop 200)
#|

20.irteusgl$ load "hoge.l"
loop:   1
union-link-list: (left_upper_shoulder left_lower_shoulder left_upper_elbow left_lower_elbow left_upper_forearm left_lower_forearm left_wrist)
move-target: (#<cascaded-coords #X60551e0 :larm-end-coords  649.768 289.751 335.873 / 8.337e-05 0.0 5.188e-06>)
targe-coordst: (#<coordinates #X62b4108  809.768 289.751 335.873 / 8.337e-05 0.0 5.188e-06>)
vel-pos :100.000  -0.008   0.031 
vel-rot :  0.000   0.000   0.000 
vel-pos-norm : 100.000/  1.000
vel-rot-norm :   0.000/  0.017
angle :   97.3  -121.2  -143.9   138.8    -5.7   -73.2   -16.2 
 min  :  -97.5  -123.0  -175.0    -2.9  -175.3   -90.0  -175.3 
 max  :   97.5    60.0   175.0   150.0   175.3   120.0   175.3 
usrwei:  1.000   1.000   1.000   1.000   1.000   1.000   1.000 
ocost :100000000000000000000.000 100000000000000000000.000 100000000000000000000.000 100000000000000000000.000 100000000000000000000.000 100000000000000000000.000 100000000000000000000.000 
cost  :2793.069 810.736   5.128  17.240   0.021  10.521   0.062 
weight:  1.000   1.000   1.000   1.000   1.000   1.000   1.000 
k     :  0.000 (manipulability:  0.038, gain:  0.001, limit:  0.100, len:7)

loop:   2
union-link-list: (left_upper_shoulder left_lower_shoulder left_upper_elbow left_lower_elbow left_upper_forearm left_lower_forearm left_wrist)
move-target: (#<cascaded-coords #X60551e0 :larm-end-coords  748.374 270.588 386.522 / -0.024 -0.059 -0.04>)
targe-coordst: (#<coordinates #X62b4108  809.768 289.751 335.873 / 8.337e-05 0.0 5.188e-06>)
vel-pos : 57.843  22.769 -53.268 
vel-rot :  0.040   0.058   0.025 
vel-pos-norm :  81.865/  1.000
vel-rot-norm :   0.075/  0.017
angle :   97.5  -123.0  -142.4   122.1    -4.7   -61.5   -19.3 
 min  :  -97.5  -123.0  -175.0    -2.9  -175.3   -90.0  -175.3 
 max  :   97.5    60.0   175.0   150.0   175.3   120.0   175.3 
usrwei:  1.000   1.000   1.000   1.000   1.000   1.000   1.000 
ocost :2793.069 810.736   5.128  17.240   0.021  10.521   0.062 
cost  :2793.069 2621.242   4.663   2.664   0.018   3.602   0.074 
weight:  0.000   0.000   1.000   1.000   1.000   1.000   0.931 
k     :  0.000 (manipulability:  0.053, gain:  0.001, limit:  0.100, len:7)

loop:  49
union-link-list: (left_upper_shoulder left_lower_shoulder left_upper_elbow left_lower_elbow left_upper_forearm left_lower_forearm left_wrist)
move-target: (#<cascaded-coords #X60551e0 :larm-end-coords  780.342 273.197 389.386 / -0.002 -0.006 -0.014>)
[... 24 lines cut ...]
(setq step 1);; 1 だと解ける 2だと解けない
(dotimes (i (/ 160 step)) 
  (print i)
  (send *baxter* :larm :move-end-pos (float-vector step 0 0) :world :debug-view t)
  (send *viewer* :draw-objects)
  )

#|
angle :   96.7  -123.0  -140.5   115.5    -4.8   -50.9   -18.0 
 min  :  -97.5  -123.0  -175.0    -2.9  -175.3   -90.0  -175.3 
 max  :   97.5    60.0   175.0   150.0   175.3   120.0   175.3 
usrwei:  1.000   1.000   1.000   1.000   1.000   1.000   1.000 
ocost :2793.069 2621.242   4.174   1.685   0.018   1.861   0.068 
cost  :2793.069 2621.242   4.175   1.684   0.018   1.861   0.068 
weight:  0.000   0.000   0.193   1.000   1.000   0.349   1.000 
k     :  0.000 (manipulability:  0.052, gain:  0.001, limit:  0.100, len:7)
|#

;; try 3
(send *baxter* :angle-vector av0)
(setq p (send *baxter* :larm :end-coords :copy-worldcoords))
(send p :translate #f(160 0 0) :world)
(send *baxter* :reset-manip-pose)
(send *baxter* :larm :inverse-kinematics p :debug-view t) ;; ng
```

aginika, 2015-04-28:
@k-okada ありがとうございます。
明日それらを試してみようと思います。

snozawa, 2015-04-28:
ぱっとみですが、右後ろ向いたときに左後ろをlook-atするIkが解けない問題と同じです。
多分ですが、勾配を降る方向でないところに解があるので、Euslispの現行IKだと相当難しいIkです。
なので、try3がおすすめです。

k-okada, 2015-04-28:
https://github.com/jsk-ros-pkg/jsk_pr2eus/issues/72
ですね。
これと同じアプローチが多軸でできたらいいですが、今度は動く時に手がグリングリン回りますね。

snozawa, 2015-04-28:
このときの
https://github.com/jsk-ros-pkg/jsk_pr2eus/issues/72#issuecomment-67111690
を思い出すと、
> 多分ですが、勾配を降る方向でないところに解があるので、Euslispの現行IKだと相当難しいIkです。

なので、`try 3`作戦がもっとも素直＋費用対効果が良いと思ったきがしました。
（今回は自由度数がおおいので、ぐりんと動かないかのチェックが必要ですが）

aginika, 2015-04-29:
Try3を実機でやりました。

[image]

aginika, 2015-04-29:
素人で申し訳ないのですが、以下の質問をさせていただきたいです。
- 今回の`reset-manip-pose`のような様々な点へのIKが解きやすい姿勢というものは存在するまたは求めることはできるのでしょうか？
- どんな愚行なプログラムやオフラインの計算でもいいのですが、可能な姿勢であればどんな姿勢でも求められる現実的な方法は存在するのでしょうか？

k-okada, 2015-04-29:
try2 出だした姿勢を送った時の比較がしたいです．

以下の答えは厳密にはnilになります．なので，ikfastやmoveitがでてきますが，
こんかいみたいに対象がかなり制限されているときは，そこそこの案が有るともいます．

https://github.com/start-jsk/2014-semi/blob/34b7cf27a5a77ea27afe98986d340c4a4b6731c3/jsk_2014_picking_challenge/scripts/main.l#L21
みたいなプログラムをつくっておくのがいいとおもいます．
ところでロボットのend-coordsはどこにあるかな？掃除機の先にちゃんと付いているかな？

aginika, 2015-04-29:
> try2 出だした姿勢を送った時の比較がしたいです．

了解です。

掃除機の先にはまだついてないですね。urdf -> colladaのところで止まっています。

k-okada, 2015-04-29:
表示はどうでもよくて，end-coordsだけがついていたらいいです．

garaemon, 2015-04-29:
try3でいいと思いますが、もしかしたらIKを解く順番は動かす順番ではなくて、必要な精度できめたほうがいいかもしれませんね。

0. 目標の位置で解く
1. 少し手前に動かして(しきい値をゆるめにして)予備姿勢を作る

aginika, 2015-04-29:
Try2:
[image]

aginika, 2015-04-30:
@k-okada 
> https://github.com/start-jsk/2014-semi/blob/34b7cf27a5a77ea27afe98986d340c4a4b6731c3/jsk_2014_picking_challenge/scripts/main.l#L21

はikが解けるかのテストを書くということですか？

また、このtry2とtry3からbaxter-interface.lはどのように変更がかかるのでしょうか？

k-okada, 2015-04-30:
３はなんで一回手が上に上がっている？そういう姿勢を送っている？
２はいいように見える．

aginika, 2015-04-30:
3は初期位置とゴールだけを与えています。３の場合は肩周りが大きく変動しているために全体として手が上がっています。
2はmove-end-posした値は逐一送っています。

k-okada, 2015-04-30:
try 1はどうだろう？

k-okada, 2015-04-30:
https://github.com/jsk-ros-pkg/jsk_robot/pull/308
をついかしたので，これをloadするようにしてください．いろいろ頑張って
IKをときます．

aginika, 2015-05-01:
@k-okada I will try to use that

closed 2015-05-01 by aginika