jsk-ros-pkg/jsk_robot #305: pepper_description and pepper_bringup conflict for doc jobs

vrabaud, 2015-04-28:
The official packages are now at:
https://github.com/ros-naoqi/
If you have anything you want to transfer, Kei has access.
Then, please remove the pepper_* packages that are not official and the doc jobs will show correctly on:
http://wiki.ros.org/pepper_bringup
Thx !

k-okada, 2015-04-29:
I see, now it completely removed from our repo
https://github.com/jsk-ros-pkg/jsk_robot/pull/307

chiwunau, 2015-05-03:
@k-okada sensei
Can I add ros-naoqi/pepper_robot to jsk_common/jsk.rosinstall ?
catkin build peppereus does not pass because of lack of pepper.urdf
and they don't have it if I do sudo apt-get install ros-hydro-pepper-description.

vrabaud, 2015-05-03:
fyi, the urdf in pepper_robot is in is at: https://github.com/ros-naoqi/pepper_robot/blob/master/pepper_description/urdf/pepper1.0_generated_urdf/pepper.urdf

k-okada, 2015-05-03:
vincent is right, and please make sure that if you use shadow-fixed repo
and if you still have problem,  paste error message for peppereus

chiwunau, 2015-05-03:
forgot to apt-get upgrade
It works now, sorry for the trouble.

k-okada, 2015-05-04:
please not that if you want to create papper.l file, you need to manually
install ros-%ROS_DISTRO-pepper-meshes and read license carefully, then run
catkin b peppereus again.

closed 2015-05-04 by k-okada