jsk-ros-pkg/jsk_robot #312: Baxterの腕がコリジョンしているかどうかを確認したい

aginika, 2015-05-01:
:collision-avoidance-linksはよける努力はするが、実際によけてない姿勢を返すことがあると聞いたのですが、
どのように返してきた姿勢でcollisionしているか調べるのが最適でしょうか？

pqp-collision-avoidanceは物体間での確認などを行っているように見えます。

おすすめは`:self-collision-check`でしょうか？ただ出力が初期姿勢の時から以下のようになっていて
思っているのと違うのですが、使い方が間違っているのでしょうか？
```
send *baxter* :self-collision-check
((#<bodyset-link #X6159770 torso  0.0 0.0 0.0 / 0.0 0.0 0.0> . #<bodyset-link #X5f7fa50 right_upper_shoulder  64.027 -259.027 129.626 / -1.134 0.0 0.0>) (#<bodyset-link #X6159770 torso  0.0 0.0 0.0 / 0.0 0.0 0.0> . #<bodyset-link #X88a21e8 right_lower_shoulder  93.188 -321.563 399.976 / -1.134 -0.436 -1.571>) (#<bodyset-link #X6159770 torso  0.0 0.0 0.0 / 0.0 0.0 0.0> . #<bodyset-link #X5faa2d0 left_upper_shoulder  64.027 259.027 129.626 / 1.134 0.0 0.0>) (#<bodyset-link #X6159770 torso  0.0 0.0 0.0 / 0.0 0.0 0.0> . #<bodyset-link #X5fab8a8 left_lower_shoulder  93.188 321.563 399.976 / 1.134 -0.436 -1.571>) (#<bodyset-link #X6159770 torso  0.0 0.0 0.0 / 0.0 0.0 0.0> . #<bodyset-link #X5efce98 screen  182.7 0.0 686.0 / 1.571 1.499e-15 1.751>) (#<bodyset-link #X5e9b320 collision_head_link_2  110.0 0.0 750.0 / 0.0 0.0 0.0> . #<bodyset-link #X58ea160 collision_head_link_1  110.0 0.0 750.0 / 0.0 0.0 0.0>) (#<bodyset-link #X6fe20b0 right_upper_elbow_visual  134.171 -409.452 445.196 / -1.134 1.134 6.693e-12> . #<bodyset-link #X8a3a868 right_upper_elbow  132.256 -405.345 443.083 / -0.03 0.767 0.943>) (#<bodyset-link #X6fe20b0 right_upper_elbow_visual  134.171 -409.452 445.196 / -1.134 1.134 6.693e-12> . #<bodyset-link #X6085480 right_lower_elbow  282.406 -622.397 506.082 / -0.485 0.401 -0.886>) (#<bodyset-link #X608add8 right_upper_forearm_visual  354.083 -660.196 471.766 / 1.531 0.794 2.161> . #<bodyset-link #X608b300 right_upper_forearm  366.781 -666.893 465.687 / 1.314 0.49 2.029>) (#<bodyset-link #X608add8 right_upper_forearm_visual  354.083 -660.196 471.766 / 1.531 0.794 2.161> . #<bodyset-link #X6075998 right_lower_forearm  589.515 -774.638 355.419 / 1.129 0.56 0.365>) (#<bodyset-link #X606ee30 right_wrist  631.558 -685.847 293.786 / 2.499 -0.307 2.162> . #<bodyset-link #X60036e0 right_hand_camera  655.517 -561.757 246.338 / -2.636 0.913 2.624>) (#<bodyset-link #X5f9b6c8 left_upper_elbow_visual  134.171 409.452 445.196 / 1.134 1.134 6.693e-12> . #<bodyset-link #X5fa25f8 left_upper_elbow  132.256 405.345 443.083 / 0.03 0.767 -0.943>) (#<bodyset-link #X5f9b6c8 left_upper_elbow_visual  134.171 409.452 445.196 / 1.134 1.134 6.693e-12> . #<bodyset-link #X5f953b8 left_lower_elbow  282.406 622.397 506.082 / 0.485 0.401 -2.256>) (#<bodyset-link #X58edfb8 left_upper_forearm_visual  354.083 660.196 471.766 / -1.531 0.794 -2.161> . #<bodyset-link #X5b9da80 left_upper_forearm  366.781 666.893 465.687 / -1.314 0.49 -2.029>) (#<bodyset-link #X58edfb8 left_upper_forearm_visual  354.083 660.196 471.766 / -1.531 0.794 -2.161> . #<bodyset-link #X5f98af0 left_lower_forearm  589.515 774.638 355.419 / -1.129 0.56 2.777>) (#<bodyset-link #X5f8f510 left_wrist  631.558 685.847 293.786 / -2.499 -0.307 -2.162> . #<bodyset-link #X5f838a8 left_hand_camera  642.678 568.861 227.346 / -0.505 -0.913 2.624>))

```

k-okada, 2015-05-02:
```
$ (send *baxter* :self-collision-check :pairs (list (cons (send *baxter*
:larm :elbow-p :child-link)  (send *baxter* :rarm :elbow-p :child-link))))
```

please use :pair option to specify which link pair to be checked, ->
https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L2388

by default we have method to calculate all link pair
https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L2373, but
this is very difficult problem, for example,
adjacent two link usually always collide, and if the two joint is
orthogonal, then three links are always collide.
 :collision-check-pairs only checks neighbor joints,

moreover, the model generated from urdf/collada has different problem,
since they use "fixed" joint to connect
two body parts, but in the euslisp layer they're two different links,
connected with fixed joint. this makes the problem
more complex.

so my suggestion is to create collision-check pair by hand, may be check
collide between head and elbow, and hand, and also check
collide of these links between left and right may sufficient.

aginika, 2015-05-03:
@k-okada Thanks a lot, I think I understood fully!

closed 2015-05-03 by aginika

k-okada, 2016-04-27:
Cc: @h-kamada