jsk-ros-pkg/jsk_robot #322: Failed to compile peppereus

garaemon, 2015-05-06:
pepper_description is not yet released

```
[peppereus] ==> '/home/jskuser/ros/hydro/build/peppereus/build_env.sh /usr/bin/m
ake cmake_check_build_system' in '/home/jskuser/ros/hydro/build/peppereus'
-- Using CATKIN_DEVEL_PREFIX: /home/jskuser/ros/hydro/devel
-- Using CMAKE_PREFIX_PATH: /home/jskuser/ros/hydro/devel;/home/jskuser/ros/hydr
o_parent/devel;/opt/ros/hydro
-- This workspace overlays: /home/jskuser/ros/hydro/devel;/home/jskuser/ros/hydr
o_parent/devel;/opt/ros/hydro
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Python version: 2.7
-- Using Debian Python package layout
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/jskuser/ros/hydro/build/peppereus/test_r
esults
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- catkin 0.5.90
-- Using these message generators: gencpp;geneus;genlisp;genpy
-- [roseus.cmake] message_generation Requires: gencpp;genlisp;genmsg;genpy
-- [roseus.cmake] generate all roseus messages
-- [roseus.cmake] peppereus depends on actionlib;actionlib_msgs;diagnostic_msgs;
geometry_msgs;nav_msgs;roscpp;rosgraph_msgs;sensor_msgs;std_msgs;std_srvs;tf2_ms
gs;tf;trajectory_msgs;visualization_msgs
-- [roseus.cmake] peppereus will compile peppereus
CMake Error at CMakeLists.txt:16 (message):
  Could not found pepper.urdf in
  /home/jskuser/ros/hydro/devel/share/pepper_description/urdf/pepper1.0_generated_urdf/pepper.urdf
  and
  /home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_robot/jsk_pepper_robot/pepper_description/urdf/pepper1.0_generated_urdf/pepper.urdf

-- Configuring incomplete, errors occurred!
make: *** [cmake_check_build_system]  1
[peppereus] <== '/home/jskuser/ros/hydro/build/peppereus/build_env.sh /usr/bin/m
ake cmake_check_build_system' failed with return code '2'
```

k-okada, 2015-05-06:
Please install pepper_description from den, now it's released from aldebaran

k-okada, 2015-05-06:
pepper_description is removes from jsk_robot, so try

   - catkin clean --orphans

before catkin build, see
http://catkin-tools.readthedocs.org/en/latest/cheat_sheet.html

2015年5月6日水曜日、Kei Okada<k-okada@jsk.t.u-tokyo.ac.jp>さんは書きました:

> Please install pepper_description from den, now it's released from
> aldebaran
>
> 2015年5月6日水曜日、Ryohei Ueda<notifications@github.com
>
>> pepper_description is not yet released
>>
>> [peppereus] ==> '/home/jskuser/ros/hydro/build/peppereus/build_env.sh /usr/bin/m
>> ake cmake_check_build_system' in '/home/jskuser/ros/hydro/build/peppereus'
>> -- Using CATKIN_DEVEL_PREFIX: /home/jskuser/ros/hydro/devel
>> -- Using CMAKE_PREFIX_PATH: /home/jskuser/ros/hydro/devel;/home/jskuser/ros/hydr
>> o_parent/devel;/opt/ros/hydro
>> -- This workspace overlays: /home/jskuser/ros/hydro/devel;/home/jskuser/ros/hydr
>> o_parent/devel;/opt/ros/hydro
>> -- Using PYTHON_EXECUTABLE: /usr/bin/python
>> -- Python version: 2.7
>> -- Using Debian Python package layout
>> -- Using CATKIN_ENABLE_TESTING: ON
>> -- Call enable_testing()
>> -- Using CATKIN_TEST_RESULTS_DIR: /home/jskuser/ros/hydro/build/peppereus/test_r
>> esults
>> -- Found gtest sources under '/usr/src/gtest': gtests will be built
>> -- catkin 0.5.90
>> -- Using these message generators: gencpp;geneus;genlisp;genpy
>> -- [roseus.cmake] message_generation Requires: gencpp;genlisp;genmsg;genpy
>> -- [roseus.cmake] generate all roseus messages
>> -- [roseus.cmake] peppereus depends on actionlib;actionlib_msgs;diagnostic_msgs;
>> geometry_msgs;nav_msgs;roscpp;rosgraph_msgs;sensor_msgs;std_msgs;std_srvs;tf2_ms
>> gs;tf;trajectory_msgs;visualization_msgs
>> -- [roseus.cmake] peppereus will compile peppereus
>> CMake Error at CMakeLists.txt:16 (message):
>>   Could not found pepper.urdf in
>>   /home/jskuser/ros/hydro/devel/share/pepper_description/urdf/pepper1.0_generated_urdf/pepper.urdf
>>   and
>>   /home/jskuser/ros/hydro/src/jsk-ros-pkg/jsk_robot/jsk_pepper_robot/pepper_description/urdf/pepper1.0_generated_urdf/pepper.urdf
>>
>>
>> -- Configuring incomplete, errors occurred!
>> make: *** [cmake_check_build_system]  1
>> [peppereus] <== '/home/jskuser/ros/hydro/build/peppereus/build_env.sh /usr/bin/m
>> ake cmake_check_build_system' failed with return code '2'
>>
>>
>
>
> ◉ Kei Okada

garaemon, 2015-05-06:
How can I install it?

k-okada, 2015-05-06:
rosdep isntall jsk_pepper_startup or apt-get ros-hydro-pepper-description

garaemon, 2015-05-06:
I see

closed 2015-05-06 by garaemon