jsk-ros-pkg/jsk_robot #341: pepper実機の関節角を動かそうとするとシミュレータが立ち上がる

kochigami, 2015-06-01:
デモ前日に、ペッパーのパソコンのパッケージをアップデートしてしまい、この現象が起こりました。
```
roscd peppereus
roseus pepper-interface.l
pepper-init
```
の後、
```
;; extending gcstack 0x6728290[32738] --> 0x7c93980[65476] top=7fe1
[ WARN] [1433162659.695376752]: [joint_trajectory] action server is not found
[ WARN] [1433162659.695472650]:      goal=0, cancel=0, feedback=0, result=0
[ WARN] [1433162659.695530702]: #<controller-action-client #X735f5c8 joint_tra\
jectory> is not respond, pepper-interface is disabled
```
という警告文が出てシミュレータが立ち上がります。
joint_trajectoryがあるべき場所を探します。

kochigami, 2015-06-01:
```
rosrun roseus generate-all-msg-srv.sh --all
```
をして、
```
 [naoqi_joint_states.py] create dummy effort (error) from sensord joint angle and desired joint angle #37 
```
がまだマージされていないので、書き足してみたけれど結果は変わりませんでした。

k-okada, 2015-06-01:
rostopic info /joint_trajectory/status

してみて下さい．これが出ていないとダメです．なにか立ち上げ忘れている気がします．

また，

rostopic list

してみて下さい．

kochigami, 2015-06-01:
ご教示ありがとうございます。
```
Type: actionlib_msgs/GoalStatusArray

Publishers: None

Subscribers: 
 * /pepper_1433170081822103285 (http://192.168.97.25:48247/)
```
となりました。
おそらく、nao_pose/launch/pose_manager.launch がpepper_robot/peppr_bringup/launch/pepper.launch に足りないのだと思います。試してみます。

closed 2015-06-01 by kochigami

k-okada, 2015-06-02:
上手くいったかな？
動いているときの
rosnode list
rostopic list
をメモってくれるとうれしいです

kochigami, 2015-06-03:
```roslaunch jsk_pepper_startup jsk_pepper_startup.launch```して、pepper_bringup/launch/pepper.launch にnao_pose/launch/pose_manager.launch があるかどうかで比較しました。

nao_pose/pose_manager.launch がないとき
```
nao@pepper:~$ rosnode list
/camera/depth/camera
/camera/front/camera
/cmd_vel_relay
/image_saver
/joy_client
/nao_alife
/nao_audio
/nao_behaviors
/nao_logger
/nao_speech
/nao_vision
/naoqi_joint_states
/naoqi_moveto
/pepper_laser
/pepper_twitter
/robot_state_publisher
/rosout
/teleop_joy
/turtlebot_teleop_joystick
/tweet
```

```
nao@pepper:~$ rostopic list
/camera/depth/camera/camera_info
/camera/depth/camera/image_raw
/camera/depth/camera/parameter_descriptions
/camera/depth/camera/parameter_updates
/camera/front/camera/camera_info
/camera/front/camera/image_raw
/camera/front/camera/parameter_descriptions
/camera/front/camera/parameter_updates
/camera/top/camera/camera_info
/camera/top/camera/image_raw
/cmd_vel
/cmd_vel_mux/input/teleop
/diagnostics
/imu
/joint_angles
/joint_states
/joint_stiffness
/joy
/move_base_simple/goal
/nao_audio/audio_source_localization
/nao_speech/parameter_descriptions
/nao_speech/parameter_updates
/odom
/pepper/laser/ground_left/pointcloud
/pepper/laser/ground_left/scan
/pepper/laser/ground_right/pointcloud
/pepper/laser/ground_right/scan
/pepper/laser/shovel/pointcloud
/pepper/laser/shovel/scan
/pepper/laser/srd_front/pointcloud
/pepper/laser/srd_front/scan
/pepper/laser/srd_left/pointcloud
/pepper/laser/srd_left/scan
/pepper/laser/srd_right/pointcloud
/pepper/laser/srd_right/scan
/pepper_navigation/front
/pepper_tweet
/rosout
/rosout_agg
/run_behavior/cancel
/run_behavior/feedback
/run_behavior/goal
/run_behavior/result
/run_behavior/status
/speech
/speech_action/cancel
/speech_action/feedback
/speech_action/goal
/speech_action/result
/speech_action/status
/speech_vocabulary_action/cancel
/speech_vocabulary_action/feedback
/speech_vocabulary_action/goal
/speech_vocabulary_action/result
/speech_vocabulary_action/status
/tf
/tweet
/word_recognized
```

nao_pose/pose_manager.launch がある時
```
nao@pepper:~$ rosnode list
/camera/depth/camera
/camera/front/camera
/cmd_vel_relay
/image_saver
/joy_client
/nao_alife
/nao_audio
/nao_behaviors
/nao_logger
/nao_speech
/nao_vision
/naoqi_joint_states
/naoqi_moveto
/pepper_laser
/pepper_twitter
/pose_controller
/pose_manager
/robot_state_publisher
/rosout
/teleop_joy
/turtlebot_teleop_joystick
/tweet
```

```
nao@pepper:~$ rosnode list
/camera/depth/camera
/camera/front/camera
/cmd_vel_relay
/image_saver
/joy_client
/nao_alife
/nao_audio
/nao_behaviors
/nao_logger
/nao_speech
/nao_vision
/naoqi_joint_states
/naoqi_moveto
/pepper_laser
/pepper_twitter
/pose_controller
/pose_manager
/robot_state_publisher
/rosout
/teleop_joy
/turtlebot_teleop_joystick
/tweet
nao@pepper:~$ rostopic list 
/body_pose/cancel
/body_pose/feedback
/body_pose/goal
/body_pose/result
/body_pose/status
/body_pose_naoqi/cancel
/body_pose_naoqi/feedback
/body_pose_naoqi/goal
/body_pose_naoqi/result
/body_pose_naoqi/status
/camera/depth/camera/camera_info
/camera/depth/camera/image_raw
/camera/depth/camera/parameter_descriptions
/camera/depth/camera/parameter_updates
/camera/front/camera/camera_info
/camera/front/camera/image_raw
/camera/front/camera/parameter_descriptions
/camera/front/camera/parameter_updates
/camera/top/camera/camera_info
/camera/top/camera/image_raw
/cmd_vel
/cmd_vel_mux/input/teleop
/diagnostics
/imu
/joint_angles
/joint_angles_action/cancel
/joint_angles_action/feedback
/joint_angles_action/goal
/joint_angles_action/result
/joint_angles_action/status
/joint_states
/joint_stiffness
/joint_stiffness_trajectory/cancel
/joint_stiffness_trajectory/feedback
/joint_stiffness_trajectory/goal
/joint_stiffness_trajectory/result
/joint_stiffness_trajectory/status
/joint_trajectory/cancel
/joint_trajectory/feedback
/joint_trajectory/goal
/joint_trajectory/result
/joint_trajectory/status
/joy
/move_base_simple/goal
/nao_audio/audio_source_localization
/nao_speech/parameter_descriptions
/nao_speech/parameter_updates
/odom
/pepper/laser/ground_left/pointcloud
/pepper/laser/ground_left/scan
/pepper/laser/ground_right/pointcloud
/pepper/laser/ground_right/scan
/pepper/laser/shovel/pointcloud
/pepper/laser/shovel/scan
/pepper/laser/srd_front/pointcloud
/pepper/laser/srd_front/scan
/pepper/laser/srd_left/pointcloud
/pepper/laser/srd_left/scan
/pepper/laser/srd_right/pointcloud
/pepper/laser/srd_right/scan
/pepper_navigation/front
/pepper_tweet
/rosout
/rosout_agg
/run_behavior/cancel
/run_behavior/feedback
/run_behavior/goal
/run_behavior/result
/run_behavior/status
/speech
/speech_action/cancel
/speech_action/feedback
/speech_action/goal
/speech_action/result
/speech_action/status
/speech_vocabulary_action/cancel
/speech_vocabulary_action/feedback
/speech_vocabulary_action/goal
/speech_vocabulary_action/result
/speech_vocabulary_action/status
/tf
/tweet
/word_recognized
```

kochigami, 2015-06-03:
rosnode listは
```
/pose_controller
/pose_manager
```
rostopic listは
```
/body_pose/cancel
/body_pose/feedback
/body_pose/goal
/body_pose/result
/body_pose/status
/body_pose_naoqi/cancel
/body_pose_naoqi/feedback
/body_pose_naoqi/goal
/body_pose_naoqi/result
/body_pose_naoqi/status

(/joint_angles)　両方ある
/joint_angles_action/cancel
/joint_angles_action/feedback
/joint_angles_action/goal
/joint_angles_action/result
/joint_angles_action/status

(/joint_stiffness) 両方ある
/joint_stiffness_trajectory/cancel
/joint_stiffness_trajectory/feedback
/joint_stiffness_trajectory/goal
/joint_stiffness_trajectory/result
/joint_stiffness_trajectory/status
/joint_trajectory/cancel
/joint_trajectory/feedback
/joint_trajectory/goal
/joint_trajectory/result
/joint_trajectory/status
```
が増えています。
（nao_pose/pose_manager.launch だけあげて確かめれば良かったです・・・申し訳ございません。）