jsk-ros-pkg/jsk_robot #342: naoeusのパッケージについて

kochigami, 2015-06-03:
もうすぐ未来館での見学対応があるのですが、その時にナオも動くといいかな、と思っています。
しかし、
roseus nao-interface.lを読み込んでも

/opt/ros/hydro/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0  0x5fe1fb8 in (apply #'ros::load-org-for-ros ros::fullname args)E:

とエラーが出るようになってしまいました。
naoeusのパッケージに、package.xmlではなくmanifest.xmlがある（パッケージが古い）ことが原因なのでしょうか。それとも、問題はもっと深いのでしょうか。

k-okada, 2015-06-03:
roseus
$ (load "nao-interface.l")

として見て下さい
https://github.com/jsk-ros-pkg/jsk_roseus/issues/314

kochigami, 2015-06-04:
ご教示ありがとうございました。
エラーの内容が読めました。

closed 2015-06-04 by kochigami

k-okada, 2015-06-04:
なんて書いてあった？

kochigami, 2015-06-04:
```
roseus ;; loading roseus("") on euslisp((9.11 host06.storm.ros.org Thu Feb 19 15:39:45 UTC 2015  1.0.3))                                                       
eustf roseus_c_util 1.irteusgl$ load "nao-interface.l"
/opt/ros/hydro/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error:  file #P"/opt/ros/hydro/share/euscollada/nao.l" not found in (apply #'ros::load-org-for-ros ros::fullname args)
```
です。
naoeus/euslisp 以下に、昔作って取っておいてあったnao.l (improved 版)を置いておくだけでは駄目みたいで、jsk_nao_robotのREADMEにあるnao.lの作成を最初からやるとうまくいきました。
generate nao.l では点が出て（ロボットではない）、
improve nao.l ではariesに接続した状態（そこからコレオグラフをダウンロードしてくる→モデルをつくる材料にするようです）であれば、可愛いナオになることが分かりました。