jsk-ros-pkg/jsk_robot #344: error: no such package NAO_INTERACTION_MSGS

jiangjun0105, 2015-06-09:
Today I tried to run the demonstration program in Miraikan on my computer. When I ran 
``load "mirai-demo-20150413.l"``
in roseus, I got this error
[image]

Kochigami tried to help me generate that message, and we did this
[image]

But I got the same error. Additionally, I had rebooted the roseus and  terminal, but nothing changed.

kochigami, 2015-06-09:
Do we need to add nao_interaction_msgs to find_package in CMakeLists.txt of peppereus?
She seems to have get nao_interaction_msgs from apt-get (because ```roscd nao_interaction_msgs``` changes the current directory /opt/ros/indigo/...).
In this case, I think we don't have to ```rosrun roseus generate-all-msg-srv.sh nao_interaction_msgs```.

I looked into package.xml of peppereus, but there was no dependency of naoqi_msgs, so I think we don't have to add nao_interaction_msgs to it. Is this right?

kochigami, 2015-06-09:
We found jsk_201504_miraikan (pepper demo package) has run dependency on jsk_pepper_startup package, which has dependency on nao_interaction_msgs, so I think we don't have to add extra dependency of nao_interaction_msgs to execute this program...

My PC, which I installed nao_interaction_msgs from source, can execute demo program without no error.
I don't know why this error happened.

jiangjun0105, 2015-06-09:
I added the ``  <run_depend>nao_interaction_msgs</run_depend>`` to the package.xml of package``jsk_pepper_startup`` and ``peppereus``
But it didn't work. Nothing changed.

jiangjun0105, 2015-06-12:
https://github.com/jsk-ros-pkg/jsk_robot/pull/347 will fix
add nao_interactoin_msgs to both build_depend and run_depend, then run ` rosrun roseus generate-all-msg-srv.sh peppereus` to re-generate `/home/jiangjun/catkin_ws/devel/share/roseus/ros/peppereus/manifest.l` 
manifest.l should contain `(ros::load-ros-package "nao_interaction_msgs")
`

jiangjun0105, 2015-06-12:
until jskeus 1.0.7 (which contains https://github.com/euslisp/jskeus/commit/94fc4f027dc17121c9d483aad68cbdda72fc1868) available, please compile euslisp from source or add 
```
(defmethod pepper-robot
  ;;;;;;;;;;;;;;;;;;;;;;;;;;;;; FIX                                                                                                                                      
  (:make-support-polygons
   ()
   (setq support-polygons
         (remove nil (list
          (send self :make-sole-polygon :rleg)
          (send self :make-sole-polygon :lleg))))
   )
  (:make-sole-polygon
   (name)
   (labels ((all-descendant-links
             (l)
             (if (and l (send l :descendants))
                 (append (list l) (flatten (mapcar #'all-descendant-links (remove-if-not #'(lambda (x) (derivedp x bodyset-link)) (send l :descendants)))))
               )))
     (unless (send self name :links) (return-from :make-sole-polygon nil))
     (send-all (all-descendant-links (car (send self name :links))) :worldcoords)
     (let* ((target-nm (read-from-string (format nil "~A-sole-body" name)))
            (vs (remove-duplicates
                 (flatten (send-all (flatten (send-all (all-descendant-links (car (send self name :links))) :bodies)) :vertices))
                 :test #'(lambda (x y) (eps-v= x y *epsilon*))))
            (min-vs (find-extream vs #'(lambda (x) (elt x 2)) #'<))
            (b (send
                (make-bounding-box
                 (remove-if
                  #'(lambda (p)
                      (< 5.0 (- (elt p 2) (elt min-vs 2)))) vs)) ;; 5.0 is margin                                                                                        
                :body)))
       (send (send self name :end-coords :parent) :assoc b)
       (send self :put target-nm b)
       (send b :worldcoords)
       (let ((f (find-if #'(lambda (x) (memq :bottom (send x :id))) (send b :faces))))
         (send f :name name)
         f))))
  ;;;;;;;;;;;;;;;;;;;;;;;;;;;;; FIX     
)
```
to your pepper.l file.

furushchev, 2015-12-16:
This may be just a problem on building.
I think it's ok to close once. @k-okada

closed 2015-12-16 by k-okada