jsk-ros-pkg/jsk_robot #349: unable to create nao.l on indigo

kochigami, 2015-06-14:
This is just a memo. (何か勘違いをしていたら申し訳ございません)
I tried to create nao.l in my indigo environment by following README in jsk_nao_robot, but it failed.
Maybe I need to install collada_urdf_jsk_patch, but I can't get a released package and complete ```catkin build collada_urdf_jsk_patch``` (actually, build failed in assimp_devel).
I have to wait until released package like ```ros-indigo-collada-urdf-jsk-patch``` is released.

First I executed  ```sudo apt-get install ros-indigo-nao-description```, then ```roscd euscollada``` and ```./nao.sh```
The result is as follows.
```
./nao.sh
DEPRECATED IN HYDRO:
  The <include> tag should be prepended with 'xacro' if that is the intended use
  of it, such as <xacro:include ...>. Use the following script to fix incorrect
  xacro includes:
     sed -i 's/<include/<xacro:include/g' `find . -iname *.xacro`
DEPRECATED IN HYDRO:
  The <include> tag should be prepended with 'xacro' if that is the intended use
  of it, such as <xacro:include ...>. Use the following script to fix incorrect
  xacro includes:
     sed -i 's/<include/<xacro:include/g' `find . -iname *.xacro`

[rospack] Error: package 'collada_urdf_jsk_patch' not found
```
Next, I executed 
```aptitude search ros-indigo-collada-urdf-jsk-patch```, but it returner none,
so I cloned jsk_common ```git clone https://github.com/jsk-ros-pkg/jsk_common.git```
However, catkin build collada_urdf_jsk_patch failed.
```
kochigami@kochigami-desktop:~/catkin_ws/src/jsk_common/jsk_ros_patch$ catkin build collada_urdf_jsk_patch 
--------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/indigo
Workspace:                   /home/kochigami/catkin_ws
Source Space:       [exists] /home/kochigami/catkin_ws/src
Build Space:        [exists] /home/kochigami/catkin_ws/build
Devel Space:        [exists] /home/kochigami/catkin_ws/devel
Install Space:     [missing] /home/kochigami/catkin_ws/install
DESTDIR:                     None
--------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
--------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
--------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------------- 
Found '51' packages in 0.1 seconds. 
Starting ==> assimp_devel                                                       

[assimp_devel] ==> '/home/kochigami/catkin_ws/build/assimp_devel/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' in '/home/kochigami/catkin_ws/build/assimp_devel'
[100%] Generating /home/kochigami/catkin_ws/devel/lib/libassimp_devel.so
make[3]: : jobserver : -j1 ,  make  `+' 
[../v3.1.1.zip]
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of ../v3.1.1.zip or
        ../v3.1.1.zip.zip, and cannot find ../v3.1.1.zip.ZIP, period.
make[3]: *** [build/assimp-3.1.1/unpacked]  9
make[2]: *** [/home/kochigami/catkin_ws/devel/lib/libassimp_devel.so]  2
make[1]: *** [CMakeFiles/libassimp_devel.dir/all]  2
make: *** [all]  2
[assimp_devel] <== '/home/kochigami/catkin_ws/build/assimp_devel/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' failed with return code '2'

Failed   <== assimp_devel           [ 0.2 seconds ]                             
[build] There were '1' errors:                                                  

Failed to build package 'assimp_devel' because the following command:

# Command to reproduce:
cd /home/kochigami/catkin_ws/build/assimp_devel && /home/kochigami/catkin_ws/build/assimp_devel/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j; cd -

# Path to log:
cat /home/kochigami/catkin_ws/build/build_logs/assimp_devel.log

Exited with return code: 2 

[build] Runtime: 0.3 seconds 
```
Actually, the actual error occurred in building assimp_devel.
The error message is below.
```
unzip v3.1.1.zip 
Archive:  v3.1.1.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of v3.1.1.zip or
        v3.1.1.zip.zip, and cannot find v3.1.1.zip.ZIP, period.
```
As this error message, I can't unzip v3.1.1.zip.
I installed assimp_devel from ```sudo apt-get ros-indigo-assimp-devel```.

k-okada, 2015-06-15:
use collada-urdf instad of collada-urdf-jsk-patch

kochigami, 2015-10-17:
I apologize for my late reply.
Since ```ros-indigo-collada-urdf-jsk-patch``` was released, nao.l could be created for the following steps:

```
get source code of euscollada from jsk_model_tools
sudo apt-get install ros-indigo-collada-jsk-urdf-jsk-patch
catkin build
source ~/catkin_ws/devel/setup.bash
roscd euscollada
./nao.sh
```

closed 2015-10-17 by kochigami