jsk-ros-pkg/jsk_robot #352: Diff of environment variable ROBOT between inside of pr2 and rosserpr1012

wkentaro, 2015-06-18:
in the shell of inside of PR2 ROBOT env val is set like,

~~~
ROBOT=pr2
~~~

but in rossetpr1012 ans rossetpr1040, it is set like,
~~~
ROBOT=PR2
~~~
See https://github.com/jsk-ros-pkg/jsk_robot

garaemon, 2015-06-18:
I thinik it should be `ROBOT=pr2`

wkentaro, 2015-06-18:
PR: https://github.com/jsk-ros-pkg/jsk_robot/pull/353

k-okada, 2015-06-19:
I dont think pr2 robot is depend on ROBOT environment, please specify which code depends on this env variable

wkentaro, 2015-06-19:
pr2

   - pr2_navigation_global
   - pr2_navigation_slam
   - pr2_2dnav_local
   - pr2_2dnav
   - pr2_2dnav_slam

jsk-ros-pkg

   - pr2eus_tutorials
   - jsk_pr2_sensors
   - jsk_pr2_move_base
   - detect_cans_in_fridge_201202

jsk-ros-pkg/jsk_pr2eus/pr2eus_tutorials/launch/template_grasp.launch:
<include file="$(find pr2_machine)/$(env ROBOT).machine" />
jsk-ros-pkg/jsk_robot/jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_sensors/pr2_teleop_robot.launch:
 <include file="$(find pr2_machine)/$(env ROBOT).machine" />
jsk-ros-pkg/jsk_robot/jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_sensors/kinect_head.launch:
 <include file="$(find pr2_machine)/$(env ROBOT).machine" />
jsk-ros-pkg/jsk_robot/jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_move_base/move_base.xml:
 <include file="$(find pr2_machine)/$(env ROBOT).machine" />
jsk-ros-pkg/jsk_robot/jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_move_base/safe_teleop.xml:
 <include file="$(find pr2_machine)/$(env ROBOT).machine" />
jsk-ros-pkg/jsk_robot/jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_move_base/pr2_2dnav.launch:
 <include file="$(find pr2_machine)/$(env ROBOT).machine" />
jsk-ros-pkg/jsk_robot/jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_move_base/pr2_teleop_general_joystick.launch:
 <include file="$(find pr2_machine)/$(env ROBOT).machine" />
jsk-ros-pkg/jsk_demos/detect_cans_in_fridge_201202/launch/perception.launch:
 <include file="$(find pr2_machine)/$(env ROBOT).machine" />
jsk-ros-pkg/jsk_demos/detect_cans_in_fridge_201202/launch/startup.launch:
 <include file="$(find pr2_machine)/$(env ROBOT).machine" />
jsk-ros-pkg/jsk_demos/detect_cans_in_fridge_201202/index.rst:
<include file="$(find pr2_machine)/$(env ROBOT).machine" />
pr2_navigation/pr2_navigation_global/move_base.xml:  <include
file="$(find pr2_machine)/$(env ROBOT).machine" />
pr2_navigation/pr2_navigation_slam/move_base.xml:  <include
file="$(find pr2_machine)/$(env ROBOT).machine" />
pr2_navigation/pr2_navigation_local/move_base_local.xml:  <include
file="$(find pr2_machine)/$(env ROBOT).machine" />
pr2_navigation_apps/pr2_2dnav_local/launch/pr2_2dnav.launch:  <include
file="$(find pr2_machine)/$(env ROBOT).machine" />
pr2_navigation_apps/pr2_2dnav/launch/pr2_2dnav.launch:  <include
file="$(find pr2_machine)/$(env ROBOT).machine" />
pr2_navigation_apps/pr2_2dnav_slam/launch/pr2_2dnav.launch:  <include
file="$(find pr2_machine)/$(env ROBOT).machine" />

​

k-okada, 2015-06-19:
pr2_machines only have pr2.machine and sim.machine and how non-jsk package
change this? do the also uses ROBOT environment?

wkentaro, 2015-06-19:
It seems that `ROBOT` is used in navigation files of pr2. (like pr2_navigation, pr2_move_base)
I don't think they changes `ROBOT` environment in its program but just use it to decide which should be used (pr2.machine or sim.machine).

k-okada, 2015-06-20:
humm ->

https://github.com/PR2/pr2_navigation/blob/86fa20c703bc349b24bb18001c48ac87d18bb7e6/pr2_navigation_slam/move_base.xml#L2