jsk-ros-pkg/jsk_robot #359: `roslaunch jsk_pr2_startup pr2_hark.launch` failed with segmentation fault

furushchev, 2015-06-24:
below is log when executed `roslaunch --screen jsk_pr2_startup pr2_hark.launch` on PR1012:

```bash
UINodeRepository::Scan()
Scanning def /usr/lib/flowdesigner/toolbox
done loading def files
loading XML document from memory
node_LocalizeMUSIC_1 : param ELEVATION no longer used
node_GHDSS_1 : param TF_CONJ no longer used
node_GHDSS_1 : param MIC_FILENAME no longer used
node_GHDSS_1 : param MIC_POS_SHIFT no longer used
node_GHDSS_1 : param FIXED_NOISE no longer used
node_GHDSS_1 : param FIXED_NOISE_FILENAME no longer used
node_GHDSS_1 : param SPEED_OF_SOUND no longer used
node_GHDSS_1 : param UPDATE_ACCEPT_ANGLE no longer used
done!
Building network  :MAIN
whatisthis: transfer function
[E] [zip_open failed] [harkio_TransferFunction_fromFile] [153]
Segmentation fault
```

on PR1040:

```bash
Toolbox load error: libharkio3.so: cannot open shared object file: No such file or directory
Toolbox load error: libharkio3.so: cannot open shared object file: No such file or directory
There were errors loading the toolboxes:
/usr/lib/flowdesigner/toolbox/hark/hark.tlb
UINodeRepository::Scan()
Scanning def /usr/lib/flowdesigner/toolbox
done loading def files
loading XML document from memory
node_LocalizeMUSIC_1 : param ELEVATION no longer used
node_GHDSS_1 : param TF_CONJ no longer used
node_GHDSS_1 : param MIC_FILENAME no longer used
node_GHDSS_1 : param MIC_POS_SHIFT no longer used
node_GHDSS_1 : param FIXED_NOISE no longer used
node_GHDSS_1 : param FIXED_NOISE_FILENAME no longer used
node_GHDSS_1 : param SPEED_OF_SOUND no longer used
node_GHDSS_1 : param UPDATE_ACCEPT_ANGLE no longer used
done!
Building network  :MAIN
UINode.cc line 221: Node not found: MultiFFT
UINode.cc line 232: Exception caught while creating node_MultiFFT_1 (type MultiFFT)
UINetwork.cc line 559: Exception caught while building network LOOP0
UINode.cc line 232: Exception caught while creating node_LOOP0_1 (type LOOP0)
UINetwork.cc line 559: Exception caught while building network MAIN

[pr2_hark-1] process has died [pid 15095, exit code 1, cmd /home/furushchev/ros/hydro/src/jsk-ros-pkg/jsk_robot/jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_lifelog/microcone_localization_MFCC.sh __name:=pr2_hark __log:=/home/furushchev/.ros/log/97892330-19a9-11e5-8bf5-001517ebc305/pr2_hark-1.log].
log file: /home/furushchev/.ros/log/97892330-19a9-11e5-8bf5-001517ebc305/pr2_hark-1*.log
```

ompugao, 2015-07-01:
basically, the code I wrote heavily depends on **hark1.0**
but hark developers are updating hark actively these days and the the latest version is **hark2.1**.
not only the current code but the installation manual(https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/master/hark_jsk_plugins/INSTALL ) is now quite obsolete and needs review.

furushchev, 2015-07-02:
@ompugao Thanks for quite looking this issue attentively!
OK, I'll try setup on latest hark environment.
Could you give me the setup procedure to work pr2_hark.launch on the old enviroment ( how to, document, link to related blog, or some else) as possible as you remember?

ompugao, 2015-07-12:
please follow http://www.hark.jp/wiki.cgi?page=HARK+Installation+Instructions
you can do it in the same way even with the old hark.
you can download the old code from [here](http://www.hark.jp/wiki.cgi?page=Softwares#p10)
i recommend you to install from *source code* if you want to debug your code, otherwise it's fine to use their pre-compiled binaries (http://www.hark.jp/wiki.cgi?page=HARK+Installation+Instructions#p1).

and as i mentioned, you can install our plugin following: https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/master/hark_jsk_plugins/INSTALL

furushchev, 2015-07-12:
Thank you very much!!I'll try it tomorrow!

Outlook から送信されました

furushchev, 2015-09-28:
@ompugao Thank you for kind advice!
The main reason of segmentation fault was change of data structure from hark version <= 2.0 to > 2.1.
It works again with PR #450

closed 2015-09-28 by furushchev