jsk-ros-pkg/jsk_robot #360: Failed in ```roslaunch jsk_pepper_startup jsk_pepper_startup.launch```

kochigami, 2015-06-29:
Could you help us when we want to delete unnecessary nodes, which wouldn't be deleted even though we executed ```rosnode kill <node name>```? 

First, we found some topics have already existed before we launch nothing on Pepper.
```
/bumper
/cmd_pose
/cmd_vel
/diagnostics
/dummy_state
/joint_angles
/joint_states
/joint_stiffness_trajectory/cancel
/joint_stiffness_trajectory/feedback
/joint_stiffness_trajectory/goal
/joint_stiffness_trajectory/result
/joint_stiffness_trajectory/status
/joint_trajectory/cancel
/joint_trajectory/feedback
/joint_trajectory/goal
/joint_trajectory/result
/joint_trajectory/status
/robot_interface_marker_array
/rosout
/rosout_agg
/speech
/tactile_touch
/tf
/tf_static
```
In this situation, we got some errors when we executed one launch file for one node (for example) ```roslaunch nao_apps tactile.launch ```.
```
kochigami@kochigami-ThinkPad-T440p:~/ros/hydro/src/naoqi_bridge/naoqi_msgs [add-HandTouch.msg]$ roslaunch nao_apps tacticle.launch 
... logging to /home/kochigami/.ros/log/eaaac870-19a9-11e5-8c68-c03fd5ac4304/roslaunch-kochigami-ThinkPad-T440p-8370.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://133.11.216.151:59226/

SUMMARY
========

PARAMETERS
 * /rosdistro
 * /rosversion

NODES
  /
    nao_tactile (nao_apps/nao_tactile.py)

ROS_MASTER_URI=http://pepper.jsk.imi.i.u-tokyo.ac.jp:11311

core service [/rosout] found
process[nao_tactile-1]: started with pid [8379]
Exception TypeError: "in method 'module_getName', argument 1 of type 'AL::module *'" in <bound method NaoTactile.__del__ of <__main__.NaoTactile;  >> ignored
================================================================================REQUIRED process [nao_tactile-1] has died!
process has died [pid 8379, exit code 1, cmd /home/kochigami/ros/hydro/src/nao_robot/nao_apps/nodes/nao_tactile.py --pip=olive.jsk.imi.i.u-tokyo.ac.jp --pport=9559 __name:=nao_tactile __log:=/home/kochigami/.ros/log/eaaac870-19a9-11e5-8c68-c03fd5ac4304/nao_tactile-1.log].
log file: /home/kochigami/.ros/log/eaaac870-19a9-11e5-8c68-c03fd5ac4304/nao_tactile-1*.log
Initiating shutdown!
================================================================================
[nao_tactile-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
The content of  log file in ```~/.ros/log/...``` is as follows.
```
inaoqi-broker: construct at :0, parent : olive.jsk.imi.i.u-tokyo.ac.jp:9559
[I] 8556 qi.eventloop: Creating event loop while no qi::Application() is running
[I] 8556 qimessaging.session: Session listener created on tcp://0.0.0.0:0
[I] 8556 qimessaging.transportserver: TransportServer will listen on: tcp://133.11.216.151:57007
[I] 8556 qimessaging.transportserver: TransportServer will listen on: tcp://127.0.0.1:57007
[I] 8556 qimessaging.transportserver: TransportServer will listen on: tcp://157.82.7.39:57007
[W] 8570 qimessaging.transportsocket: connect: Operation canceled
Could not connect to NaoQi's main broker
```
The result of ```rosnode list``` is as follows. 
rosnode list
/follow_hand_pulled_1435246029578825762
/follow_hand_pulled_1435246873184638926
/follow_hand_pulled_1435294065294784081
/follow_hand_pulled_1435296387535976156
/follow_hand_pulled_1435302718024993440
/nao_diagnostic_updater
/nao_tactile
/pepper_1435152016604059735
/rosout
/rostopic_11126_1435065410635
/rostopic_8007_1435561499308
```
However, we can't delete them by executing ```rosnode kill <node name>```.
Thanks in advance.

k-okada, 2015-06-29:
1. try `rosnode cleanup` to clean zombie process
2.`ps -auxw | grep roscore` to find othre roscore process is running and
kill them
3. reboot computer

kochigami, 2015-07-19:
Thank you very much for your help. ```rosnode cleanup``` works well.
I'll try ```rosnode cleanup``` every times before we start our demonstration.

closed 2015-07-19 by kochigami