jsk-ros-pkg/jsk_robot #363: How to stop Pepper's go-pos method after it starts

kochigami, 2015-07-03:
I am grateful if Pepper stops safely when he or she subscribes some rostopic related to obstacle detecting signal 
even though he or she is moving by using ```:go-pos``` method in ```pepper-interface.l```

I think Pepper's ```:go-pos``` method is like PR2's ```:go-pos-unsafe``` method. 

Could you give me some hints to tackle this problem?
Finding go-pos-safe method from NAOqi and integrated it into pepper-interface.l is required?

chiwunau, 2015-07-04:
In PR2, we have something like 
(send *ri* :go-stop)
does that work on Pepper?

kochigami, 2015-07-04:
Thank you very much for your advice.
In PR2, you can do something like that?

'''
(defun calback(msg)
  (send ri :go-stop))
(subscribe /some_sensor_topic  msg_type callback)
and in main code, it includes
(send ri :go-pos some_distance)
'''
The robot subscribes topic data by using (ros::ros-spin).
Does PR2 stop when he or she moves if something like this code is used?

Sorry for not clear explanation…
Anyway, thank you very much!!

chiwunau, 2015-07-04:
(send *ri* :go-stop) also stops the motion of joint trajectory command.
how about (send *ri* :go-pos 0 0 0)

k-okada, 2015-07-04:
go-stop only stops base motion
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/pr2-interface.l#L328

kochigami, 2015-07-05:
Thank you very much for all of your advice.

I tried on something like
```(defun bumper-cb (msg)                                                 
  (send *ri* :go-pos 0 0 0))
(ros::subscribe "/bumper"  naoqi_msgs::Bumper #'bumper-cb)
```
However, it seems to fail.
I cannot understand whether the situation is ```(send *ri* :go-pos 0 0 0)``` is excuted but it takes some time to stop Pepper completely or ```(send *ri* :go-pos 0 0 0)``` is no effect.
I think the former is answer.
I will continue considering this tomorrow.

k-okada, 2015-07-06:
1, please check which naoqi API is called in ros_driver, if you use
https://github.com/ros-naoqi/naoqi_bridge/blob/master/naoqi_driver/src/naoqi_driver/naoqi_moveto.py#L55
it calls moveto described here
http://doc.aldebaran.com/2-1/naoqi/motion/control-walk-api.html#ALMotionProxy::move__floatCR.floatCR.floatCR
2. please check what will happen if
proxy.moveTo(1,0,0)
time.sleep(1)
proxy.moveTo(-1,0,0)

will the robot moves 1 meter ahead and then go back back 1 meter. or the
robot move forward 1 sec and back to 1 meter
it it moves as the former description, we should change naoqi_driver (and
better to ask question that how to interrupt moveto motion)

3) create naoqi_movetoward, that takes cmd_vel and send moveToward, note
that the robot should be stopped if there are no /cmd_vel input
for 1 sec,
https://github.com/tork-a/spur/blob/master/spur_controller/nodes/base_controller.py#L185
may help

kochigami, 2015-07-06:
Thank you very much for your advice.

We found moveTo is a block function which is not interrupted.
As to question2, Pepper moved as the former description.

By the way, I found :go-velocity method in pepper-interface.l. 
I expect this method will relate to moveToward or something, and when I go back to school, I want to examine it.

Now I am wondering how pepper-interface.l and NAOqi API are combined.
(ex Which function in NAOqi API does correspond with which method in pepper-interface.l?)

kochigami, 2015-07-06:
I tried a simple code written rospy in which Pepper continues to move by using ```moveTo``` or ```moveToward``` and is interrupted when bumper detects an obstacle. (I forgot euslisp at that time.)
If the bumper sensor is activated, Pepper stops for 1sec and starts moving again.

As a result, I seem to have to write naoqi_movetoward code.

There are results of Pepper' s movement after I executed the simple code.
1. condition of using moveTo 
Pepper stops only once and never starts moving again. 
(I think it is caused by the interruption of moveTo.)

2. condition of using moveToward  
Pepper stops for 1 sec and starts moving.

Thank you very much for your advice.

closed 2015-07-06 by kochigami

k-okada, 2015-07-07:
from
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_pepper_robot/peppereus/pepper-interface.l#L102
go-pos uses cmd_pose and go-velocity uses cmd_vel

and it seems nao_wheel.py listens cmd_vel, and cmd_pose, even though
nao_wheel.py is not launched pepper.launch, it seems we launch manually
`roslaunch nao_apps nao_wheel.launch`

then looking in to nao_walker.py and it uses setWalkTargetVelocity which
 may not work on Pepper robot
https://github.com/ros-naoqi/nao_robot/blob/aa4e1e9fa6277fac31c5ea9a568067f908d7fa9b/nao_apps/nodes/nao_walker.py#L138

kochigami, 2015-07-07:
I found ```:go-velocity``` works well (even though nao_wheel.py needs some changes to check in every 1 sec ).
```:go-velocity``` calls moveToward and therefore Pepper can stop by using this.
```
pepper-interface.l         topic          nao_wheel.py
:go-pos ->               /cmd_pose ->  moveTo
:go-velocity ->          /cmd_vel    -> moveToward 
```