jsk-ros-pkg/jsk_robot #366: pr1012,pr1040のapplicationsユーザでデモを動かせるようにする

furushchev, 2015-07-07:
#### 条件
  - deb + jsk-ros-pkg/jsk_demosで動くのが理想的
  - pr1012はkinect_head名前空間, pr1040はopenni名前空間

#### 状況
以下のものは問題があるのでソースから入れています。
- [ ] `robot start`
  - [x] jsk-ros-pkg/jsk_smart_apps
    - [x] not released
  - [x] PR2/app_manager
    - [x] `app_master` node not installed -> https://github.com/PR2/app_manager/pull/2
    - [x] python scripts not installed -> https://github.com/PR2/app_manager/pull/3
    - [ ] release待ち (see. https://gist.github.com/furushchev/0580d2adf129d163b05b )
  - [ ] PR2/pr2_navigation
    - [x] `self_filter` node not installed -> https://github.com/PR2/pr2_navigation/issues/18
    - [ ] https://github.com/PR2/pr2_navigation/issues/18 not released -> https://github.com/PR2/pr2_navigation/issues/19 released only `shadow_fixed`?
- [ ] `roslaunch jsk_pr2_startup pr2.launch`
  - [x] pr2/pr2_apps
    - [x] `pr2_mannequin_mode` yaml files installed to incorrect dest -> https://github.com/PR2/pr2_apps/pull/17
    - [ ] wait to release
  - [ ] jsk-ros-pkg/jsk_topic_tools
    - [ ] `config` not installed -> https://github.com/jsk-ros-pkg/jsk_common/pull/1028
    - [ ] wait to release
  - [ ] SharedAutonomyToolkit/shared_autonomy_manipulation
    - [ ] `safe_teleop_base` not released -> https://github.com/SharedAutonomyToolkit/shared_autonomy_manipulation/issues/5

k-okada, 2015-07-07:
>          -  app_master node not installed -> PR2/app_manager#2
>          -  python scripts not installed -> PR2/app_manager#3
>          -  self_filter node not installed -> PR2/pr2_navigation#18
>          -  config not installed -> jsk-ros-pkg/jsk_common#1028

mongodb_store 関係のエラーはどうなったかな？

garaemon, 2015-07-07:
jsk_commonはtagを切っておきました

furushchev, 2015-07-07:
@k-okada 
> mongodb_store 関係のエラーはどうなったかな？
今はpr1040が空いていないので、pr1012で作業していますが、こちらだと問題にならないです。
そういえば前にmongodプロセスに特殊な許可を与えないと複数ユーザで同じdatabase pathを使うmongodbを立ち上げられないことを思い出したので、それを試してみます。

furushchev, 2015-07-07:
pr1040でもmongodbが動きました。

```bash
sudo killall mongod
sudo service mongodb stop
sudo chown -R robot:users /path/to/db
sudo chmod 777 /path/to/db # http://stackoverflow.com/questions/30814288/cannot-start-mongodb-service
sudo setcap cap_fowner+ep /usr/bin/mongod # https://jira.mongodb.org/browse/SERVER-7583
```

k-okada, 2015-07-07:
nice!, where is /path/to/db in pr2 robot

furushchev, 2015-07-08:
@k-okada for PR2, it is /var/lib/robot/mongodb_store
(see. https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_pr2_robot/jsk_pr2_startup/pr2_bringup.launch#L165 )

furushchev, 2015-07-21:
mongodbが新しく作ってしまうファイルの権限の問題があるようです。
https://github.com/strands-project/mongodb_store/pull/139
でLinuxデフォルトのmongodも使えるようなPRをだしました。

furushchev, 2015-08-04:
pr1012でもデモができてきていて、openni名前空間を留めておく意味があまりないので、pr1040も標準名前空間に戻しました。

furushchev, 2015-12-16:
hydroはEOLなのでapplicationsユーザでデモが出来るように.rosinstallでバージョンを固定しています。

closed 2015-12-16 by furushchev