jsk-ros-pkg/jsk_robot #369: naoqi のｍｓｇを更新するとき、msgファイルを更新する他に何が必要？

jiangjun0105, 2015-07-15:
naoqi_msgs の中のBumper.msgに後ろのバンパーの情報も入れたいです、Bumper.msgのファイルを更新して、そのメッセージを使っているlaunch file を立ち上げる時、以下のエラーが出てきた。
[image]
今知りたいのは、msgを更新するとき、その　.ｍｓｇ　を更新する他に、コンパイルとかなんかやらないといけない事はありますか。
使う言語はpythonです。
よろしくお願いします。

kochigami, 2015-07-15:
naoqi_msgsをcatkin make してみてくれますか。
気づくのが遅くなってごめんなさい！

jiangjun0105, 2015-07-15:
やってみましたが、
```
jiangjun@jiangjun-ThinkPad-T420s:~/catkin_ws$ catkin_make --only-pkg-with-deps naoqi_msgs
Base path: /home/jiangjun/catkin_ws
Source space: /home/jiangjun/catkin_ws/src
Build space: /home/jiangjun/catkin_ws/build
Devel space: /home/jiangjun/catkin_ws/devel
Install space: /home/jiangjun/catkin_ws/install
Packages "naoqi_msgs" not found in the workspace
```
どうやって違うworkspaceのpackageをcatkin_make?

kochigami, 2015-07-15:
naoqi_msgsがopt以下にあるということだよね…

ソースコードをとってくる以外に方法はないのか、詳しい方がいらっしゃればご教示ください。
（役に立たずごめんなさい。）

garaemon, 2015-07-15:
nao_appsをソースで使うためには、naoqi_msgsをソースコードを取ってくるしかないと思います.

jiangjun0105, 2015-07-15:
ごめん、ソースコードをてってくるという方法は具体的にどうやってするかは教えてもらえませんか
@kochigami

kochigami, 2015-07-15:
授業でした。ごめんなさい。
もう解決していたらごめんなさい。

Githubのnaoqi_msgsのページからgit cloneしてください。

jiangjun0105, 2015-07-18:
I solved it by `git clone` the source to catkin workspace.
Thank you for you help!

closed 2015-07-18 by jiangjun0105