jsk-ros-pkg/jsk_robot #381: 冷蔵庫デモで缶を取った後に手渡しする動作が動かない

furushchev, 2015-07-22:
on pr1012 applications environment

```lisp 
[ INFO] [1437538837.007874540]: wait-interpolation debug: start
[ INFO] [1437538843.388988552]: wait-interpolation debug: end
[ WARN] [1437538845.125728626]: continuous joint (l_wrist_roll_joint) moves -251.0 degree, commanded joint differs from original trajectory to avoid unintentional 360 rotation
[ WARN] [1437538845.125896244]: original trajectory command :
[ WARN] [1437538845.125961585]:                        : (#f(12.0 13.0 21.0 62.0 -105.0 -117.0 -66.0 -71.0 -4.0 74.0 -104.0 -89.0 70.0 -5.0 20.0 4.0 40.0) #f(12.0 9.0 24.0 50.0 -94.0 -158.0 -70.0 39.0 -4.0 74.0 -104.0 -89.0 70.0 -5.0 20.0 5.0 30.0)) (1000 1000 1000 1000)
[ WARN] [1437538845.126017110]:   current angle vector : #f(50.0 60.0 74.0 70.0 -120.0 20.0 -30.0 180.0 -60.0 74.0 -70.0 -120.0 -20.0 -30.0 180.0 0.0 3.180555e-15)
[ WARN] [1437538845.126049368]: new trajectory command : dvi = 2
[ WARN] [1437538845.126112644]:                        : #f(31.0 36.5 47.5 66.0 -112.5 -48.5 -48.0 54.5 -32.0 74.0 -87.0 -104.5 25.0 -17.5 100.0 2.0 20.0) 500
[ WARN] [1437538845.126162375]:                        : #f(12.0 13.0 21.0 62.0 -105.0 -117.0 -66.0 -71.0 -4.0 74.0 -104.0 -89.0 70.0 -5.0 20.0 4.0 40.0) 500
[ WARN] [1437538845.126218458]: new trajectory command :
[ WARN] [1437538845.126305551]:                        : (#f(31.0 36.5 47.5 66.0 -112.5 -48.5 -48.0 54.5 -32.0 74.0 -87.0 -104.5 25.0 -17.5 100.0 2.0 20.0) #f(12.0 13.0 21.0 62.0 -105.0 -117.0 -66.0 -71.0 -4.0 74.0 -104.0 -89.0 70.0 -5.0 20.0 4.0 40.0) #f(12.0 9.0 24.0 50.0 -94.0 -158.0 -70.0 39.0 -4.0 74.0 -104.0 -89.0 70.0 -5.0 20.0 5.0 30.0)) (500 500 1000 1000 1000)
;; #<rotational-joint #X8e567c8 r_wrist_flex_joint> :joint-angle(-5.0) violate max-angle(-5.72958)
/opt/ros/hydro/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: vector expected in (v- end start)
```

k-okada, 2015-07-22:
>  span cla
>  ss="pl-c1">74.0 -104.0 -89.0 70.0 -5.0 20.0 5.0 30.0)) (1000 1000 1000 1000)

end と start はなにがはいっていますか？
エラーが出ているのはrobot-interface.l? pr2-interface.l? irtmodel.l ?
そのバージョンは？

furushchev, 2015-07-22:
説明不足でした。
:angle-vector-sequenceの問題なので、原因はrobot-interface.lでした。
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/151

closed 2015-07-22 by furushchev

k-okada, 2015-07-22:
どういう状態でこれが起きますか？
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/test/pr2-ri-test-simple.l#L32
でいろいろチェックしているつもりですが．

furushchev, 2015-07-23:
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L355-L356
ここでavsを走査しているのですが、最後にavsが空になるloopで
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L372
で(car avs) => nilを:angle-vector-durationに渡してしまうので、エラーになると思われます。

k-okada, 2015-07-23:
> pr2-ri-test-simple.lに:angle-vector-sequenceのtestを追加すれば良いでしょうか？

はいついかしましょう．なんで今までこれに引っかかっていないんだろう？

furushchev, 2015-07-23:
@k-okada 
>> pr2-ri-test-simple.lに:angle-vector-sequenceのtestを追加すれば良いでしょうか？

> はいついかしましょう．なんで今までこれに引っかかっていないんだろう？

よくみたら既に`test-angle-vector`などで`:angle-vector-sequence`もテストされているようです。

もっと詳しくテストしてみると, (length avs) より(length tms)が大きくなると`:angle-vector-durationの第2引数にnilが渡ってしまうようです。

```lisp
(load "package://pr2eus/pr2-interface.l")
(ros::roseus "hoge")
(pr2-init)
(setq avs (list
           #f(50.0352 60.0006 73.999 69.9915 -119.999 19.9992 -30.0006 180.001 -59.998\
6 73.998 -70.0037 -119.999 -20.0017 -30.0015 180.003 0.004854 -1.146)
           #f(50.0 60.0 74.0 70.0 -120.0 20.0 -30.0 180.0 -49.6879 33.0369 -90.0555 -1\
21.542 75.5768 -39.6244 94.367 0.0 0.0)))

(send *ri* :angle-vector-sequence avs (list 1000 1000 1000))
```

等でも落ちることが分かりました。

k-okada, 2015-07-23:
1) なんでそういうことになるんだろう？でもコードにどこかバグがあるかな？
２） チェックの仕方はtms とavsの長さを比較するようにしましょう．

furushchev, 2015-07-24:
とりあえず今回のように
`(length avs) < (length tms)`になってしまうときのための修正を
https://github.com/furushchev/jsk_pr2eus/commit/da3fcfe0584943f282de25da3571bcb9ed4d3025
として入れてみました。
今回のエラーが起こった箇所は`jsk_demo_common/euslisp/pr2-action.l`の`:hand-over`なので、そちらも修正します。

furushchev, 2015-07-24:
修正しました。
https://github.com/jsk-ros-pkg/jsk_demos/pull/1003