jsk-ros-pkg/jsk_robot #383: speak-jpができない

furushchev, 2015-07-22:
現在applicationsユーザ@pr1012で

/robotsound_jpにトピックを送ると、

```bash
SIOD ERROR: unbound variable : aq_rm.phont
[ERROR] [WallTime: 1437543186.739402] Sound synthesis failed. Is festival installed? Is a festival voice installed? Try running "rosdep satisfy sound_play|sh". Refer to http://pr.willowgarage.com/wiki/sound_play/Troubleshooting
```

と出ます。
`rosdep satisfy sound_play`というコマンドは既に無く, かわりに`rosdep check sound_play`をしても、All dependencies satisfiedというメッセージが出ます。

k-okada, 2015-07-22:
voice text で喋るはずだけど．

また，他のユーザだとしゃべりますか．

furushchev, 2015-07-24:
**memo** `voice_text` package has to be installed from source because of licence problem.
After installed from source, it works awesome.

closed 2015-07-24 by furushchev