jsk-ros-pkg/jsk_robot #396: pr2_gazebo.launch will die in rgbd_launch launching phase.

aginika, 2015-07-30:
This is because of the difference of indigo and hydro.
In hydro , there was a machine tag and in indigo there is no more machine tag related args...

What is the best way to make  pr2_gazebo  working both in hydro and indigo?

Where the args passing is below.
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_pr2_robot/jsk_pr2_startup/pr2_gazebo.launch#L38

k-okada, 2015-07-30:
does gazebo requires rgdb_launch?

aginika, 2015-08-03:
This rgbd_launch is for publishing `image_rect_color` which is publiing in real but not in gazebo according to ＠furushchev .
So this might be replaced with `relay`, cause image_color is published.

k-okada, 2015-08-09:
Bisides we need to API stability for rgbd_launch
it seems gazebo publishes perfect depth image, so we can use relay to create `image_rect_color`
```
$ rostopic echo -n 1 /head_mount_kinect/depth/camera_info
header: 
  seq: 22
  stamp: 
    secs: 480
    nsecs: 790000000
  frame_id: head_mount_kinect_ir_optical_frame
height: 480
width: 640
distortion_model: plumb_bob
D: [1e-08, 1e-08, 1e-08, 1e-08, 1e-08]
K: [589.3664541825391, 0.0, 320.5, 0.0, 589.3664541825391, 240.5, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [589.3664541825391, 0.0, 320.5, -0.0, 0.0, 589.3664541825391, 240.5, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---
```

furushchev, 2015-08-09:
OK. I'll now create PR to use relay instead of rgbd_launch

k-okada, 2015-08-09:
https://github.com/ros-drivers/rgbd_launch/issues/5 says current hydro API is not correct, so we may have to change jsk's pr2.launch for indigo

furushchev, 2015-08-09:
@k-okada OK. Anyway I created PR to use relay or `image_transport republish` node instead of rgbd_launch.
This will cover more, almost all kinect topics published on real robot, and, I still can not determine if it's just "Placebo" effect or not, will become faster (less use CPU resource?)

aginika, 2015-08-10:
@furushchev Thank you!!
It works well.

closed 2015-08-10 by aginika

k-okada, 2015-08-10:
@aginika
How did you avoid
https://github.com/jsk-ros-pkg/jsk_robot/issues/409#issuecomment-129337271 ?