jsk-ros-pkg/jsk_robot #409: pr2_gazebo.launch fails on indigo

furushchev, 2015-08-07:
please execute
rosdep install jsk_pr2_startup --ignore-src -r -y

execute 
roslaunch jsk_pr2_startup pr2_gazebo_73b2.launch
then open another terminal:
/opt/ros/indigo/lib/move_base/move_base odom:=base_odometry/odom cmd_vel:=navigation/cmd_vel __name:=move_base_node

NOTE:
http://answers.ros.org/question/11076/edit-logger-levels-programmatically-or-from-the-command-line/

Cc: @sasabot

sasabot, 2015-08-07:
/opt/ros/indigo/lib/move_base/move_base odom:=base_odometry/odom cmd_vel:=navigation/cmd_vel __name:=move_base_node
[ WARN] [1438950592.974247250, 29.270000000]: Waiting on transform from base_footprint to map to become available before running costmap, tf error: Could not find a connection between 'map' and 'base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 29.27 timeout was 0.1.
[ INFO] [1438950593.163195488, 29.360000000]: Using plugin "footprint"
[ INFO] [1438950593.266436398, 29.398000000]: Using plugin "inflation"
[ INFO] [1438950593.277218161, 29.401000000]: Using plugin "static_map"
[ INFO] [1438950593.282266375, 29.403000000]: Requesting the map...
[ INFO] [1438950593.506232290, 29.507000000]: Resizing costmap to 1200 X 1200 at 0.050000 m/pix
[ INFO] [1438950593.701750886, 29.606000000]: Received a 1200 X 1200 map at 0.050000 m/pix
[ INFO] [1438950593.937352114, 29.707000000]: Using plugin "inflation"
[ INFO] [1438950594.052055121, 29.750000000]: Using plugin "obstacles"
[ INFO] [1438950594.055983053, 29.751000000]:     Subscribed to Topics: base_scan_marking base_scan tilt_scan ground_object_cloud
[ INFO] [1438950594.189358192, 29.802000000]: Using plugin "people"
terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
  what():  According to the loaded plugin descriptions the class social_navigation_layers::ProxemicLayer with base class type costmap_2d::Layer does not exist. Declared types are  costmap_2d::FootprintLayer costmap_2d::InflationLayer costmap_2d::ObstacleLayer costmap_2d::StaticLayer costmap_2d::VoxelLayer costmap_2d::VoxelWithFootprintLayer
Aborted (core dumped)

k-okada, 2015-08-07:
http://wiki.ros.org/social_navigation_layers seems interesting

install these packages from deb, also we'd like to figure out why they are
not installed,
Should we add depend to jsk_pr2_xx ? or move-base or pr2 package should
dpend on social navigation?

furushchev, 2015-08-07:
@k-okada OK. I'll add.
@sasabot please note https://github.com/jsk-ros-pkg/jsk_robot/pull/365. if you have time, please notify us the slow nodes. we will debug. :-)

k-okada, 2015-08-08:
merged #412, @sasabot now you can go on

when you paste your output,  you'd better to use `cold block` syntax, three back-quote
https://help.github.com/articles/github-flavored-markdown/

sasabot, 2015-08-09:
Thanks. Now, I'm trying pr2_gazebo_73b2.launch.

A few things I've notices is that, with indigo+Gazebo5 (actually, the Gazebo related packages have to be Jade branch to do this), the gzclient and gzserver fails sometimes, so I had to set
> respawn="true" 

for the Gazebo related nodes in gazebo_ros.

Another thing, the Real Time Factor is 0.33 at default, and by setting the Physics property in Gazebo to,
>  real time update rate : 200
  max step size : 0.005

I get Real Time Factor 0.76, which is a bit slow compared to pr2_gazebo.launch which was 0.93
and when I set
>  real time update rate : 266
  max step size : 0.007

I get Real Time Factor about 0.97, but then the PR2 starts sliding (This happens with pr2_gazebo as well)
Maybe changing the Gazebo physics is not really a good solution, so I'll try looking for the slow nodes.

k-okada, 2015-08-09:
running realtime factor on 1.00 depends on your machine performance and  if you run all ros node, then that's very difficult

BTW: I think you need to locally modify some source code to run pr2_gazebo.launch

- use hydro-devel branch for rgdb_launch (https://github.com/jsk-ros-pkg/jsk_robot/issues/396)
- https://github.com/jsk-ros-pkg/jsk_recognition/pull/1043
- https://github.com/jsk-ros-pkg/jsk_smart_apps/pull/56

k-okada, 2015-08-09:
I staill have following errors, how did you fix that?
```
$ roslaunch jsk_pr2_startup pr2_gazebo.launch
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Metapackage "jsk_roseus" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
... logging to /home/k-okada/.ros/log/6ab2962a-3e7a-11e5-a43e-28b2bde75e69/roslaunch-kokada-t440s-24995.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Metapackage "jsk_roseus" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Metapackage "jsk_roseus" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Metapackage "jsk_roseus" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Metapackage "jsk_roseus" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Metapackage "jsk_roseus" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Metapackage "jsk_roseus" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Metapackage "jsk_roseus" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Metapackage "jsk_roseus" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Metapackage "jsk_roseus" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Metapackage "jsk_roseus" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Metapackage "jsk_roseus" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Metapackage "jsk_roseus" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Metapackage "jsk_roseus" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Metapackage "jsk_roseus" should not have other dependencies besides a buildtool_depend on catkin and run_depends.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
[... cut ...]
[tilt_shadow_filter-67] restarting process
process[tilt_shadow_filter-67]: started with pid [6240]
[ERROR] [1439113471.494651895, 76.680000000]: Lookup would require extrapolation into the past.  Requested time 76.413000000 but the earliest data is at time 76.581000000, when looking up transform from frame [base_footprint] to frame [base_link]
[ERROR] [1439113472.405922588, 76.885000000]: Lookup would require extrapolation into the past.  Requested time 76.463000000 but the earliest data is at time 76.58

sasabot, 2015-08-09:
I forgot to mention but I modified pr2_gazebo.launch by commenting out ```<arg name="machine" value="c1" />``` to launch pr2_gazebo_73b2.launch.

I haven't yet got rid of all the errors, some of the critical ones I'm having is 
```
[amcl-43] process has died
```
and I still have
```
[ERROR] [1439112250.398724708]: Skipping XML Document "/home/laala/ros/indigo/src/ros_naoqi/naoqi_bridge/naoqi_sensors_py/naoqicamera_nodelet.xml" which had no Root Element.  This likely means the XML is malformed or missing.
```
And for pr2_gazebo_73b2.launch, I have
```
[tilt_shadow_filter-66] process has died
[base_shadow_filter-69] process has died
```

Furthermore, when I try to use ```(send *ri* :go-pos 0.1 0 0)``` I get
```
[ERROR] [1439111028.531715240, 186.784000000]: Could not find a connection betw\
een 'map' and 'base_footprint' because they are not part of the same tree.Tf ha\
s two or more unconnected trees.
```
however, ```go-pos-unsafe``` seems to work fine.

Also, for some reason, I have to ```rosrun rviz rviz``` before launching pr2_gazebo_73b2.launch. Rviz dies if I run it afterwards.

k-okada, 2015-08-09:
@sasabot, did you make it running without  https://github.com/jsk-ros-pkg/jsk_robot/issues/409#issuecomment-129145378

sasabot, 2015-08-10:
I tested without switching to hydro-devel, instead I just commented out the arg "machine."
If switching the branch is recommended, I can try. Would that be better?

k-okada, 2015-08-10:
that's has been fixed via https://github.com/jsk-ros-pkg/jsk_robot/pull/413, please try this.

Also we have
```
[tilt_shadow_filter-66] process has died
[base_shadow_filter-69] process has died
```
and `/map` is not published, did you fixed that?

sasabot, 2015-08-10:
I'm not sure why we're having the filter process errors, sometimes it doesn't happen.
I was able to use ```go-pos``` when it didn't die for pr2_gazebo.launch.
However, I'm not having any luck with pr2_gazebo_73b2.launch. That dies every time I try.
```
$ /opt/ros/indigo/lib/laser_filters/scan_to_cloud_filter_chain scan:=tilt_scan cloud_filtered:=tilt_scan_shadow_filtered __name:=tilt_shadow_filter
[ WARN] [1439188180.931147893, 45.433000000]: Shutdown request received.
[ WARN] [1439188180.931220117, 45.433000000]: Reason given for shutdown: [new node registered with same name]
```

I don't know if this is related to the ```/map``` problem, but I'm also having
```
$ /opt/ros/indigo/lib/amcl/amcl scan:=base_scan __name:=amcl
[ INFO] [1439190196.207977197]: Subscribed to map topic.
[ INFO] [1439190196.350743369, 69.555000000]: Received a 1200 X 1200 map @ 0.050 m/pix

amcl: /tmp/buildd/ros-indigo-amcl-1.12.4-0trusty-20150604-0435/src/amcl/pf/pf_kdtree.c:378: pf_kdtree_cluster: Assertion `node == pf_kdtree_find_node(self, self->root, node->key)' failed.
Aborted (core dumped)
```

aginika, 2015-08-10:
@k-okada I couldn't have any problem around launch to execute `go-pos` ...
I couldn't find any dying nodes.

[image]
pr2_gazebo_go_pos_test_x2

```
$ roslaunch jsk_pr2_startup pr2_gazebo_73b2.launch 
... logging to /home/aginika/.ros/log/c7013aa0-3f68-11e5-970f-001f29030e5b/roslaunch-remi-16946.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/aginika/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://remi:48162/

SUMMARY
========

CLEAR PARAMETERS
 * /depth_image_creator_nodelet/
 * /move_base_node/
 * /pointcloud_screenpoint_nodelet/

PARAMETERS
 * /active_user/launch_user_name: False
 * /amcl/base_frame_id: /base_footprint
 * /amcl/gui_publish_rate: 10.0
 * /amcl/kld_err: 0.05
 * /amcl/kld_z: 0.99
 * /amcl/laser_lambda_short: 0.1
 * /amcl/laser_likelihood_max_dist: 2.0
 * /amcl/laser_max_beams: 30
 * /amcl/laser_model_type: likelihood_field
 * /amcl/laser_sigma_hit: 0.2
 * /amcl/laser_z_hit: 0.5
 * /amcl/laser_z_max: 0.05
 * /amcl/laser_z_rand: 0.5
 * /amcl/laser_z_short: 0.05
 * /amcl/max_particles: 5000
 * /amcl/min_particles: 500
 * /amcl/odom_alpha1: 0.2
 * /amcl/odom_alpha2: 0.2
 * /amcl/odom_alpha3: 0.8
 * /amcl/odom_alpha4: 0.2
 * /amcl/odom_alpha5: 0.1
 * /amcl/odom_frame_id: /odom_combined
 * /amcl/odom_model_type: omni
 * /amcl/recovery_alpha_fast: 0.0
 * /amcl/recovery_alpha_slow: 0.0
 * /amcl/resample_interval: 1
 * /amcl/transform_tolerance: 0.1
 * /amcl/update_min_a: 0.5
 * /amcl/update_min_d: 0.2
 * /amcl/use_map_topic: True
 * /base_controller/bl_caster_l_wheel_joint/d: 0.0
 * /base_controller/bl_caster_l_wheel_joint/i: 0.0
 * /base_controller/bl_caster_l_wheel_joint/i_clamp: 2.0
 * /base_controller/bl_caster_l_wheel_joint/p: 3.0
 * /base_controller/bl_caster_r_wheel_joint/d: 0.0
 * /base_controller/bl_caster_r_wheel_joint/i: 0.0
 * /base_controller/bl_caster_r_wheel_joint/i_clamp: 2.0
 * /base_controller/bl_caster_r_wheel_joint/p: 3.0
 * /base_controller/bl_caster_rotation_joint/position_controller/d: 0.0
 * /base_controller/bl_caster_rotation_joint/position_controller/i: 0.0
 * /base_controller/bl_caster_rotation_joint/position_controller/i_clamp: 3.45
 * /base_controller/bl_caster_rotation_joint/position_controller/p: 80.0
 * /base_controller/bl_caster_rotation_joint/velocity_controller/d: 0.0
 * /base_controller/bl_caster_rotation_joint/velocity_controller/i: 0.0
 * /base_controller/bl_caster_rotation_joint/velocity_controller/i_clamp: 4.0
[... 1552 lines cut ...]
[ INFO] [1439216107.787504546, 259.300000000]: current-map /eng2/7f/ #<cascaded-coords #Xa392828 /world  0.0 0.0 30000.0 / 0.0 0.0 0.0>
[ INFO] [1439216129.544069069, 269.301000000]: current-map /eng2/7f/ #<cascaded-coords #Xa5625b8 /world  0.0 0.0 30000.0 / 0.0 0.0 0.0>
[ INFO] [1439216151.297716611, 279.300000000]: current-map /eng2/7f/ #<cascaded-coords #Xa68bc28 /world  0.0 0.0 30000.0 / 0.0 0.0 0.0>
[ INFO] [1439216173.479202327, 289.300000000]: current-map /eng2/7f/ #<cascaded-coords #Xa7d5f48 /world  0.0 0.0 30000.0 / 0.0 0.0 0.0>
[ INFO] [1439216195.839850320, 299.300000000]: current-map /eng2/7f/ #<cascaded-coords #Xb09e358 /world  0.0 0.0 30000.0 / 0.0 0.0 0.0>
[ INFO] [1439216217.805534197, 309.301000000]: current-map /eng2/7f/ #<cascaded-coords #Xb7bdf20 /world  0.0 0.0 30000.0 / 0.0 0.0 0.0>
[ INFO] [1439216240.187521757, 319.300000000]: current-map /eng2/7f/ #<cascaded-coords #Xb892c08 /world  0.0 0.0 30000.0 / 0.0 0.0 0.0>
[ INFO] [1439216261.168018427, 329.300000000]: current-map /eng2/7f/ #<cascaded-coords #Xb9b9428 /world  0.0 0.0 30000.0 / 0.0 0.0 0.0>
[ INFO] [1439216282.142961164, 339.300000000]: current-map /eng2/7f/ #<cascaded-coords #Xba8d2a0 /world  0.0 0.0 30000.0 / 0.0 0.0 0.0>
[ INFO] [1439216303.159682741, 349.300000000]: current-map /eng2/7f/ #<cascaded-coords #Xbbb1078 /world  0.0 0.0 30000.0 / 0.0 0.0 0.0>
[ INFO] [1439216324.494423896, 359.301000000]: current-map /eng2/7f/ #<cascaded-coords #Xbcd5150 /world  0.0 0.0 30000.0 / 0.0 0.0 0.0>
/home/aginika/ros/indigo/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/voice_text//text2wave: line 23: nkf: command not found
ssh: Could not resolve hostname c2: Name or service not known
lost connection
command-line: line 0: Bad configuration option: /tmp/sound_playm5df1t.wav
ssh: Could not resolve hostname c2: Name or service not known
[ERROR] [WallTime: 1439216326.570516] [360.238000] Sound synthesis failed. Is festival installed? Is a festival voice installed? Try running "rosdep satisfy sound_play|sh". Refer to http://pr.willowgarage.com/wiki/sound_play/Troubleshooting
[ INFO] [1439216346.462296876, 369.300000000]: current-map /eng2/7f/ #<cascaded-coords #Xbdac580 /world  0.0 0.0 30000.0 / 0.0 0.0 0.0>
[ INFO] [1439216368.361651548, 379.301000000]: current-map /eng2/7f/ #<cascaded-coords #X6e8f1b0 /world  0.0 0.0 30000.0 / 0.0 0.0 0.0>
[ INFO] [1439216391.212530281, 389.300000000]: current-map /eng2/7f/ #<cascaded-coords #X7096490 /world  0.0 0.0 30000.0 / 0.0 0.0 0.0>
[ INFO] [1439216413.992963397, 399.300000000]: current-map /eng2/7f/ #<cascaded-coords #X72cdb90 /world  0.0 0.0 30000.0 / 0.0 0.0 0.0>
[ INFO] [1439216436.295656966, 409.300000000]: current-map /eng2/7f/ #<cascaded-coords #X71e8018 /world  0.0 0.0 30000.0 / 0.0 0.0 0.0>
[ INFO] [1439216459.595081974, 419.300000000]: current-map /eng2/7f/ #<cascaded-coords #X74404d8 /world  0.0 0.0 30000.0 / 0.0 0.0 0.0>

```

aginika, 2015-08-11:
I found that in my environment, the tilt_laser died sometime too.
I try to fix now.

aginika, 2015-09-03:
https://bitbucket.org/osrf/gazebo/issues/1344/intensities-of-laser-scan-is-too-big

furushchev, 2015-12-12:
To summarize, this issue is caused by segfault of `laser_filter` package, which was fixed at https://github.com/ros-perception/laser_filters/pull/34.
So I close this issue.
Please create new issue if this problem is reproduced.

closed 2015-12-12 by furushchev