jsk-ros-pkg/jsk_robot #432: PR1012 チルトレーザ不具合

wkentaro, 2015-09-07:
Kentaro Wada [20:36]
まずはネットワーク問題の切り分け
問題：pr1012がpr1040よりping応答が10倍遅い
prbase1012 -> hub_a, pr1012 -> hub_b, prbase1040 -> hub_c, pr1040 -> hub_d
だったのを
prbase1012 -> hub_a
pr1012 -> hub_b
prbase1040 -> hub_a
pr1040 -> hub_b
にしても変わらず・・・ (edited)

Kentaro Wada [20:40]
LANケーブルも変えましたが同様・・・

Yuki Furuta [20:54]
ケーブル変えて雑談してたところ、ネットワーク速度は変わらないが、チルトが抜ける（＝ｔｆが抜ける）現象は治った？

[image]

なぜか地面を飛び抜ける問題も改善している？

Yuki Furuta [21:00]
さらに別の問題が発生したかもしれません・・・

[image]

画像左のようにチルトの上下で地面位置にギャップがある気がします。

Kei Okada [21:23] 
Serviceportからやってみてください。issue にしましょう

wkentaro, 2015-09-07:
@k-okada @furushchev

k-okada, 2015-09-07:
問題はレーザーとエンコーダーのタイムスタンプがあってい無い可能性がたかい

サービスポートにしてネットワークの問題を回避
C1, C2の同期を確認

furushchev, 2015-09-07:
c1:
```
chronyc> tracking
Reference ID    : 133.11.216.200 (basestation-novpn)
Stratum         : 4
Ref time (UTC)  : Mon Sep  7 13:39:25 2015
System time     : 0.000000175 seconds fast of NTP time
Frequency       : 8.781 ppm slow
Residual freq   : 0.000 ppm
Skew            : 0.018 ppm
Root delay      : 0.004227 seconds
Root dispersion : 0.008101 seconds
```

c2:
```
chronyc> tracking
Reference ID    : 10.68.255.1 (prbase1012)
Stratum         : 4
Ref time (UTC)  : Mon Sep  7 13:41:40 2015
System time     : 0.000000386 seconds slow of NTP time
Frequency       : 4.336 ppm slow
Residual freq   : -0.006 ppm
Skew            : 1.529 ppm
Root delay      : 0.004581 seconds
Root dispersion : 0.009403 seconds
```

となっています。どちらもbasestationを見ていて、ずれてはなさそうです。
が、c1から見るときにvpnの外から見ているのが気になります。
pr1040も確認します。

garaemon, 2015-09-07:
c2は本当はc1を刺さないといけない
けれどもなんか前にうまくあわなかったからbase-stationにしました、みたいな話があった気がする。
とりあえずc2->c1にするといいのかな

furushchev, 2015-09-07:
ｐｒ1040

c1: 
```
chronyc> tracking
Reference ID    : 10.68.255.1 (prbase1040)
Stratum         : 11
Ref time (UTC)  : Mon Sep  7 11:45:59 2015
System time     : 0.000000388 seconds slow of NTP time
Frequency       : 1.612 ppm fast
Residual freq   : -0.000 ppm
Skew            : 0.004 ppm
Root delay      : 0.000352 seconds
Root dispersion : 0.000030 seconds
```

c2:

```
chronyc> tracking
Reference ID    : 10.68.255.1 (prbase1040)
Stratum         : 11
Ref time (UTC)  : Mon Sep  7 13:42:46 2015
System time     : 0.000000463 seconds fast of NTP time
Frequency       : 2.792 ppm slow
Residual freq   : 0.001 ppm
Skew            : 0.019 ppm
Root delay      : 0.000517 seconds
Root dispersion : 0.000007 seconds
```

pr1040もc1,c2ともにbasestationを見ていますが、どちらもVPN内で参照していますね。
@garaemon さんの意見も合わせて
pr10XX <- c1 <- c2
をすべて10.68系で参照するようにしてみます。

garaemon, 2015-09-07:
pingが遅いというのはどこからどこへのping?

furushchev, 2015-09-07:
133系列で
musca -> pr1012
と
musca -> pr1040
です

garaemon, 2015-09-07:
どちらも有線?

furushchev, 2015-09-07:
はい。

garaemon, 2015-09-07:
tracepathで違いはある?

garaemon, 2015-09-07:
あと、ethtoolを入れて各計算機で速度のチェックかな

furushchev, 2015-09-07:
```
musca $ tracepath 133.11.216.201                                                      
 1?: [LOCALHOST]                                         pmtu 1500
 1:  pr1012                                                0.294ms 
 2:  pr1012                                                0.876ms reached
     Resume: pmtu 1500 hops 2 back 63 
```

```
musca $ tracepath 133.11.216.211                                                      
 1?: [LOCALHOST]                                         pmtu 1500
 1:  pr1040                                                0.344ms 
 1:  pr1040                                                0.290ms 
 2:  pr1040                                                0.732ms reached
     Resume: pmtu 1500 hops 2 back 63 
```

変わりはあまり無いようです。
一方pingは

```
 ~ ⮀  ⮀  ⮀ 🍣  ping 133.11.216.201                                                           ⮂  
PING 133.11.216.201 (133.11.216.201) 56(84) bytes of data.
64 bytes from 133.11.216.201: icmp_req=1 ttl=63 time=4.95 ms
64 bytes from 133.11.216.201: icmp_req=2 ttl=63 time=4.52 ms
64 bytes from 133.11.216.201: icmp_req=3 ttl=63 time=5.37 ms
64 bytes from 133.11.216.201: icmp_req=4 ttl=63 time=4.79 ms
^C
--- 133.11.216.201 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3004ms
rtt min/avg/max/mdev = 4.520/4.911/5.375/0.317 ms
 ~ ⮀  ⮀  ⮀ 🍣  ping 133.11.216.211                                                           ⮂  
PING 133.11.216.211 (133.11.216.211) 56(84) bytes of data.
64 bytes from 133.11.216.211: icmp_req=1 ttl=63 time=0.700 ms
64 bytes from 133.11.216.211: icmp_req=2 ttl=63 time=0.567 ms
64 bytes from 133.11.216.211: icmp_req=3 ttl=63 time=0.648 ms
64 bytes from 133.11.216.211: icmp_req=4 ttl=63 time=0.607 ms
^C
--- 133.11.216.211 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3001ms
rtt min/avg/max/mdev = 0.567/0.630/0.700/0.055 ms
```

というように10倍pr1012が遅いです。

ethtoolでprbase1012, prbase1040, pr1012, pr1040, muscaはすべて1000MbのFullであることを確認しました。

garaemon, 2015-09-07:
musca -> {pr1012,pr1040}でmtrしてみてもいいかもしれない

garaemon, 2015-09-07:
3号館のPCからだとどちらもあんまり変わらないですね

garaemon, 2015-09-07:
iperfしてみて適当に通信速度は買ってみるとどうなりますか

furushchev, 2015-09-07:
pr1012:

```bash
🍣  iperf -c 133.11.216.201 -d                                                    ⮂  
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
------------------------------------------------------------
Client connecting to 133.11.216.201, TCP port 5001
TCP window size:  104 KByte (default)
------------------------------------------------------------
[  5] local 133.11.216.42 port 53794 connected with 133.11.216.201 port 5001
[  4] local 133.11.216.42 port 5001 connected with 133.11.216.201 port 42186
[ ID] Interval       Transfer     Bandwidth
[  5]  0.0-10.0 sec   105 MBytes  87.4 Mbits/sec
[  4]  0.0-10.0 sec  90.2 MBytes  75.4 Mbits/sec
```

pr1040:

```bash
🍣  iperf -c 133.11.216.211 -d                                                    ⮂  
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
------------------------------------------------------------
Client connecting to 133.11.216.211, TCP port 5001
TCP window size:  117 KByte (default)
------------------------------------------------------------
[  5] local 133.11.216.42 port 57044 connected with 133.11.216.211 port 5001
[  4] local 133.11.216.42 port 5001 connected with 133.11.216.211 port 39023
[ ID] Interval       Transfer     Bandwidth
[  5]  0.0-10.0 sec   203 MBytes   170 Mbits/sec
[  4]  0.0-10.0 sec   101 MBytes  84.8 Mbits/sec
```

ギガビットと考えると遅いですね。（こんなものなのでしょうか？）

muscaを別のPCに変えて（有線）試しても変わらないので、何かがボトルネックになってそうなのですが。。

garaemon, 2015-09-07:
どっかに100Mの線がありそう。
basestationのvpnなしのローカルネットワークで試してみるとどうなる?

garaemon, 2015-09-07:
間のVPNの有無で速度が変わる場合 => basestationの負荷をチェックして違いがあるかみてみる。rebootして挙動が変わるか見てみる

furushchev, 2015-09-07:
prbase1012:

```bash
🍣  iperf -c 133.11.216.200 -d                                                    ⮂  
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
------------------------------------------------------------
Client connecting to 133.11.216.200, TCP port 5001
TCP window size:  314 KByte (default)
------------------------------------------------------------
[  5] local 133.11.216.42 port 54148 connected with 133.11.216.200 port 5001
[  4] local 133.11.216.42 port 5001 connected with 133.11.216.200 port 51313
[ ID] Interval       Transfer     Bandwidth
[  5]  0.0-10.0 sec   690 MBytes   578 Mbits/sec
[  4]  0.0-10.0 sec  1.02 GBytes   878 Mbits/sec
```

prbase1040:

```bash
🍣  iperf -c 133.11.216.210 -d                                                    ⮂  
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
------------------------------------------------------------
Client connecting to 133.11.216.210, TCP port 5001
TCP window size:  298 KByte (default)
------------------------------------------------------------
[  5] local 133.11.216.42 port 38598 connected with 133.11.216.210 port 5001
[  4] local 133.11.216.42 port 5001 connected with 133.11.216.210 port 55520
[ ID] Interval       Transfer     Bandwidth
[  5]  0.0-10.0 sec   602 MBytes   505 Mbits/sec
[  4]  0.0-10.0 sec  1.07 GBytes   915 Mbits/sec
```

明らかに違いました。basestationの負荷を確認してみます。

k-okada, 2015-09-07:
などと通信の問題だけで時間が経っていか無いようにserviceポートにつなげてパーツ交換必要どうかの目処をつけたいところであります

furushchev, 2015-09-08:
MEMO:

```
export PYTHONPATH=`rospack find pr2_calibration_launch`/pr2_urdf_parser_py:$PYTHONPATH
```

```
Prosilicaが画像をpublishしないときがありますが、
rosrun rqt_reconfigure rqt_reconfigure
から、procilica_cameraを選択
auto_maxを24->12に変更した状態で、Streaming Modeからてきとうな別モードを選択して、
サイド、StreamingModeを選択。
そのあと、prosilica_nodeを再度立ち上げ直すと治ります。
```

furushchev, 2015-09-09:
キャリブが完了しました。上の画像にあるヒステリシス的地面にはならなくなったと思います。
これでチルトがずれていかないか様子を見ていこうと思います。
[image]

furushchev, 2015-09-09:
@wkentaro 一旦closeしましょうか。

closed 2015-09-09 by wkentaro

furushchev, 2015-09-09:
@wkentaro :+1:

furushchev, 2015-09-10:
今度はチルトが上にあがっていってしまっているようです。

[image]

garaemon, 2015-09-10:
徐々に上がっていくのかな?

furushchev, 2015-09-10:
@garaemon 実際に確認はしていませんが、おそらくそうだと思われます。エンコーダのケーブルをWillow製ケーブルからJSKケーブルへ変装してエンコーダケーブルが原因かを特定してみることにします。

furushchev, 2015-09-11:
[image]

JSKで作ったケーブルに変えると
- オフセットはなくなった
- 一部のレーザが床を飛び抜けるようになった。

うーん、どうしたらよいのか。。

garaemon, 2015-09-11:
床が抜けるというのはc2で確認しましたか?

furushchev, 2015-09-11:
モータを駆動させているのはc1でtilt_hokuyo_nodeはc2なのですが、
c1,c2で同期がとれていないという事でしょうか？
今からchronyで調整してもう一度やってみます。

garaemon, 2015-09-11:
「床が時々抜ける」というのは僕の予想だと「レーザが時々読めていない」という事だと思うので、レーザを呼んでいるのはc2だったと記憶しているので、レーザの出力を他に何もかませずにc2で確認すると良いと思います。

garaemon, 2015-09-11:
レーザが抜けなく読めているなら、tfやら同期、ネットワーク通信の問題になります

furushchev, 2015-09-11:
@garaemon c2でこのような抜けを確認するにはどうするのがいいでしょうか。
tfは

```
$ rosrun tf tf_monitor map laser_tilt_link
All Broadcasters:
Node: /amcl 19.8291 Hz, Average Delay: 0.0291049 Max Delay: 0.486613
Node: /image_transport_nodelet_manager 5.54 Hz, Average Delay: 0.242387 Max Delay: 0.555064
Node: /map_tf00_pr1012_29273_4333092886284769788 10.0001 Hz, Average Delay: -0.0990513 Max Delay: 0
Node: /map_tf02_pr1012_29273_4735240064901639361 9.9988 Hz, Average Delay: -0.0990094 Max Delay: 0
Node: /map_tf04_pr1012_29273_3796203073115208613 9.99891 Hz, Average Delay: -0.0990474 Max Delay: 0
Node: /map_tf06_pr1012_29273_8784645973815653104 10.0001 Hz, Average Delay: -0.0990797 Max Delay: 0
Node: /map_tf08_pr1012_29273_6556130392805840466 9.99932 Hz, Average Delay: -0.0989811 Max Delay: 0
Node: /map_tfA_pr1012_29273_1150689146252351765 9.99931 Hz, Average Delay: -0.0989602 Max Delay: 0
Node: /map_tfB_pr1012_29273_5990862063646585778 9.99965 Hz, Average Delay: -0.0990193 Max Delay: 0
Node: /robot_pose_ekf 30.0049 Hz, Average Delay: 0.0578756 Max Delay: 0.115086
Node: /robot_state_publisher 76.9118 Hz, Average Delay: -0.311412 Max Delay: 0.0713511
Node: /tf_mux 9.99852 Hz, Average Delay: -0.0987751 Max Delay: 0
```

という感じで10Hzくらいで引けそうです。

時刻の同期は
c2 -> c1
c1 -> prbase
というようにchronyを設定して同期して見ました。

```
chronyc> tracking
Reference ID    : 10.68.0.1 (pr1012)
Stratum         : 5
Ref time (UTC)  : Fri Sep 11 13:03:41 2015
System time     : 0.000007390 seconds slow of NTP time
Last offset     : -0.000007595 seconds
RMS offset      : 0.000040731 seconds
Frequency       : 4.274 ppm slow
Residual freq   : -0.003 ppm
Skew            : 0.207 ppm
Root delay      : 0.000195 seconds
Root dispersion : 0.000228 seconds
Update interval : 356.2 seconds
Leap status     : Normal
```

しかし未だ床を抜けるレーザがあるようです。（変わってません）

garaemon, 2015-09-11:
c2でレーザのトピックをrostopic hzしたらいいんじゃない?

furushchev, 2015-09-11:
なるほど。確認してみました。

```bash
$ rostopic hz /tilt_scan -w 10
subscribed to [/tilt_scan]
average rate: 42.854
        min: 0.000s max: 0.040s std dev: 0.01844s window: 10
average rate: 43.268
        min: 0.000s max: 0.040s std dev: 0.01876s window: 10
average rate: 42.844
        min: 0.000s max: 0.041s std dev: 0.01874s window: 10
average rate: 43.151
        min: 0.000s max: 0.040s std dev: 0.01871s window: 10
average rate: 43.060
        min: 0.000s max: 0.040s std dev: 0.01861s window: 10
average rate: 43.277
        min: 0.000s max: 0.040s std dev: 0.01893s window: 10
average rate: 42.643
        min: 0.000s max: 0.040s std dev: 0.01839s window: 10
average rate: 43.070
        min: 0.000s max: 0.040s std dev: 0.01869s window: 10
average rate: 43.072
        min: 0.000s max: 0.040s std dev: 0.01843s window: 10
average rate: 43.065
        min: 0.000s max: 0.040s std dev: 0.01880s window: 10
average rate: 42.875
        min: 0.000s max: 0.041s std dev: 0.01882s window: 10
average rate: 43.063
        min: 0.000s max: 0.040s std dev: 0.01876s window: 10
average rate: 43.038
        min: 0.000s max: 0.040s std dev: 0.01859s window: 10
average rate: 42.645
        min: 0.000s max: 0.040s std dev: 0.01856s window: 10
average rate: 42.844
        min: 0.000s max: 0.040s std dev: 0.01870s window: 10
average rate: 43.054
        min: 0.000s max: 0.040s std dev: 0.01881s window: 10
average rate: 43.046
        min: 0.000s max: 0.040s std dev: 0.01862s window: 10
average rate: 42.670
        min: 0.000s max: 0.040s std dev: 0.01858s window: 10
```

レーザが欠損するという感じではない気がします。

garaemon, 2015-09-11:
うーん、だと次はエンコーダが怪しそうだという事で、jont_statesのチルトレーザの角度をプロットしてみるのかな?

furushchev, 2015-09-11:
@garaemon それはすでに確認していて、pr1040と違いはなさそうでした。

garaemon, 2015-09-11:
> 一部のレーザが床を飛び抜けるようになった。

ではこれはなぜ起きるんだろうか。
レーザの値自体がおかしくなってしまうのか?

garaemon, 2015-09-11:
変更したのはエンコーダケーブルで会って、hokuyo自体のケーブルではないんだよね?
なにか接触が悪くなってしまった?

garaemon, 2015-09-11:
時々抜けるというのはいつも同じタイミングなのかな?
だとしたらその関節角で止めると常に再現するようになったりするんだろうか.

タイミングが関係ないとすると、任意の角度で散るとレーザを止めて、常にセンサデータをrvizにだしていると時々、急に変なoutlierがのってくる、という挙動を示すはずですね。

furushchev, 2015-09-11:
@garaemon 手がかりになるかわかりませんが、今のチルトの動きはpr2_move_base.pyが作っている（画像参照）のですが、
これを

```
rosservice call laser_tilt_controller/set_periodic_cmd '{ command: { header: { stamp: 0 }, profile: "linear" , period: 3 , amplitude: 1 , offset: 0 }}'
```

などとして上下対称的な動きにすると抜けはなくなります。

[image]
[image]

garaemon, 2015-09-11:
回転速度変えてみない?

furushchev, 2015-09-11:
了解です。変えたのはエンコーダのケーブルだけでhokuyo自体にはノータッチです。
なので、もう考えられる原因としては「何らかのキャリブデータがおかしい？」「エンコーダが早い方の速度についていけてない」「センサの方も故障している」くらいかなと思っています。
ただpr1040（とおそらく他のPR2）ではこの速度でも大丈夫なのでおかしいなと思っています。

明日朝もう一度回転速度などを遅くして試してみます。

garaemon, 2015-09-11:
もうひとつあり得る可能性としては、速い速度の時にrvizがうまく三次元復元できていないという可能性があります。
そのときは`scan_to_cloud_filter_chain`を`high_fidelity:=true`で使ってみてください
https://github.com/jsk-ros-pkg/jsk_common/blob/master/jsk_tilt_laser/launch/multisense_laser_pipeline.launch#L31

これをかけるとscan -> pointcloudに変換するときにレーザがスキャン中に変化したtfを考慮してくれます.

garaemon, 2015-09-11:
時刻kのスキャンデータdata[i]と時刻k+1のスキャンデータdata_next[i]を考えると、3次元環境はだいたい連続なはずだから
`data[i] - data_next[i]`とかをうまく可視化するとなにかわかるのかもしれない。

ところでケーブル交換と判断するに至った「じょじょに上がっていってしまう現象」はなくなったのかな?

furushchev, 2015-09-27:
たくさんのアドバイスありがとうございました。
いろいろ試しているうちに、
接点復活剤をつけてJSK製のケーブルを何回か挿しなおしたところ、治ってしまいました。
現在１週間程度経過していて、冷蔵庫デモ、エレベータ移動を複数回行なっていますが、異常はないようです。
[image]
[image]

k-okada, 2015-09-28:
よかった。
Supportにも接点復活剤の写真付きで報告していきましょう
60万浮いたので、欲しいものリスト作ってみてください

furushchev, 2015-09-28:
報告しました。
https://support.clearpathrobotics.com/hc/en-us/requests/3308
欲しいものリスト考えます。