jsk-ros-pkg/jsk_robot #468: Mongo DB Error in pr2_bringup.launch

aginika, 2015-10-11:
@furushchev 

```
Error: could not connect to localhost:27017: [Errno 111] Connection refused
Could not connect to mongo server localhost:27017
Make sure mongod is launched on your specified host/port
Traceback (most recent call last):
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rospy/core.py", line 399, in 
signal_shutdown
    h()
  File "/home/inagaki/ros/hydro/src/mongodb_store/mongodb_store/scripts/config_m
anager.py", line 210, in _on_node_shutdown
    self._mongo_client.disconnect()
AttributeError: 'ConfigManager' object has no attribute '_mongo_client'
Error: could not connect to localhost:27017: [Errno 111] Connection refused
Could not connect to mongo server localhost:27017
Make sure mongod is launched on your specified host/port
Traceback (most recent call last):
  File "/home/inagaki/ros/hydro/src/mongodb_store/mongodb_store/scripts/message_
store_node.py", line 236, in <module>
    store = MessageStore()
  File "/home/inagaki/ros/hydro/src/mongodb_store/mongodb_store/scripts/message_
store_node.py", line 31, in __init__
    raise Exception("No Daemon?")
Exception: No Daemon?

```

furushchev, 2015-10-14:
@aginika https://github.com/jsk-ros-pkg/jsk_robot/pull/469

aginika, 2015-10-29:
I confirmed it works in my account on PR2. Thanks, @furushchev 
The mongodb_store and jsk_robot has merged
https://github.com/jsk-ros-pkg/jsk_robot/pull/469
https://github.com/strands-project/mongodb_store/pull/148
https://github.com/strands-project/mongodb_store/pull/151

closed 2015-10-29 by aginika