jsk-ros-pkg/jsk_robot #473: package.xml, CMakeLists.txt and nao-interface.l for naoeus package

kochigami, 2015-10-17:
I wrote package.xml and CMakeLists.txt for ```naoeus```, which uses old ```rosbuild``` compiling system.
Could you give me advice?
If I make a mistake, please let me know. Sorry for my trouble.

What I did:
I created ```nao.l``` by following the first section of ```jsk_nao_robot``` README.
I modified nao-interface.l a little by following pepper-interface.l.
Using ```catkin_create_pkg```, I created ```naoeus_test``` package. Then I edited package.xml and CMakeLists.txt, put nao.l and nao-interface.l in euslisp folder.  
I executed ```roslaunch jsk_nao_startup jsk_nao_startup.launch network_interface:=eth1```
I tried all methods in nao-interface.l, all of them worked correctly.

CMakeLists.txt
```
cmake_minimum_required(VERSION 2.8.3)
project(naoeus_test)

find_package(catkin REQUIRED roseus rostest)

catkin_package()
```
package.xml: https://gist.github.com/kochigami/d5fa49cb19e2339d4c24

nao-interface.l: https://gist.github.com/kochigami/65c1d9ec19c451ebf19e

k-okada, 2015-10-18:
Ok, we have several choices;

Ideally, we should create naoqieus package and create naoqi-interface.l,
also create cmake function for generate <robot>.l. And let peppereus and
naoeus depend on naoqieus.

But keeping three packages such as naoqieus, peppereus and naoqieus
required a lot of maintaining jobs so usually we'll maintain main robot's
package and other package drapeding on that. For example, baxtereus depends
on pr2eus. We should create rosroboteus package and put robot-interface.l
in this. But so far we did not do that.

So;
1) create naoqieus, if you want to become naoqieus maintainer. Beat but
hard
2) let naoeus packages depends in peppereus; easy but not clean
3) update naoeus and let peppereus depends on this; so so,

kochigami, 2015-10-19:
Thank you very much for your advice.
I could understand three ways (
1) means pepereus and naoeus depend on naoqieus
2) means naoeus depends on peppereus
3) means peppereus depends on naoeus
), but I couldn't understand what I should do actually about dependency...

For example, what I did three days ago was just creating catkin package, which has dependency on ```pr2eus```, it has no dependency on ```peppereus```. In this case, I think my ```naoeus``` is independent of ```peppereus```. Is it right...?
 
I have two questions.
1. One package (A) depends on another package (B) means that it has ```build_depend``` and ```run_depend``` to another package (B) in package.xml?
2. this example is right?
naoeus/peppereus package : depends on naoqieus package
naoqieus class is at the top level of naoeus/peppereus robot class
naoqieus class has some common methods with naoeus and peppereus class methods, but each of them overrides other methods if necessary. 
Therefore, the difficulty of making naoqieus package is that understanding the common methods of both robots.
(よく分かっておらず申し訳ございません。英語も意味不明で申し訳ございません。）

k-okada, 2015-11-06:
jsk_nao_robotとjsk_pepper_robotを統合してjsk_naoqi_robotにしようと思っているんですがよいでしょうか？
また，ヒストリが残るように全部書き換えてみましたが（多分いままでcheckoutしたディレクトリがあると,pushするとconflictに成る気がしますが，，，）大丈夫そうでしょうか？，jsk_robotはユーザも多いのでmv するだけにしておいたほうがいいでしょうか．
https://github.com/jsk-ros-pkg/jsk_robot/tree/rename_to_naoqi

k-okada, 2015-11-06:
やっぱヒストリを壊すのはアレかとおもい，
https://github.com/k-okada/jsk_robot/tree/move_to_naoqi
みたいに一回rmして，nao, pepperのコミットを追加してみました．どうでしょうか．
https://github.com/jsk-ros-pkg/jsk_robot/pull/483

k-okada, 2015-11-06:
ok, so I created 
> 1) means pepereus and naoeus depend on naoqieus

version of peppereus at #484, 
see https://github.com/k-okada/jsk_robot/commit/ce363500eef0befac47a6b9bde3bcafa1d02bf4a
and https://github.com/k-okada/jsk_robot/commit/1fefb7a9fe410ab3ee0fb79bd16a6f1532e8c2d0

k-okada, 2015-11-30:
I confirmed this works well on Nao robot with @ban-masa

closed 2015-11-30 by k-okada