jsk-ros-pkg/jsk_robot #475: Creating Pepper/Nao's laboratory introducing demo

kochigami, 2015-10-22:
1. use choregraphe
2. deadline: 24/10/2015

requirements:
following slides contents 
time length: more than eight minutes
checking compatibility of Naoqi 2.0 (old Pepper), 2.1 (Nao), 2.3 (new Pepper)
installing it into robots as an application 
(if possible)
In the future, I want robots (especially Pepper) to show pictures of slides by their tablets. 
As a first step, I want robots to talk instead of me. I want to show slides next to them.

kochigami, 2015-10-28:
what I did: made ten minutes introduction demo (Nao uses gestures while reading dialog, but no interaction)
what I want to do next:
1. making four versions of introduction (three, ten minutes / Japanese, English)
2. learning how to show pictures by using Pepper's tablet
3. designing interaction during the demo (ex. start / end trigger, question towards audiences)
4. installing apps on Pepper

kochigami, 2016-02-20:
目的：デモの時，ペッパーにタブレットでスライドを説明してくれる機能をつくる
　　  変数を持ったペッパー日記を表示できるようにする
　　  rosを使う
- [ ] choregrapheの使い方を勉強する
- [ ] スライド画像をタブレットに表示するプログラムを書く 
- [ ] euslispのプログラムで喋りながらスライド画像を見せるプログラムを書く
- [ ] ペッパー日記を作って表示できるようにする

kochigami, 2016-02-21:
> スライド画像をタブレットに表示するプログラムを書く 

Choregrapheからはできるが，端末からpythonのサンプルプログラムが実行できない．
Choregrapheで無名のプロジェクトを作ると，最終的に以下のパスを渡している．
``` tabletService.showImage("http://198.18.0.1/apps/.lastUpdatedChoregrapheBehavior/cute_pepper.jpg") ```
(Pepperの体内で，写真やアプリは以下にある．
``` /home/nao/.local/share/PackageManager/apps/.lastUploadedChoregrapheBehavior/html/pepper_cute.jpg ```)

しかし，pythonのサンプルプログラムでChoregrapheのプログラムを実行した後，（= ファイル群がペッパーの体内にある）
同じパスを与えてプログラムを実行しても，何も起こらない．
```python altabletservice_showimage.py --ip PepperIP```  のまま，端末にもロボットにも変化が起きない．
NaoqiAPI documentation 2.4.2 ALTabletServiceのShowImageのサンプルコードを使用．

```
#! /usr/bin/env python                                                          
#Example: Use showImage Method

import qi
import argparse
import sys
import time

def main(session):
                                                                             
    #This example uses the showImage method.                                     
    #To Test ALTabletService, you need to run the script ON the robot.           
    
    # Get the service ALTabletService.                                          

    try:
        tabletService = session.service("ALTabletService")
        print "session start"
        # Display a local image located in img folder in the root of the web se\
rver                                                                            
        # The ip of the robot from the tablet is 198.18.0.1                     
        tabletService.showImage("http://198.18.0.1/apps/.lastUpdatedChoregrapheBehavior/cute_pepper.jpg")
        print "show image done"
        time.sleep(3)

        # Hide the web view                                                     
        tabletService.hideImage()
    except Exception, e:
        print "Error was: ", e

if __name__ == "__main__":
    parser = argparse.ArgumentParser()
    parser.add_argument("--ip", type=str, default="127.0.0.1",
                        help="Robot IP address. On robot or Local Naoqi: use '1\
27.0.0.1'.")
    parser.add_argument("--port", type=int, default=9559,
                        help="Naoqi port number")

    args = parser.parse_args()
    session = qi.Session()
    try:
        session.connect("tcp://" + args.ip + ":" + str(args.port))
    except RuntimeError:
        print ("Can't connect to Naoqi at ip \"" + args.ip + "\" on port " + st\
r(args.port) +".\n"
               "Please check your script arguments. Run with -h option for help\
.")
        sys.exit(1)
    main(session)
```

kochigami, 2016-02-22:
プログラムの中に，
To Test ALTabletService, you need to run the script ON the robot.
というコメントがあったため，ペッパー体内で動かさないといけないようだ．
多分，タブレットがロボットとしかつながっていない初期設定になっている．外部から操作できないようになっている．

- [ ] sshしてサンプルコードを試す
- [ ] 外部からタブレットにつなげられないか調べる

k-okada, 2016-02-22:
体内でPythonはしったっけ(pythonは走るけどSDKがない気がする）？走ら無ければ，外部でいいのでCでプログラムを書く方法を調べてみて下さい．

kochigami, 2016-02-22:
アドバイスありがとうございます．
体内でpythonは動きました．(say，ソースコードは下,ShowImageの上のプログラムと違うのは，読み込むサービス（ALTabletServiceなど）と関数のみ )

体内でpythonでNAOqiOSが動くが　（say）
体内でpythonでandroidが動かない   (ShowImage)　ということが分かりました．
ロボットとandroidは接続しています．
（```ping 198.18.0.1```と体内で打つと値が返ってきます．これは体内でifconfigすると，usb0に現れるipであることを確認．）

qimessaging.transportsocket: connect: Connection refused と出るので，qiMessagingが何なのか調べています．
どうしても分からなかったらサンプルコードの実行法をCommunityで質問してみます．

```
#! /usr/bin/env python                                                          
"""Example: Use say Method"""

import qi
import argparse
import sys

def main(session):
    """                                                                         
    This example uses the say method.                                           
    It makes the robot say some text using the module.                          
    """
    # Get the service ALTextToSpeech.                                           
    tts = session.service("ALTextToSpeech")

    #Says a test std::string                                                    
    tts.say("This is a sample text!")

if __name__ == "__main__":
    parser = argparse.ArgumentParser()
    parser.add_argument("--ip", type=str, default="127.0.0.1",
                        help="Robot IP address. On robot or Local Naoqi: use '127.0.0.1'.")
    parser.add_argument("--port", type=int, default=9559,
                        help="Naoqi port number")

    args = parser.parse_args()
    session = qi.Session()
    try:
        session.connect("tcp://" + args.ip + ":" + str(args.port))
    except RuntimeError:
        print ("Can't connect to Naoqi at ip \"" + args.ip + "\" on port " + str(args.port) +".\n"
               "Please check your script arguments. Run with -h option for help.")
        sys.exit(1)
    main(session)
```
実行時
このプログラムを体内の/home/nao以下に置いて，
```python say_test.py```と実行

kochigami, 2016-02-22:
```
qicli call ALTabletService.showImage "http://198.18.0.1/apps/.lastUploadedChoregrapheBehavior/cute_pepper.jpg"　（または "http://198.18.0.1/apps/img/cute_pepper.jpg"　）
```
を体内で実行したら，

```
[W] 9113 qimessaging.transportsocket: connect: Connection refused
true
```
となって実行できた．サンプルコードに何か足すのかもしれない．
画像ファイルの置き場所は，
```
~/.local/share/PackageManager/apps/.lastUploadedChoregrapheBehavior/html
~/.local/share/PackageManager/apps/img/html (imgはフォルダ．下にhtmlフォルダしかなく，アプリですらないができた)
```

kochigami, 2016-02-22:
タブレットをwifiにつなげば，ローカルPCから，体内で先ほど動かしたサンプルプログラムが動くことを確認した．
wifiは，ロボットがつながっているものと一致している必要はなさそう．

ローカルPCから画像を表示するプログラムを実行するのに必要なこと：
・タブレットをwifiにつなぐこと　（sshしてqicliを使うのが便利）
　使ったコマンドは

```
qicli call ALTabletService.getWifiStatus
qicli call ALTabletService.configureWifi security ssid key
```

・ ``` ~/.local/share/PackageManager/apps/img/html ```以下に表示したい画像を置いて，
``` tabletService.showImage("http://198.18.0.1/apps/img/cute_pepper.jpg") ```　のようにパスを渡してあげること

セキュリティに気をつける必要がある
（あまり分かっていないので，問題がありましたらご教示ください．申し訳ございません．）

kochigami, 2016-02-22:
> euslispのプログラムで喋りながらスライド画像を見せるプログラムを書く

写真の切り替えとペッパーの喋りが同期しない．
写真を見せる関数： showImage (file name) サービスリクエストを出す関数
話す： ``` (send *ri* :speak "sentence")```

今は，
```  
(showImage "cute_pepper.jpg")
(send *ri* :speak "こんにちは，\\pau=500\\研究室のアイドルこと，\\pau=500\\ペッパーです．\\pau=1000\\今日は皆さんに研究室の紹介をしますね！\\pau=1000\\おっと，\\pau=100\\これは自分の写真だった．\\pau=1000\\ごめんごめん．")
(unix:sleep 15)　次の写真を見せるまで前の写真をキープ
(showImage "penguin_shitamuki.jpg")
(send *ri* :speak "反省してます")
```
のように話す時間に合わせて(unix:sleep 15)のように入れている．
これしかないのか明日調べる．
この場合，同じスピードで話すように指定する必要がある．

``` jsk_201504_miraikan``` を作った時も同じ問題（動きと喋りが同期しない）で困ったはず．

kochigami, 2016-02-24:
> デモに使えそうな内容に整える

- [ ] AutonomousLifeの状態をとってくる関数がほしい（一緒に歩くデモの時も毎回わざわざdisableしている）
- [ ] AnimatedSayをeuslispで実行する機能がほしい (ALAnimatedSpeech.say)

kochigami, 2016-02-27:
``` pepper_presentation ``` というパッケージを作って，READMEに従って他のパソコンからもデモを動かせることを確認しました．
プルリクエストの議論に動きがあったとき，随時.rosintallの内容・プログラムの内容を書き換えていきます．

とりあえず一段落したので閉じます．ご教示ありがとうございました．

closed 2016-02-27 by kochigami