jsk-ros-pkg/jsk_robot #489: [jsk_naoqi_robot] try executing jsk_pepper_startup.launch

kochigami, 2015-11-17:
It is just a memo, sorry for my trouble. I will close this issue as soon as possible.
I want to know whether the discussion "prevent calling ros::init without name; use node_name command option instead of namespace #44 in naoqi_driver" reached some conclusion or not.

First, I installed jsk_robot package as source codes. 
Then, I tried ```roslaunch jsk_pepper_startup jsk_pepper_startup.launch```.
In this case, naoqi_driver deb package seems to be old (there is a bad node "//pepper_robot"), so I had to install naoqi_driver package as source codes.
  
I have to complete these steps.
1. I have to remember how to switch the roscd path from /opt/ros/indigo... to /catkin_ws/src/... (I had a memo in my home, so it is ok)
2. Check if the installed code works well, if not, read the discussion there carefully.

k-okada, 2015-11-17:
On Tue, Nov 17, 2015 at 10:48 PM, Kanae Kochigami
<notifications@github.com> wrote:
> It is just a memo, sorry for my trouble. I will close this issue as soon as possible.
> I want to know whether the discussion "prevent calling ros::init without name; use node_name command option [...]

you do not have to install naoqi_driver from source code, use deb
package unless you want to use latest changes, and #44 is not a
critical problem, if you use oslaunch jsk_pepper_startup
jsk_pepper_startup.launch.

--
◉ Kei Okada

kochigami, 2015-11-18:
Thank you very much for teaching me.
I checked that there is no problem.
Sorry for my trouble.

closed 2015-11-18 by kochigami