jsk-ros-pkg/jsk_robot #498: pr2 のroslaunch が遅い

k-okada, 2015-11-30:
ちゃんとパッチがあたっているか心配なので，

~k-okada/work に debをおきました．
何かあっても元に戻せるので試してみて下さい．

スクリプトは以下
https://gist.github.com/k-okada/a1990dd4586d9f9fa819

k-okada, 2015-11-30:
you can revert this by
sudo dpkg  -i /var/cache/apt/archives/ros-hydro-roslaunch_1.10.12-0precise-20150327-1953-+0000_amd64.deb 
the deb is also find at 
http://packages.ros.org/ros/ubuntu/pool/main/r/ros-hydro-roslaunch/

h-kamada, 2015-11-30:
robot start　実測で 55秒から 38 秒。そこまで効果はないでしょうか。
とりあえずpr1012は入れたままにしておきます。

garaemon, 2015-12-05:
roslaunch以外も必要という話だったりしませんか. catkin_pkgとか

furushchev, 2015-12-20:
roslibも変える必要がありました。
@k-okada のpatchをforkして追加しました。
https://gist.github.com/furushchev/6c0e1fb3f6ecdb8b744c

furushchev, 2015-12-20:
pr1012にpatchを当てて試して見ました。
robot startはEnterキーを押してから21秒になりました。

k-okada, 2015-12-20:
debはどこにああるかな．バージョンは99ではなくて100にしておきましょう．後でトラック出来るように．

furushchev, 2015-12-20:
debも上のリンクにあります。
今から100に変更します。

furushchev, 2015-12-20:
@k-okada roslaunchとroslibのdebがあって、roslaunchに関しては前のものから変更なしですが、それでも両方100に上げますか？

k-okada, 2015-12-21:
ああ，だったら大丈夫です．~k-okada/workにもdebを置いておきました．

closed 2015-12-21 by k-okada