jsk-ros-pkg/jsk_robot #540: pr2をindigoに上げる

furushchev, 2016-01-23:
そろそろ上げても良いタイミングだと思っていますが、どうでしょうか？
もしよろしければまずは片方だけ(pr1012)上げてみようと思います。

garaemon, 2016-01-23:
上げましょう。

PR2がボトルネックになっています。

一応backupをとっておくとよいですね

furushchev, 2016-01-23:
@garaemon 
了解です。backup取ります。
pr2 user manual と wilikiを参考にアップグレードしていこうとしていますが、他に参照したら良いドキュメントはありますか？

furushchev, 2016-01-23:
前にgroovy -> hydroにしたときはどのような手順だったのでしょうか？

- 単純にdo-release-upgradeをして14.04にする→indigoにするのでいいのか
- basestationも一緒に上げる必要があるのか

が知りたいです。

garaemon, 2016-01-23:
オフィシャルなupgradeマニュアルみたいなのが存在した気がするけど、、、
MLにbasestationのイメージないの?っていう質問なかったかな?

k-okada, 2016-01-23:
https://pr2s.clearpathrobotics.com/wiki/Tutorials/UpgradingToPrecise
が正しい方法（upgradeではなく，netbootからのclean install)だったはずだけど，
http://lists.willowgarage.com/pipermail/pr2-users/2013-November/001832.html
upgradeでもいけるとは思う．https://linuxjm.osdn.jp/html/util-linux/man1/script.1.html
はしらせてログをとりながらすすめましょう．

closed 2017-10-04 by furushchev