jsk-ros-pkg/jsk_robot #547: [Pepper] 手をつないで歩くプログラムを復活させる

kochigami, 2016-02-03:
目的：@jiangjun0105と一緒に作った手を繋いで歩くプログラムを，現在のros-naoqi環境で動くようにする
やること：
- [ ] hand touchに反応するプログラム 
- [ ] 安全装置を切れるプログラム -> 放置していた自分のプログラムを完成させる
- [ ] 実機で確認する
- [ ] 一緒に歩くためのプログラムをどこかに置かせてもらう

I wrote here since I didn't know the right place. Sorry for my trouble.

kochigami, 2016-02-03:
> hand touchに反応するプログラム 
（案１． naoqi_driverでbumper, tactileと一緒にhand touchを加えてもらう (C++)）
案２． 
naoqi_appsにbumper, tactile, hand touch に反応するプログラムを置かせてもらう(python)

-> 以前プルリクエストしたものを，NaoqiNodeを使って書いている
-> NaoqiNodeはThreadの継承クラスなので，案1のプログラムとやっていることは同じはず．

k-okada, 2016-02-03:
https://github.com/ros-naoqi/naoqi_driver
を workspace に入れてコンパイルしてみて下さい．
これでtactile が使えるようになるはずです．

kochigami, 2016-02-04:
ご教示ありがとうございます．
そちらのプログラムには，バンパーと頭の触覚(tactile)の対応はありますが，手のセンサはありません．

今日はあまり作業できなかったため，明日やりかけのpythonのプログラムから考えてみようと思います．
C++のプログラムは自分には暗号ですが，今解読しています．

kochigami, 2016-02-07:
> 案２． naoqi_appsにbumper, tactile, hand touch に反応するプログラムを置かせてもらう(python)

をつくった．
案１は明日やる．

kochigami, 2016-02-08:
> 安全装置を切れるプログラム -> 放置していた自分のプログラムを完成させる

- [ ] 障害物への接近距離の定義を変える
- [ ] 接近距離がいくつになっているか返す （サービス）
- [ ] 安全機能を切る
- [ ] どうして動けないか返す　（サービス）

一緒に”手を繋いで”歩くデモで必要なのは3, 4番目だが，
1, 2 番目も大切と思うので一緒につくる．

> 障害物への接近距離の定義を変える

1. Pepperを籠（障害物）までgo-posで近づける
止まったら自分のプログラムのサービスを使い接近許可距離を変える -> もっと近づけるようになった
2. service call実行時：
```
rosservice call /orthogonal_distance "orthogonal_distance:
  data: 1.0" 
```
を行うと，
```
ERROR: service [/orthogonal_distance] responded with an error: service cannot process request: handler returned invalid value: Invalid number of arguments, args should be [] args are(<class 'std_srvs.srv._Empty.EmptyResponse'>,)
```
となるが，実際は命令は届いている．

k-okada, 2016-02-08:
2016-02-08 19:45 GMT+09:00 Kanae Kochigami <notifications@github.com>:

> rosservice call /orthogonal_distance "orthogonal_distance:
>   data: 1.0"
>
>
これはROSでも最も難しいところなんだけど，http://wiki.ros.org/ROS/YAMLCommandLine
rosservice call /orthogonal_distance "{orthogonal_distance: {  data: 1.0}}"
かな？
orthogonal_distance の メッセージのrossrv show は何が帰ってくる？

--
◉ Kei Okada

kochigami, 2016-02-09:
ご教示ありがとうございます．
> orthogonal_distance の メッセージのrossrv show は何が帰ってくる？

```rossrv show naoqi_bridge_msgs/OrthogonalSecurityDistance```の結果は
```
std_msgs/Float32 orthogonal_distance
  float32 data
---
```
です．

プログラムの中で，返り値を```return EmptyResponse``` -> ```return EmptyResponse()```に直して，

```rosservice call /orthogonal_distance "{orthogonal_distance: { data: 1.0}}"```後の現在のエラーは
```
ERROR: service [/set_orthogonal_distance] responded with an error: service cannot process request: handler returned invalid value: Invalid number of arguments, args should be [] args are(,)
```
になっています．リストとして認識されなくて，引数は2個あるはずと，怒られているのかな，と思います．解決はまだできていません．
もっと奇妙なのは，エラーが出てもサービスは実行されていることです．

入力引数: Float Number+ 出力引数: Empty ->今回の場合
入力引数: Empty + 出力引数: Empty -> エラーなく実行できた
ので，入力の引数の与え方に問題がありそうです．

kochigami, 2016-02-09:
> プログラムの中で，返り値をreturn EmptyResponse -> return EmptyResponse()に直して，
入力引数: Float Number+ 出力引数: Empty ->今回の場合
入力引数: Empty + 出力引数: Empty -> エラーなく実行できた

は勘違い．
srvファイルの中の定義（入力引数，出力引数は別タイプでもよい）： Float Number -> Empty など
service callbackの中の処理： Float Number が引数だったら，返り値はFloat Numberのresponse型で,
EmptyResponseではない
この2つを混同していた．

ビルド時にsrvファイルから，request, responseファイルが作られる．

kochigami, 2016-02-09:
```naoqi_bridge_msgs/srv```以下にファイルを作って```catkin build```しても，ビルドされない．

catkin buildは cmake + make してくれるが，CMakeLists.txtに変更がない限りcmakeしない = naoqi_bridge_msgsのメッセージファイルは追加されない． 
無理やりcmakeしてもらうために，
```catkin bt -v -i --force-cmake```する．

kochigami, 2016-02-10:
jsk_robotの更新をしたら，jsk_pepper_robot, jsk_nao_robotがjsk_naoqi_robot以下に入った（= パスが変わる）：
```catkin build```すると２つのエラー
```
[jsk_pepper_startup] CMake Error: The source "/home/kochigami/catkin_ws/src/jsk_robot/jsk_naoqi_robot/jsk_pepper_startup/CMakeLists.txt" does not match the source "/home/kochigami/catkin_ws/src/jsk_robot/jsk_pepper_robot/jsk_pepper_startup/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
```
-> ```catkin clean -c (= --cmake-cache)```, ```catkin build -v -i --force-cmake```したらビルドできた．
```
[jsk_robot_startup] CMake Error at /opt/ros/indigo/share/dynamic_reconfigure/cmake/dynamic_reconfigure-macros.cmake:57 (add_custom_target): 
[jsk_robot_startup]   add_custom_target cannot create target "jsk_robot_startup_gencfg" because 
[jsk_robot_startup]   another target with the same name already exists.  The existing target is a 
[jsk_robot_startup]   custom target created in source directory                 
[jsk_robot_startup]   "/home/kochigami/catkin_ws/src/jsk_robot/jsk_robot_common/jsk_robot_startup". 
[jsk_robot_startup]   See documentation for policy CMP0002 for more details.    
[jsk_robot_startup] Call Stack (most recent call first):                        
[jsk_robot_startup]   CMakeLists.txt:17 (generate_dynamic_reconfigure_options)  
```
-> 調べ中

k-okada, 2016-02-10:
catkin clean -a
として devel/build を一回全て消してもう一度 catkin build するとどうでしょう．
catkin build jsk_pepper_startup でもいいですね．

kochigami, 2016-02-10:
```catkin clean -b```, ```catkin clean -d```, ```catkin build```をしたらできました．これからは何も考えず消します．
ご教示ありがとうございました．

kochigami, 2016-02-10:
実機で試した．
0.
ペッパーの言語を日本語に
AutonomousLifeはenableのままにした（顔が動いてかわいい）
１. 
```
roslaunch jsk_pepper_startup jsk_pepper_startup.launch
roslaunch naoqi_apps externalCollisionAvoidance.launch
```

２. safety reflexの無効化をロボットウェブページから
３. 
```
rosservice call /set_external_collision_protection_status "name:
  data: 'Move'
status: false"
```
４．
```
roseus roseus <walking demo program>
main
```
５．終わったら
```
rosservice call /set_external_collision_protection_status "name:
  data: 'Move'
status: True"
```
safety reflexの無効化を元に戻す

話す機能：speechトピック -> SpeechWithFeedbackActionで動かしたい
実験結果：
ペッパーと一緒に歩けた．
久しぶりにやったら早くて怖かった． -> スピードを調整する必要がある
ペッパーは3分くらいで落ちた．疲れたのか，自分のプログラムのせいかまだ分からない．

kochigami, 2016-02-18:
> たくさん喋らないようにする

前と進む方向が変わった時だけ話すようにした

> 顔が動くようにする

AutonomousLife：enabled => 自動的にExternalCollisionProtectionEnabled(Move): enabled になるので
手を繋いで歩けなくなる
FaceTrackerを使えるようにしたが，
・たまにあらぬ方向を向く
・前を全く見ない
ところを改善したい．

0.　ペッパーの言語を日本語に
１.  ```roslaunch jsk_pepper_startup jsk_pepper_startup.launch```
2.
```
roslaunch naoqi_apps externalCollisionAvoidance.launch
safety reflexの無効化をロボットウェブページから 
rosservice call /set_external_collision_protection_status "name:
  data: 'Move'
status: false"
```
3．
```
roslaunch naoqi_apps tracker.launch
rosservice call /face_tracker
```
4.
```
roseus roseus walk_together_hand_in_hand.l
main
```
5.
ペッパーの左手の方向に応じて進む方向が決まる
手を繋がない限り進まない（= 危ない時は手を離す）

6．終わったら
```
rosservice call /set_external_collision_protection_status "name:
  data: 'Move'
status: True"
rosservice call /stop_tracker
```
safety reflexの無効化を元に戻す

kochigami, 2016-02-19:
walk_togetherという新しいパッケージを作った
取ってくる必要があるプログラム（walk_togetherの中に説明がある）：
１．　naoqi_apps
https://github.com/ros-naoqi/naoqi_bridge/pull/48
と顔追跡のプログラム 
2. 　naoqi_driver
https://github.com/ros-naoqi/naoqi_driver/pull/63
3. 　naoqi_bridge_msgs
https://github.com/ros-naoqi/naoqi_bridge_msgs/pull/6
（naoqi_driverはビルドするときに0.0.4バージョンのリリースを取ってくる．最新は0.0.5であることに注意）

k-okada, 2016-02-19:
On Fri, Feb 19, 2016 at 9:00 PM, Kanae Kochigami <notifications@github.com>
wrote:

> （naoqi_driverはビルドするときに0.0.4バージョンのリリースを取ってくる．最新は0.0.5であることに注意）
>

これはどういう意味でしょうか？0.0.5はなんか問題あったのかな．

--
◉ Kei Okada

kochigami, 2016-02-19:
0.5.2のリリース以降（去年の8月頃），
naoqi_driverのpackage.xmlにversion_gte="0.0.4"と指定が入っています．
おそらく，更新されていないだけだと思いますが，そもそもどうしてバージョン指定が書かれているのかは分かっていません．

jsk_pepper_startup, naoqieusは，
package.xmlにnaoqi_driverをrun/build dependに加えることでnaoqi_bridge_msgsをインストールしていると思います．
だからソースで入れない限り，0.0.4のバージョンが入ると思いました．（間違っていたらすみません・・・）

自分にとっては，0.0.5でHandTouch.msgが使えるので，ソースで入れないといけないという意味でした．

k-okada, 2016-02-20:
https://github.com/ros-naoqi/naoqi_driver/blob/master/package.xml#L19

ですか．gte は greather than or equal で，naoqi_bridge_msgs
は0.0.4以降を使って下さいね，といういみですが，これは
何も影響はなくて，rosdep
installではその時の最新が入ってきて，もしnaoqi_bridge_msgsが古い時にコンパイルの時にワーニングが出るだけです．

ちなみに，aptでいれても0.0.5はインストールされますね．色々まだいじるだろうからソースで入れていいとは思うけど．
http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/r/ros-indigo-naoqi-bridge-msgs/
http://packages.ros.org/ros/ubuntu/pool/main/r/ros-indigo-naoqi-bridge-msgs/

kochigami, 2016-02-20:
すみません．version_eqの説明と見間違えていました．

> 自分にとっては，0.0.5でHandTouch.msgが使えるので，ソースで入れないといけないという意味でした．

これも今考えると意味がよく分かりません．0.0.5でHandTouch.msgが使えるが，それ以降このデモのために追加したファイルがあるので，自分のブランチのソースを取ってくる必要がある，という意味でした．

プログラムはパッケージにして，jsk_pepper_startup, peppereusの動く他のパソコンでも使えるようになったので，
一旦閉じようと思います．
ご教示ありがとうございました．

closed 2016-02-20 by kochigami