jsk-ros-pkg/jsk_robot #549: [jsk_naoqi_robot/ naoeus] nao-interface.lが古く，nao-interfaceがnaoqi-interfaceを継承していない

kochigami, 2016-02-10:
```naoeus/euslisp``` の中の```nao-interface.l```は，2015年1月５日までの変更で止まっているようです．
github上で```naoeus/euslisp```以下まで移動して，historyを見るとそうなります．

一方，```jsk_naoqi_robot```のページからhistoryを見ると，2015年11月7日の変更で，
```nao-interface.l``` に ```naoqi-interface```を継承することになっています．

naoeus/euslisp/nao-interface.l を```roseus roseus nao-interface.l```で動かそうとしたら．
```
ERROR th=0 no such package "NAOQI_MSGS""NAOQI_MSGS" in #<compiled-code #X4afc910>E:
```
と出て，naoqi-interfaceクラスを継承していないことに気がついたのですが，
nao-interface.lを動かす一番正しい方法は何でしょうか．
```jsk_naoqi_robot/peppereus/pepper-interface.l``` は ```naoqi-interface```を継承しています．

自分に何かできることがあればご教示ください．

k-okada, 2016-02-10:
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/naoeus/nao-interface.l
を使ってください。ややこしいですね。

kochigami, 2016-02-10:
```peppereus/pepper-interface.l``` と同じように，
```naoeus/nao-interface.l``` になったということですね．

見落としておりすみません．ご教示ありがとうございました．

closed 2016-02-10 by kochigami