jsk-ros-pkg/jsk_robot #567: [jsk_pepper_robot] jsk_pepper_startup.launchの立ち上がりが遅いときがある

kochigami, 2016-03-14:
```roslaunch jsk_pepper_startup jsk_pepper_startup.launch```を実行した時，中に含まれる```naoqi_driver/launch/naoqi_driver.launch```の立ち上がりが遅い時がある．（またはnaoqi_dashboardのノードが落ちて，naoqi_driver.launchも上がらないこともある　（ログは取り忘れ））

可能性は，
起動直後　or
AutonomousLifeがsolitaryかinteractive 
ではないか．
分かったらこのissueを閉じる．

```roslaunch jsk_pepper_startup jsk_pepper_startup.launch```の実行結果
###ここで止まる###　で止まる（```naoqi_driver```の立ち上がる前の部分）
```
 roslaunch jsk_pepper_startup jsk_pepper_startup.launch network_interface:=eth1
... logging to /home/kochigami/.ros/log/199232b6-e98b-11e5-8619-bc5ff4fcfea0/roslaunch-kochigami-desktop-26517.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/kochigami/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://192.168.97.146:57461/

SUMMARY
========

CLEAR PARAMETERS
 * /naoqi_dashboard_aggregator/

PARAMETERS
 * /image_saver/filename_format: /tmp/pepper_camer...
 * /image_saver/save_all_image: False
 * /joy_node/autorepeat_rate: 20
 * /joy_node/deadzone: 0.3
 * /joy_node/dev: /dev/input/js0
 * /naoqi_dashboard_aggregator/analyzers/joystick/discard_stale: True
 * /naoqi_dashboard_aggregator/analyzers/joystick/find_and_remove_prefix: joy_node
 * /naoqi_dashboard_aggregator/analyzers/joystick/path: Joystick
 * /naoqi_dashboard_aggregator/analyzers/joystick/type: diagnostic_aggreg...
 * /naoqi_dashboard_aggregator/analyzers/naoqi/analyzers/Computer/find_and_remove_prefix: naoqi_driver_comp...
 * /naoqi_dashboard_aggregator/analyzers/naoqi/analyzers/Computer/path: Computer
 * /naoqi_dashboard_aggregator/analyzers/naoqi/analyzers/Computer/type: diagnostic_aggreg...
 * /naoqi_dashboard_aggregator/analyzers/naoqi/analyzers/Joints/find_and_remove_prefix: naoqi_driver_joints:
 * /naoqi_dashboard_aggregator/analyzers/naoqi/analyzers/Joints/path: Joints
 * /naoqi_dashboard_aggregator/analyzers/naoqi/analyzers/Joints/type: diagnostic_aggreg...
 * /naoqi_dashboard_aggregator/analyzers/naoqi/analyzers/PowerSystem/find_and_remove_prefix: naoqi_driver_batt...
 * /naoqi_dashboard_aggregator/analyzers/naoqi/analyzers/PowerSystem/path: Power System
 * /naoqi_dashboard_aggregator/analyzers/naoqi/analyzers/PowerSystem/type: diagnostic_aggreg...
 * /naoqi_dashboard_aggregator/analyzers/naoqi/path: NAOqi
 * /naoqi_dashboard_aggregator/analyzers/naoqi/type: diagnostic_aggreg...
 * /naoqi_dashboard_aggregator/analyzers/tf/find_and_remove_prefix: tf_monitor:
 * /naoqi_dashboard_aggregator/analyzers/tf/path: TF
 * /naoqi_dashboard_aggregator/analyzers/tf/type: diagnostic_aggreg...
 * /pepper_robot/pose/pose_manager/poses/init/joint_names: ['Body']
 * /pepper_robot/pose/pose_manager/poses/init/positions: [0.0, 0.0, 1.39, ...
 * /pepper_robot/pose/pose_manager/poses/init/time_from_start: 1.5
 * /pepper_robot/pose/pose_manager/poses/zero/joint_names: ['Body']
 * /pepper_robot/pose/pose_manager/poses/zero/positions: [0.0, 0.0, 0.0, 0...
 * /pepper_robot/pose/pose_manager/poses/zero/time_from_start: 1.5
 * /pepper_robot/pose/pose_manager/xap: /home/kochigami/c...
 * /robot/name: dhcp41
 * /robot/type: pepper
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /teleop_twist_joy/axis_angular/yaw: 3
 * /teleop_twist_joy/axis_linear/x: 1.0
 * /teleop_twist_joy/axis_linear/y: 0.0
 * /teleop_twist_joy/enable_button: 4
 * /teleop_twist_joy/enable_turbo_button: 0
 * /teleop_twist_joy/scale_angular/yaw: 1.0
 * /teleop_twist_joy/scale_augular_turbo/yaw: 1.5
 * /teleop_twist_joy/scale_linear/x: 0.7
 * /teleop_twist_joy/scale_linear/y: 0.7
[... 118 lines cut ...]
camera/ir/image_raw is resetting
camera/ir/image_raw reset
/joint_states is resetting
/joint_states reset
laser is resetting
laser reset
/rosout is resetting
/rosout reset
sonar is resetting
sonar reset
resetting subscriber teleop
teleop is resetting
teleop reset
resetting subscriber moveto
moveto is resetting
moveto reset
resetting subscriber speech
speech is resetting
speech reset
resetting subscriber animated_speech
animated_speech is resetting
animated_speech reset
resetting service robot config service
robot config service is resetting
```

kochigami, 2016-05-24:
I think it's different from the problem mentioned above: roslaunch jsk_pepper_startup jsk_pepper_startup.launch sometimes fails. 
I use ubuntu 14.04, ros indigo, naoqi_driver from source (master branch , top commit is "0.5.8" )

- yesterday I updated my naoqi_driver package
- I have to wait for a long time before launching my other programs?
- my created programs do bad things on jsk_pepper_startup? (<=> after trying three times, my programs seem to work properly.)

- [ ] see commits of naoqi_driver
- [ ] wait for a while and then launching my other programs
- [ ] check topic names of my programs

I think this part is problem.
```
[WARN] [WallTime: 1464073031.515735] unable to communicate with ROS Master, registrations are now out of sync
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/diagnostic_common_diagnostics/tf_monitor.py", line 185, in <module>
    diag_updater.update()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/diagnostic_updater/_diagnostic_updater.py", line 245, in update
    self._check_diagnostic_period()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/diagnostic_updater/_diagnostic_updater.py", line 320, in _check_diagnostic_period
    self.period = rospy.get_param("~diagnostic_period", 1)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 462, in get_param
    return _param_server[param_name] #MasterProxy does all the magic for us
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/msproxy.py", line 121, in __getitem__
    code, msg, value = self.target.getParam(rospy.names.get_caller_id(), resolved_key)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1233, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1587, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1273, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1301, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1448, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 975, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 835, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 797, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 778, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 571, in create_connection
    raise err
socket.error: [Errno 111] Connection refused
[tf_monitor-10] process has died [pid 27185, exit code 1, cmd /opt/ros/indigo/lib/diagnostic_common_diagnostics/tf_monitor.py __name:=tf_monitor __log:=/home/kochigami/.ros/log/4c7438a4-217c-11e6-9d6a-bc5ff4fcfea0/tf_monitor-10.log].
log file: /home/kochigami/.ros/log/4c7438a4-217c-11e6-9d6a-bc5ff4fcfea0/tf_monitor-10*.log
```

All launching message is below.
```
roslaunch jsk_pepper_startup jsk_pepper_startup.launch network_interface:=eth1
... logging to /home/kochigami/.ros/log/4c7438a4-217c-11e6-9d6a-bc5ff4fcfea0/roslaunch-kochigami-desktop-27099.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/kochigami/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://192.168.97.25:45190/

SUMMARY
========

CLEAR PARAMETERS
 * /naoqi_dashboard_aggregator/

PARAMETERS
 * /image_saver/filename_format: /tmp/pepper_camer...
 * /image_saver/save_all_image: False
[... 184 lines cut ...]
  File "/usr/lib/python2.7/xmlrpclib.py", line 1233, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1587, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1273, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1301, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1448, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 975, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 835, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 797, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 778, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 571, in create_connection
    raise err
socket.error: [Errno 111] Connection refused
[tf_monitor-10] process has died [pid 27185, exit code 1, cmd /opt/ros/indigo/lib/diagnostic_common_diagnostics/tf_monitor.py __name:=tf_monitor __log:=/home/kochigami/.ros/log/4c7438a4-217c-11e6-9d6a-bc5ff4fcfea0/tf_monitor-10.log].
log file: /home/kochigami/.ros/log/4c7438a4-217c-11e6-9d6a-bc5ff4fcfea0/tf_monitor-10*.log

```

k-okada, 2016-05-24:
please make sure that you do not have existing roscore service (`ps -auxwww
| grep rosmaster` or reboot) and run again, if it does not help, please
provide following information
```
env | grep ROS
cat /etc/hosts
ifconfig -a
```

kochigami, 2016-09-09:
I'm sorry for keeping this issue open for a long time.
Recently, I haven't met the situation.

If the same problem occurs, I'll re-open the issue.

closed 2016-09-09 by kochigami