jsk-ros-pkg/jsk_robot #570: [naoeus, peppereus] :send-stiffness-controller :servo-on :servo-off をnaoqi-interface.lに移しても良いでしょうか．

kochigami, 2016-03-15:
共通のメソッドが2つのプログラムで定義されています．

１．ペッパーが登場する前のナオ：:send-stiffness-controllerを使って:servo-on, offでサーボオン，サーボオフしていた
２．登場したばかりのペッパー：ナオと同じように出来ないので，wakeUp, restという関数を呼び出すように:servo-on, off をオーバーライド
３．現在：ナオもペッパーと同じようにオーバーライドになっている

ペッパーでは，:send-stiffness-controllerは，一部だけサーボオン・オフする時に使っています．
ナオも同じ使い方ができると思います．

#568 でやろうとしていることを，naoqi-interface.l上で書きたいです．
何か勘違いしていたらすみません．

k-okada, 2016-03-17:
はい．そうしてください．

kochigami, 2016-03-26:
ご教示ありがとうございました．

closed 2016-03-26 by kochigami

kochigami, 2016-04-01:
> #568 でやろうとしていること（:start-grasp, :stop-graspで手のstiffnessを変更したい）
を，naoqi-interface.l上で書きたいです．

は以下のコマンドで行えるため，
```:start (stop)-grasp```メソッドを変更してstiffnessという引数を加えるのはやめました．

```
send *ri* :send-stiffness-controller :rhand (lhand) 0.5
send *ri* :start-grasp :rarm (:larm)
```