jsk-ros-pkg/jsk_robot #571: [naoeus, peppereus] catkin btする時にビルドするパッケージnaoqi_bridge_msgs, naoqi_driver がどこから来ているか分からないです．

kochigami, 2016-03-16:
```catkin clean -a``` ```catkin bt``` でpeppereus naoeusをビルドするときに，
naoeus: nao_apps, naoqieus, nao_description, euscollada, naoqi_bridge_msgs, naoqi_driver

peppereus: euscollada, naoqi_bridge_msgs, naoqi_driver_py, naoqi_pose, pepper_bringup, pepper_description, pepper_sensors_py, naoqieus, naoqi_driver, naoqi_sensors_py

をそれぞれビルドします．
今ビルドが通らなくなっている，naoqi_driverがどこで呼ばれているか知りたいです，
（naoqi_driverに依存せずpeppereusが動かせたら，ペッパーはnaoqi_driver_pyを使って動かせるようになるはず）

package.xmlでbuild dependになっているパッケージ(run dependにも含まれている)はそれぞれ
naoeusは
nao_apps, naoqieus, nao_description, rostest 

peppereusは
pepper_bringup, naoqieus, pepper_description, rostest 

それぞれのパッケージの package.xml の build/ run depend を見ると，```catkin bt```でビルドしようとするパッケージのうち，
```naoqi_bridge_msgs```, ```naoqi_driver```以外は含まれます．

k-okada, 2016-03-16:
$ rospack depends-why --target naoqi_driver peppereus
Dependency chains from peppereus to naoqi_driver:
* peppereus -> pepper_bringup -> naoqi_driver
* peppereus -> naoqieus -> naoqi_driver

こういうことかな．

ただ，もんだいはアルデバランのひとが不用意にlibqiのバージョンを上げたことにあるので，

$ wget
http://packages.ros.org/ros/ubuntu/pool/main/r/ros-indigo-naoqi-libqi/ros-indigo-naoqi-libqi_2.3.0-1trusty-20160109-140942-0800_amd64.deb
$ sudo dpkg -i
ros-indigo-naoqi-libqi_2.3.0-1trusty-20160109-140942-0800_amd64.deb

で解決しないかな．

これがこのまえのshadow-fixedにして最新のバイナリにする，ということの弊害ですね．テスト期間がすくないので，こういうことがおこりますね．

kochigami, 2016-03-17:
ご教示ありがとうございます．ビルドできました．

http://answers.ros.org/question/84616/help-understanding-system-and-package-dependencies-with-rosbuild-and-catkin/
http://www.ros.org/reps/rep-0127.html#run-depend-multiple
を読んで，run depend, build dependの役割を勉強し，納得しました．

昨日はペッパー動かなくなったけれど，qiについて少し勉強できたので良かったです．

closed 2016-03-17 by kochigami