jsk-ros-pkg/jsk_robot #572: jsk_pr2_robot gazebo simulation error

mmurooka, 2016-03-18:
On Ubuntu 14.04/Indigo machine,

```
murooka@ThinkPad-P50:~/ros/indigo/src/jsk-ros-pkg/jsk_robot/jsk_pr2_robot/jsk_pr2_startup$ roslaunch jsk_pr2_startup pr2_bringup.launch 
... logging to /home/murooka/.ros/log/37d25bc0-ecec-11e5-997c-507b9da5ba9d/roslaunch-ThinkPad-P50-7591.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

while processing /opt/ros/indigo/share/pr2_machine/pr2.machine:
environment variable 'ROS_ENV_LOADER' is not set. 

Machine xml is <machine address="c2" env-loader="$(env ROS_ENV_LOADER)" name="c2"/>
The traceback for the exception was written to the log file
```

furushchev, 2016-03-18:
Currenlty `pr2_bringup.launch` is only for bringing up PR2 real robot. (this is used only for `robot start`)

mmurooka, 2016-03-18:
I'd like to use :go-pos from robot-interface in gazebo.
default pr2_gazebo seems to be not enough.
How can I do that?

furushchev, 2016-03-18:
you can use `pr2_gazebo.launch` instead.
Please check if environment variable `ROBOT`is set to be `sim` before launching.

mmurooka, 2016-03-21:
Thank you. It worked.

I got some troubles to launch pr2_gazebo.launch without errors.
So, I'll make another Issues.

closed 2016-03-21 by mmurooka