jsk-ros-pkg/jsk_robot #576: jsk_pr2_startup dependency

mmurooka, 2016-03-21:
https://github.com/jsk-ros-pkg/jsk_robot/issues/572#issuecomment-199180495

1.
following pacakges are not installed by `rosdep install jsk_pr2_startup`

pr2_navigation_perception
pr2_navigation_config

2.
following packages are not built by `catkin build jsk_pr2_startup`

jsk_maps
jsk_android_gui_api9
face_recognition
pr2_move_base
jsk_smart_gui
pr2_arm_kinematics
pr2_tuck_arms_action
safe_teleop_base
pr2_teleop_general

k-okada, 2016-03-22:
please add them to package.xml, if they're released
-> http://repositories.ros.org/status_page/ros_indigo_default.html?q=pr2
-> http://repositories.ros.org/status_page/ros_indigo_default.html?q=jsk_maps

mmurooka, 2016-03-31:
close via https://github.com/jsk-ros-pkg/jsk_robot/pull/577.

closed 2016-03-31 by mmurooka