jsk-ros-pkg/jsk_robot #578: [naoeus] naoeus / nao.l が作れません

kochigami, 2016-03-26:
```
roscd naoeus 
git pull origin master
catkin bt
```
した結果
```
==> Expanding alias 'bt' from 'catkin bt' to 'catkin b --this'
==> Expanding alias 'b' from 'catkin b --this' to 'catkin build --this'
--------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/indigo
Workspace:                   /home/kochigami/catkin_ws
Source Space:       [exists] /home/kochigami/catkin_ws/src
Build Space:        [exists] /home/kochigami/catkin_ws/build
Devel Space:        [exists] /home/kochigami/catkin_ws/devel
Install Space:     [missing] /home/kochigami/catkin_ws/install
DESTDIR:                     None
--------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
--------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
--------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------------- 
Found '66' packages in 0.1 seconds. 
Starting ==> nao_apps                                                           
Starting ==> nao_description                                                    
Starting ==> naoqi_bridge_msgs                                                  
Finished <== nao_apps          [ 0.2 seconds ]                                  
Finished <== nao_description   [ 0.2 seconds ]                                  
Finished <== naoqi_bridge_msgs [ 1.5 seconds ]                                  
Starting ==> naoqi_driver                                                       
Finished <== naoqi_driver      [ 0.4 seconds ]                                  
Starting ==> naoqieus                                                           
Finished <== naoqieus          [ 0.2 seconds ]                                  
Starting ==> naoeus                                                             
Finished <== naoeus            [ 0.2 seconds ]                                  
[build] Finished.                                                               
[build] Runtime: 2.5 seconds 
```
ビルドは通るのですが，naoeus / nao.l が作られません．
何か必要な手順が欠けていますでしょうか．

CMakeLists.txtの中身を勉強する必要があります．

k-okada, 2016-03-26:
rm -fr build/naoeus; catkin_build -v -i naoeus --make-args VERBOSE=1
とどうなるかな．
sudo apt-get install ros-indigo-nao-meshes
したらいいかも

kochigami, 2016-03-26:
すみません．できました．
初めは，
```
[naoeus] -- Found nao.urdf at /home/kochigami/catkin_ws/src/nao_robot/nao_description/urdf/naoV50_generated_urdf/nao.urdf 
[naoeus] CMake Warning at /home/kochigami/catkin_ws/src/jsk_robot/jsk_naoqi_robot/naoqieus/cmake/compile_naoqi_model.cmake:23 (find_package): 
[naoeus]   By not providing "Findnao_meshes.cmake" in CMAKE_MODULE_PATH this project 
[naoeus]   has asked CMake to find a package configuration file provided by       
[naoeus]   "nao_meshes", but CMake did not find one.                              
[naoeus]                                                                          
[naoeus]   Could not find a package configuration file provided by "nao_meshes" with 
[naoeus]   any of the following names:                                            
[naoeus]                                                                          
[naoeus]     nao_meshesConfig.cmake                                               
[naoeus]     nao_meshes-config.cmake                                              
[naoeus]                                                                          
[naoeus]   Add the installation prefix of "nao_meshes" to CMAKE_PREFIX_PATH or set 
[naoeus]   "nao_meshes_DIR" to a directory containing one of the above files.  If 
[naoeus]   "nao_meshes" provides a separate development package or SDK, be sure it has 
[naoeus]   been installed.                                                        
[naoeus] Call Stack (most recent call first):                                     
[naoeus]   CMakeLists.txt:11 (compile_naoqi_model)                                
[naoeus]                                                                          
[naoeus]                                                                          
[naoeus] CMake Warning at /home/kochigami/catkin_ws/src/jsk_robot/jsk_naoqi_robot/naoqieus/cmake/compile_naoqi_model.cmake:27 (message): 
[naoeus]   Please install ros-$ROS_DISTRO-nao-meshes manually                     
[naoeus] <== '/home/kochigami/catkin_ws/build/naoeus/build_env.sh /usr/bin/cmake /home/kochigami/catkin_ws/src/jsk_robot/jsk_naoqi_robot/naoeus -DCATKIN_DEVEL_PREFIX=/home/kochigami/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/kochigami/catkin_ws/install' finished with return code '0' 
[naoeus] ==> '/home/kochigami/catkin_ws/build/naoeus/build_env.sh /usr/bin/make VERBOSE=1 --jobserver-fds=4,6 -j' in '/home/kochigami/catkin_ws/build/naoeus' 
[naoeus] /usr/bin/cmake -H/home/kochigami/catkin_ws/src/jsk_robot/jsk_naoqi_robot/naoeus -B/home/kochigami/catkin_ws/build/naoeus --check-build-system CMakeFiles/Makefile.cmake 0 
[naoeus] /usr/bin/cmake -E cmake_progress_start /home/kochigami/catkin_ws/build/naoeus/CMakeFiles /home/kochigami/catkin_ws/build/naoeus/CMakeFiles/progress.marks 
[naoeus] /usr/bin/make -f CMakeFiles/Makefile2 all                                
[naoeus] make[1]:  `/home/kochigami/catkin_ws/build/naoeus'                       
[naoeus] make[1]: `all' .                                                         
[naoeus] make[1]:  `/home/kochigami/catkin_ws/build/naoeus'                       
[naoeus] /usr/bin/cmake -E cmake_progress_start /home/kochigami/catkin_ws/build/naoeus/CMakeFiles 0 
[naoeus] <== '/home/kochigami/catkin_ws/build/naoeus/build_env.sh /usr/bin/make VERBOSE=1 --jobserver-fds=4,6 -j' finished with return code '0' 
Finished <== naoeus            [ 1.8 seconds ]                                    
```
と出たため，```sudo apt-get install ros-indigo-nao-meshes``` したところ，

```
Starting ==> naoeus                                                                                                                   
[naoeus] ==> '/home/kochigami/catkin_ws/build/naoeus/build_env.sh /usr/bin/cmake /home/kochigami/catkin_ws/src/jsk_robot/jsk_naoqi_robot/naoeus -DCATKIN_DEVEL_PREFIX=/home/kochigami/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/kochigami/catkin_ws/install' in '/home/kochigami/catkin_ws/build/naoeus' 
[naoeus] -- The C compiler identification is GNU 4.8.4                                                                                
[naoeus] -- The CXX compiler identification is GNU 4.8.4                                                                              
[naoeus] -- Check for working C compiler: /usr/bin/cc                                                                                 
[naoeus] -- Check for working C compiler: /usr/bin/cc -- works                                                                        
[naoeus] -- Detecting C compiler ABI info                                                                                             
[naoeus] -- Detecting C compiler ABI info - done                                                                                      
[naoeus] -- Check for working CXX compiler: /usr/bin/c++                                                                              
[naoeus] -- Check for working CXX compiler: /usr/bin/c++ -- works                                                                     
[naoeus] -- Detecting CXX compiler ABI info                                                                                           
[naoeus] -- Detecting CXX compiler ABI info - done                                                                                    
[naoeus] -- Using CATKIN_DEVEL_PREFIX: /home/kochigami/catkin_ws/devel                                                                
[... cut ...]
[naoeus] make[1]:  `/home/kochigami/catkin_ws/build/naoeus'                                                                           
[naoeus] /usr/bin/cmake -E cmake_progress_start /home/kochigami/catkin_ws/build/naoeus/CMakeFiles 0                                   
[naoeus] <== '/home/kochigami/catkin_ws/build/naoeus/build_env.sh /usr/bin/make VERBOSE=1 --jobserver-fds=4,6 -j' finished with return code '0' 
Finished <== naoeus            [ 7.3 seconds ]                                                   
```
となりうまくいきました．
ご教示ありがとうございました．

closed 2016-03-26 by kochigami