jsk-ros-pkg/jsk_robot #589: joint_stateが短いロボットがいるとactive_user.l がsegfault する

pazeshun, 2016-04-15:
以下のような変則的なjoint_satesを出すロボットがあるんですが、以下の部分でsegfaultします。そもそもこのコードは何をしているでしょうか？

https://github.com/jsk-ros-pkg/jsk_robot/commit/ec4cef26926905c0ee89c14f8e3179ac9feb5e52#diff-02bfe2e6e2f7f48f7105aba603d9b379R24

```
---
header: 
  seq: 3590
  stamp: 
    secs: 1460710778
    nsecs: 509445315
  frame_id: ''
name: ['right_gripper_vacuum_pad_joint']
position: [0.0]
velocity: [0.0]
effort: [0.0]
---
header: 
  seq: 18282779
  stamp: 
    secs: 1460710778
    nsecs: 517841580
  frame_id: ''
name: ['head_nod', 'head_pan', 'left_e0', 'left_e1', 'left_s0', 'left_s1', 'left_w0', 'left_w1', 'left_w2', 'right_e0', 'right_e1', 'right_s0', 'right_s1', 'right_w0', 'right_w1', 'right_w2', 'torso_t0']
position: [0.0, -0.05292233718204677, -1.6497963373707625, 2.350058567040454, 1.7011846937649238, -0.04141748127290617, 1.5953400194008303, 0.1928980840765908, 0.0003834951969713534, 1.6402089574464787, 2.350442062237425, -1.7008011985679525, -0.04180097646987752, -1.5953400194008303, 0.16413594430373926, -0.0003834951969713534, -12.565987119160338]
velocity: [0.0, 0.012566370612001944, -0.013744467856877126, -0.011780972448751823, 0.006675884387626033, 0.009817477040626518, -0.006675884387626033, 0.01610066234662749, 0.01649336142825255, -0.0007853981632501215, 0.012173671530376884, -0.0007853981632501215, 0.0, -0.023169245815878585, -0.013351768775252066, 0.0, 0.0]
effort: [0.0, 0.0, -21.928, -1.148, 1.652, 10.184, 1.016, -4.644, 0.532, 21.832, -2.604, -2.272, 8.568, -0.364, -4.064, -0.588, -20.48]
---

```

pazeshun, 2016-04-15:
```

(defun joint-cb (msg)
  ;;                                                                                                                                                  
  (cond ((ros::has-param "/robot/type")
         (setq *position* (concatenate float-vector (subseq (send msg :position) 12 16) (subseq (send msg :position) 17))))
	(t
	 (setq *position* (send	msg :position))))
  (update-activeness)
  )
```
みたいなコードにしたらpr2では動くでしょうか？ > @furushchev

pazeshun, 2016-04-15:
```
(defun joint-cb (msg)
  ;;                                                                                                                                                  
  (cond ((ros::has-param "/robot/type")
         (setq *position* (concatenate float-vector (subseq (send msg :position) 12 16) (subseq (send msg :position) 17))))
        ((ros::has-param "/rethink/software_version")
         (if (< (length (send msg :position)) 12)
             (return-from joint-cb nil))
         (setq *position* (send msg :position)))
        (t
         (setq *position* (send msg :position))))
  (update-activeness)
  )
```
はどうでしょうか。

k-okada, 2016-05-13:
Closed via #590

closed 2016-05-13 by k-okada