jsk-ros-pkg/jsk_robot #597: [jsk_robot_startup] Build fails when roseus is installed from source

wkentaro, 2016-05-01:


wkentaro, 2016-05-01:
```
==> Expanding alias 'b' from 'catkin b jsk_robot_startup --summary --no-status --no-color' to 'catkin build jsk_robot_startup --summary --no-status --no-color'
--------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/indigo
Workspace:                   /home/wkentaro/ros/indigo
--------------------------------------------------------------
Source Space:       [exists] /home/wkentaro/ros/indigo/src
Log Space:          [exists] /home/wkentaro/ros/indigo/logs
Build Space:        [exists] /home/wkentaro/ros/indigo/build
Devel Space:        [exists] /home/wkentaro/ros/indigo/devel
Install Space:      [unused] /home/wkentaro/ros/indigo/install
DESTDIR:            [unused] None
--------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------
[build] Found '134' packages in 0.0 seconds.                                   
[build] Package table is up to date.                                           
Starting >>> roseus                          
Finished <<< roseus                           [ 0.2 seconds ]
Starting >>> jsk_robot_startup               
_______________________________________________________________________________
Errors << jsk_robot_startup:cmake /home/wkentaro/ros/indigo/logs/jsk_robot_startup/build.cmake.005.log
CMake Error at /opt/ros/indigo/share/dynamic_reconfigure/cmake/dynamic_reconfigure-macros.cmake:71 (add_custom_target):
  add_custom_target cannot create target "jsk_robot_startup_gencfg" because
  another target with the same name already exists.  The existing target is a
  custom target created in source directory
  "/home/wkentaro/ros/indigo/src/jsk-ros-pkg/jsk_robot/jsk_robot_common/jsk_robot_startup".
  See documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  CMakeLists.txt:17 (generate_dynamic_reconfigure_options)

cd /home/wkentaro/ros/indigo/build/jsk_robot_startup; catkin build --get-env jsk_robot_startup | catkin env -si  /usr/bin/cmake /home/wkentaro/ros/indigo/src/jsk-ros-pkg/jsk_robot/jsk_robot_common/jsk_robot_startup --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/wkentaro/ros/indigo/devel/.private/jsk_robot_startup -DCMAKE_INSTALL_PREFIX=/home/wkentaro/ros/indigo/install; cd -

...............................................................................
Failed << jsk_robot_startup:cmake           [ Exited with code 1 ]
Failed <<< jsk_robot_startup                [ 1.0 seconds ]

[build] Successful packages:

 [   Ignored] baxter_common                      
 [   Ignored] baxter_core_msgs                   
 [   Ignored] baxter_description                 
 [   Ignored] baxter_examples                    
 [   Ignored] baxter_gazebo                      
 [   Ignored] baxter_in_bin_recognition          
 [   Ignored] baxter_interface                   
 [   Ignored] baxter_loop_fold_verify            
 [   Ignored] baxter_maintenance_msgs            
 [   Ignored] baxter_sdk                         
 [   Ignored] baxter_sim_controllers             
 [   Ignored] baxter_sim_examples                
 [   Ignored] baxter_sim_hardware                
 [   Ignored] baxter_sim_io                      
 [   Ignored] baxter_sim_kinematics              
 [   Ignored] baxter_simulator                   
 [   Ignored] baxter_tools                       
 [   Ignored] baxtereus                          
[... 102 lines cut ...]
 [   Ignored] rqt_shell                          
 [   Ignored] rqt_srv                            
 [   Ignored] rqt_top                            
 [   Ignored] rqt_topic                          
 [   Ignored] rqt_web                            
 [   Ignored] rrbot_control                      
 [   Ignored] rrbot_description                  
 [   Ignored] rrbot_gazebo                       
 [   Ignored] stereo_image_proc                  
 [   Ignored] virtual_force_publisher            
 [   Ignored] vision_opencv                      
 [   Ignored] wkentaro_ros_pkg                   
 [   Ignored] wkentaro_tools                     

[build] Failed packages:

 [    Failed] jsk_robot_startup                  

[build] Summary: 1 of 2 packages succeeded.
[build] Ignored: 132 packages were skipped or are blacklisted.
[build] Warnings: None.
[build] Abandoned: No packages were abandoned.
[build] Failed: 1 packages failed.
[build] Runtime: 1.9 seconds total.
```

k-okada, 2016-05-06:
is this related to jsk_robot_startup_gencfg ?
how about grep -ri jsk_robot_startup_gencfg build logs

--
◉ Kei Okada

wkentaro, 2016-05-20:
```
% grep -ri jsk_robot_startup_gencfg build 
build/build_logs/jsk_robot_startup.log:  add_custom_target cannot create target "jsk_robot_startup_gencfg" because
build/jsk_robot_startup/catkin_generated/installspace/jsk_robot_startupConfig.cmake:set(jsk_robot_startup_EXPORTED_TARGETS "jsk_robot_startup_gencfg")
```

ban-masa, 2016-09-23:
I have the same problem.
Is this solved?

devel/share/jsk_robot_startup/cmake/jsk_robot_startupConfig.cmake is called from roseus/cmake/roseus.cmake:90
```
find_package(${_pkg} QUIET) ## this may fail
```

k-okada, 2016-09-23:
please provide all output messages, you can use gist.github.com

ban-masa, 2016-09-23:
These are output logs.
catkin build log of jsk_robot_startup
https://gist.github.com/ban-masa/db0eb7c46c4a6c433d10ad286196c25e

manually cmake log with debug option CMAKE_FIND_DEBUG_MODE
https://gist.github.com/ban-masa/9249f3ad4e8d255748dc59d2cffedaaf

k-okada, 2016-09-23:
please re-run catkin bt after rm -fr ~/ros/indigo/build/jsk_robot_startup
~/ros/indigo/build/roseus

ban-masa, 2016-09-23:
re-run catkin bt after removed build/jsk_robot_startup and build/roseus
https://gist.github.com/ban-masa/01f917e85b12755c4b10029dbea476d9

k-okada, 2016-09-23:
do you have /home/banmasa/ros/indigo/src/jsk-ros-pkg/jsk_robot/jsk_robot_common/jsk_robot_startup/build/devel/share/jsk_robot_startup/cmake/jsk_robot_startupConfig.cmake
file?
if so something wrong

cd src/jsk-ros-pkg/jsk_robot
git clean -xfd  # this removes untracked file, make sure what you do
before you run this command
find . -iname .catkin_tools
and I think you'll find this directory under
`/home/banmasa/ros/indigo/src/jsk-ros-pkg/jsk_robot/jsk_robot_common/jsk_robot_startup/`
so remove that directory and run catkin b again
--
◉ Kei Okada

ban-masa, 2016-09-23:
I made /home/banmasa/ros/indigo/src/jsk-ros-pkg/jsk_robot/jsk_robot_common/jsk_robot_startup/build and did cmake in order to debug.

But I did catkin bt after I removed /home/banmasa/ros/indigo/src/jsk-ros-pkg/jsk_robot/jsk_robot_common/jsk_robot_startup/build, indigo/build/roseus, indigo/build/jsk_robot_startup.

I tried ```find . -iname .catkin_tools```, but I can't find anything.
I did ```git reset --hard; git status``` too, but there is no untracked directory named build.

I think $HOME/ros/indigo/build/jsk_robot_startup/catkin_generated/installspace/jsk_robot_startupConfig.cmake 
or
$HOME/ros/indigo/devel/.private/jsk_robot_startup/share/jsk_robot_startup/cmake/jsk_robot_startupConfig.cmake
may be called from roseus.cmake.

wkentaro, 2016-09-23:
@ban-masa This issue won't fix on my environment, so I currently I stopped to install roseus from source.

@k-okada 
My error log is like below, do you have any idea?
```bash
% mkdir -p ~/ros/jsk_robot_ws/src
% cd ~/ros/jsk_robot_ws
% catkin init
% cd src
% wstool set jsk_robot https://github.com/jsk-ros-pkg/jsk_robot.git --git -vmaster -y
% wstool set jsk_roseus https://github.com/jsk-ros-pkg/jsk_roseus.git --git -vmaster -y
% wstool update
% wstool info
workspace: /home/wkentaro/ros/jsk_robot_ws/src

 Localname  S SCM Version (Spec) UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------  - --- -------------- -----------  ---------------------------
 jsk_roseus   git master         b270a6cb49d6 git://github.com/jsk-ros-pkg/jsk_roseus.git
 jsk_robot    git master         5cf2265cd31a git://github.com/jsk-ros-pkg/jsk_robot.git
% pwd
/home/wkentaro/ros/jsk_robot_ws
% ls
src
% catkin build -iv --no-color --no-status > log.txt
```
https://gist.github.com/d59531f931ea892492f712d9d69ddeab

k-okada, 2016-09-23:
I see, how about changing
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_robot_common/jsk_robot_startup/CMakeLists.txt#L4-L13
to
find_package(catkin REQUIRED COMPONENTS
  dynamic_reconfigure
)

k-okada, 2016-09-23:
may be we need to undef XXX_gencfg after we find_packge() in roseus.cmake

ban-masa, 2016-09-23:
catkin bt succeeded after I fixed find_package as described above.

catkin bt also succeeded when I removed only roseus from find_package in jsk_robot_startup/CMakeLists.txt.

k-okada, 2016-09-23:
nice, please create PR for changing find_package()

ban-masa, 2016-09-23:
Should I remove all package except for dynamic_reconfigure or only roseus from find_package?

ban-masa, 2016-09-23:
I made PR #663

k-okada, 2016-09-26:
hope https://github.com/jsk-ros-pkg/jsk_roseus/pull/477 solves this problem