jsk-ros-pkg/jsk_robot #619: [jsk_naoqi_robot] pepperの点群の出し方についてのご相談

kochigami, 2016-05-31:
画像認識がうまくいかないのは，点群も原因なのかなと思いここに相談を書かせていただきます．
パッケージとは関係がなく申し訳ございません．

rviz のdepth_imageを見ると以下の様になっていますが．
[image]
 
以下のプログラム（depth_image_procを使って点群を出す）だと，画像が曲がってしまいます．（柱が折れ曲がっている）
[image]

```
<!-- -*- mode: xml -*- -->
<launch>
  <node pkg="tf" name="depth_to_top_camera_frame_publisher" type="static_transform_publisher" args="0.039 -0.044 0.035 0.000 0.000 0.000 CameraTop_optical_frame CameraDepth_optical_frame 10"/>

  <node pkg="nodelet" type="nodelet" name="register_depth" output="screen"
        args="standalone depth_image_proc/register" >
    <remap from="rgb/camera_info" to="/pepper_robot/camera/front/camera_info"/>
    <remap from="depth/camera_info" to="/pepper_robot/camera/depth/camera_info"/>
    <remap from="depth/image_rect" to="/pepper_robot/camera/depth/image_raw"/>
    <remap from="depth_registered/camera_info" to="/pepper_robot/camera/depth_registered/camera_info"/>
    <remap from="depth_registered/image_rect" to="/pepper_robot/camera/depth_registered/image_rect"/>
  </node>

  <node pkg="nodelet" type="nodelet" name="cloudify" output="screen"
        args="standalone depth_image_proc/point_cloud_xyzrgb" >
    <remap from="rgb/camera_info" to="/pepper_robot/camera/front/camera_info"/>
    <remap from="rgb/image_rect_color" to="/pepper_robot/camera/front/image_raw"/>
    <remap from="depth_registered/image_rect" to="/pepper_robot/camera/depth_registered/image_rect"/>
    <remap from="depth_registered/points" to="/pepper_robot/camera/depth/points"/>
  </node>
</launch>
```
image_proc を使ってimage_raw ではなく，image_rect_colorを使うようにしましたが，あまり変わりませんでした．
[image]

```
<!-- -*- mode: xml -*- -->
<launch>
  <node pkg="tf" name="depth_to_top_camera_frame_publisher" type="static_transform_publisher" args="0.039 -0.044 0.035 0.000 0.000 0.000 CameraTop_optical_frame CameraDepth_optical_frame 10"/>

  <node pkg="image_proc" type="image_proc" name="rgb_processing" output="screen" ns="/pepper_robot/camera/front">
  </node>

  <node pkg="nodelet" type="nodelet" name="register_depth" output="screen"
        args="standalone depth_image_proc/register" >
    <remap from="rgb/camera_info" to="/pepper_robot/camera/front/camera_info"/>
    <remap from="depth/camera_info" to="/pepper_robot/camera/depth/camera_info"/>
    <!-- <remap from="depth/image_rect" to="/pepper_robot/camera/depth/image_raw"/> -->
    <remap from="depth/image_rect" to="/pepper_robot/camera/depth/image_raw"/>
    <remap from="depth_registered/camera_info" to="/pepper_robot/camera/depth_registered/camera_info"/>
    <remap from="depth_registered/image_rect" to="/pepper_robot/camera/depth_registered/image_rect"/>
  </node>

  <node pkg="nodelet" type="nodelet" name="cloudify" output="screen"
        args="standalone depth_image_proc/point_cloud_xyzrgb" >
    <remap from="rgb/camera_info" to="/pepper_robot/camera/front/camera_info"/>
    <remap from="rgb/image_rect_color" to="/pepper_robot/camera/front/image_rect_color"/>
    <remap from="depth_registered/image_rect" to="/pepper_robot/camera/depth_registered/image_rect"/>
    <remap from="depth_registered/points" to="/pepper_robot/camera/depth/points"/>
  </node>
</launch>
```

k-okada, 2016-05-31:
１． ペッパーのカメラつかっている？それとも追加でつけたデプスセンサ？
ペッパーのカメラのデプスはこのまえみたみたいにガタガタ．（そういえば，あの画像をキャプチャしてくれると嬉しい）
治そうとしているんだけど，たぶん，naoqiでデータを取ってる時か，データを送るときにjpeg圧縮しているんじゃないかと思うんだけど，心当たりないかな？

２． ずれているのは点群？それとも，点群の上のカメラの色情報？
まずは色無しで点群だけで 平面が平面に見えているか？ 線（柱）が線に見えているか？を確認する
色情報はここから点群の各点に色を追加している．これをregistrationという．これは色カメラのパラメータと距離カメラのパラメータから，
どの点には，どの色カメラ画像の座標の色を使えばいいか，という計算して，点に色をつける．正確に言うとrectをつかわないとけない．
ただ，rectとrawで
http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration の
http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration?action=AttachFile&do=get&target=mono_1.png
と
http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration?action=AttachFile&do=get&target=mono_2.png
ぐらい違っていないと，どっちを使っても結果はそうかわらない．

kochigami, 2016-06-01:
ご教示ありがとうございます．
> １． ペッパーのカメラつかっている？それとも追加でつけたデプスセンサ？

ペッパーのカメラです．
rvizでdepth imageから点群を出した結果は以下になります．縦に伸びているのは壁です．
[image]

調べたこと：
- 画像の圧縮はバッファを使ってやるらしい（http://opencv.jp/opencv2-x-samples/imencode_imdecode）
- naoqi_driver/src/recorder/camera.cpp はバッファの処理たくさん + history を見ると，途中からバッファの変更が追加されている
- naoqi_driver/src/converters/camera.cpp void CameraConverter::callAll　内の214-215行目
```
cv::Mat cv_img(image.height, image.width, cv_mat_type_, image.buffer);
msg_ = cv_bridge::CvImage(std_msgs::Header(), msg_colorspace_, cv_img).toImageMsg();
```

- buffer:  naoqi_driver/src/tools/naoqi_image.hpp NaoqiImage構造体の要素 
- naoqi_driver/src/tools/from_any_value.cpp fromAnyValueToNaoqiImage内でセット（中身の意味は分からない...）
```
 /** buffer **/
  ref = anyref[6].content();
  if(ref.kind() == qi::TypeKind_Raw)
  {
    result.buffer = (void*)ref.asRaw().first;
  }
  else
  {
    ss << "Could not retrieve buffer";
    throw std::runtime_error(ss.str());
  }
```
- http://doc.aldebaran.com/2-1/naoqi/vision/alvideodevice-api.html?highlight=alvideo#ALVideoDeviceProxy::getImageRemote__ssCR
も配列の[6]にアクセスして画像データとしている
- convert -> record -> publish topic の順に処理しているようだ

まだわからないこと：
- CameraConverter::callAll　どこで呼んでいるのか

> ２． ずれているのは点群？それとも，点群の上のカメラの色情報？

ずれているのは点群の上の色情報ですので，デプスカメラとフロントカメラ間のtfを直します．
[image]

kochigami, 2016-06-01:
kinectとpepperカメラのトピックのスピードの比較
=> 
- ペッパーはrgb, depth 共にpublishの周期が早くなっている
- naoqi_sensors_py/nodes/camera.py を先に見てみる
```
kinect rgb
rostopic hz /camera/rgb/image_raw
subscribed to [/camera/rgb/image_raw]
average rate: 30.090
	min: 0.031s max: 0.035s std dev: 0.00146s window: 20
average rate: 30.035
	min: 0.031s max: 0.035s std dev: 0.00130s window: 50
average rate: 30.015
	min: 0.031s max: 0.036s std dev: 0.00126s window: 81
average rate: 30.011
	min: 0.031s max: 0.036s std dev: 0.00123s window: 111
average rate: 30.008
	min: 0.031s max: 0.036s std dev: 0.00117s window: 141
average rate: 29.998
	min: 0.031s max: 0.036s std dev: 0.00115s window: 167
```

```
pepper rgb
rostopic hz /pepper_robot/camera/front/image_raw
subscribed to [/pepper_robot/camera/front/image_raw]
average rate: 2.438
	min: 0.410s max: 0.410s std dev: 0.00000s window: 2
average rate: 2.086
	min: 0.410s max: 0.550s std dev: 0.05716s window: 4
average rate: 2.116
	min: 0.386s max: 0.550s std dev: 0.06577s window: 6
average rate: 2.229
	min: 0.360s max: 0.550s std dev: 0.06890s window: 8
average rate: 2.174
	min: 0.360s max: 0.550s std dev: 0.06923s window: 11
average rate: 2.101
	min: 0.360s max: 0.637s std dev: 0.08333s window: 12
```

```
kinect point cloud
rostopic hz /camera/depth_registered/points 
subscribed to [/camera/depth_registered/points]
average rate: 29.059
	min: 0.025s max: 0.059s std dev: 0.00696s window: 23
average rate: 29.627
	min: 0.022s max: 0.059s std dev: 0.00621s window: 53
average rate: 29.777
	min: 0.022s max: 0.059s std dev: 0.00545s window: 83
average rate: 29.807
	min: 0.022s max: 0.059s std dev: 0.00530s window: 113
average rate: 29.858
	min: 0.022s max: 0.059s std dev: 0.00510s window: 143
average rate: 29.875
	min: 0.022s max: 0.059s std dev: 0.00517s window: 173
average rate: 29.890
	min: 0.022s max: 0.059s std dev: 0.00504s window: 203
average rate: 29.760
	min: 0.012s max: 0.067s std dev: 0.00577s window: 232
average rate: 29.800
	min: 0.012s max: 0.067s std dev: 0.00559s window: 262
average rate: 29.810
	min: 0.012s max: 0.067s std dev: 0.00541s window: 292
^Caverage rate: 29.807
	min: 0.012s max: 0.067s std dev: 0.00534s window: 315
```

```
pepper point cloud
rostopic hz /pepper_robot/camera/depth/points 
subscribed to [/pepper_robot/camera/depth/points]
average rate: 2.002
	min: 0.500s max: 0.500s std dev: 0.00000s window: 2
average rate: 2.167
	min: 0.411s max: 0.500s std dev: 0.03745s window: 4
average rate: 2.094
	min: 0.411s max: 0.507s std dev: 0.03516s window: 6
average rate: 2.068
	min: 0.411s max: 0.507s std dev: 0.03154s window: 8
average rate: 2.054
	min: 0.411s max: 0.512s std dev: 0.02919s window: 10
average rate: 2.050
	min: 0.411s max: 0.512s std dev: 0.02666s window: 12
average rate: 2.040
	min: 0.411s max: 0.523s std dev: 0.02631s window: 14
average rate: 1.975
	min: 0.411s max: 0.713s std dev: 0.06058s window: 16
average rate: 1.978
	min: 0.411s max: 0.713s std dev: 0.05742s window: 18
```

```
roslaunch pepper_bringup pepper_full_py.launch nao_ip:=...
 rostopic hz /pepper_robot/camera/front/camera/image_raw 
subscribed to [/pepper_robot/camera/front/camera/image_raw]
average rate: 3.060
	min: 0.148s max: 0.500s std dev: 0.14387s window: 4
average rate: 3.064
	min: 0.148s max: 0.500s std dev: 0.11144s window: 6
average rate: 3.000
	min: 0.148s max: 0.500s std dev: 0.10097s window: 10
average rate: 2.926
	min: 0.148s max: 0.500s std dev: 0.09317s window: 12
average rate: 2.896
	min: 0.148s max: 0.500s std dev: 0.08902s window: 15
average rate: 3.049
	min: 0.148s max: 0.500s std dev: 0.08765s window: 19
^Caverage rate: 3.169
	min: 0.148s max: 0.500s std dev: 0.08508s window: 23
```

kochigami, 2016-06-01:
I added 
```
std::cout << "The following transports were declared locally" << std::endl;
std::vector<std::string> tmp = it.getDeclaredTransports();
for(size_t i = 0; i < tmp.size(); ++i)
  std::cout << " * " << tmp[i] << std::endl;
```
to publishers/camera.cpp
in https://github.com/ros-naoqi/naoqi_driver/commit/99c010b9001420a17a095ed612cfe6cfaf4c8ba7

roslaunch naoqi_driver output is as follows:
```
NOT going to re-register the converters
camera/bottom/image_raw is resetting
The following transports were declared locally
 * image_transport/compressed
 * image_transport/raw
camera/bottom/image_raw reset
camera/depth/image_raw is resetting
The following transports were declared locally
 * image_transport/compressed
 * image_transport/raw
camera/depth/image_raw reset
...
camera/front/image_raw is resetting
The following transports were declared locally
 * image_transport/compressed
 * image_transport/raw
camera/front/image_raw reset
```

```
topic related to image
/pepper_robot/camera/bottom/camera_info
/pepper_robot/camera/bottom/image_raw
/pepper_robot/camera/bottom/image_raw/compressed
/pepper_robot/camera/bottom/image_raw/compressed/parameter_descriptions
/pepper_robot/camera/bottom/image_raw/compressed/parameter_updates
/pepper_robot/camera/depth/camera_info
/pepper_robot/camera/depth/image_raw
/pepper_robot/camera/depth/image_raw/compressed
/pepper_robot/camera/depth/image_raw/compressed/parameter_descriptions
/pepper_robot/camera/depth/image_raw/compressed/parameter_updates
/pepper_robot/camera/front/camera_info
/pepper_robot/camera/front/image_raw
/pepper_robot/camera/front/image_raw/compressed
/pepper_robot/camera/front/image_raw/compressed/parameter_descriptions
/pepper_robot/camera/front/image_raw/compressed/parameter_updates
```

```
rostopic hz /pepper_robot/camera/depth/image_raw/
subscribed to [/pepper_robot/camera/depth/image_raw]
average rate: 1.411
	min: 0.709s max: 0.709s std dev: 0.00000s window: 2
average rate: 1.802
	min: 0.360s max: 0.709s std dev: 0.14546s window: 4
average rate: 1.851
	min: 0.360s max: 0.709s std dev: 0.11426s window: 6
average rate: 1.828
	min: 0.360s max: 0.709s std dev: 0.10546s window: 7
average rate: 1.794
	min: 0.360s max: 0.709s std dev: 0.09307s window: 9
^Caverage rate: 1.803
	min: 0.360s max: 0.709s std dev: 0.09159s window: 11

kochigami@kochigami-desktop:~/catkin_ws/src/grub_toy/launch$ rostopic hz /pepper_robot/camera/depth/image_raw/compressed
subscribed to [/pepper_robot/camera/depth/image_raw/compressed]
average rate: 1.800
	min: 0.555s max: 0.555s std dev: 0.00000s window: 2
average rate: 1.968
	min: 0.471s max: 0.555s std dev: 0.03532s window: 4
average rate: 1.997
	min: 0.383s max: 0.597s std dev: 0.07367s window: 6
average rate: 1.895
	min: 0.383s max: 0.693s std dev: 0.09166s window: 8
average rate: 1.987
	min: 0.311s max: 0.693s std dev: 0.10573s window: 10
^Caverage rate: 1.953
	min: 0.311s max: 0.693s std dev: 0.10357s window: 11
```
rviz output (depth image from /pepper_robot/camera/depth/image_raw + .../front/image_raw)
(I couldn't receive /compressed topic and based on the rviz display, /compressed topic might be published as /image_raw ? )
[image]

kochigami, 2016-06-01:
- getImageRemote (NAOqi API) : we can get uncompressed image
reference: 
https://community.ald.softbankrobotics.com/en/forum/live-video-stream-nao-cameras-833
https://community.ald.softbankrobotics.com/ja/node/2871

<=> I made pcd file of wall, but showing it on Cloud viewer seemed to take a long time...
- naoqi_driver/publishers/camera.cpp: it publishes compressed topic, but I think it is not used.

- Somehow I increased the problem. Pepper color resolution became weird. => reboot pepper

[to be continued]

kochigami, 2016-06-02:
I created pcd file by ```getImageRemote``` NAOqi API function.
comparison of pepper model: I think there is no difference 
- front view 

pepper developer version
[image]

pepper for sale version
[image]

- wall view from sky

pepper developer version
[image]

pepper for sale version
[image]

- table view from side

pepper developer version
[image]

pepper for sale version
[image]

kochigami, 2016-06-02:
現状の結論：
- 元々こういうものなのではないか

- 処理の流れ

NAOqi API ```getImageRemote``` => Naoqi MSG => ROS MSG => publish ROS topic
(naoqi_driver/src/converters)          (naoqi_driver/src/converters)   (naoqi_driver/src/publishers)
圧縮して，代入しているものはない（topicとして出してはいるけれど）
- 最初の```getImageRemote``` のデプス画像をpcdファイルにすると，点の数は240(height)*320(width)=76800 で圧縮なし

この時点でガタガタしている．

まだ理解できていないが理解したいこと：
- pepper体内のNAOqiAPIの扱い方

参考画像：　xtion (height: 480 width: 640)
[image]
[image]

kochigami, 2016-06-02:
とりあえずこのissueは閉じて，本当に困ったらフォーラムで質問します．
ご教示ありがとうございました．

closed 2016-06-02 by kochigami

kochigami, 2016-06-02:
カメラの視野角が狭い分，遠くへ行けば行くほど歪むのかもしれません．
近くを覗き見たら点群はまっすぐになりました．
```
The distance from Pepper to the table is 40 cm.
send *pepper* :head :neck-p :joint-angle 30
```

[image]

k-okada, 2016-06-06:
これは同じロボット？だいぶ違うね．
カメラの位置姿勢はいかが参考になるかな．
http://doc.aldebaran.com/2-4/family/pepper_technical/video_3D_pep.html#d-camera-pepper
http://doc.aldebaran.com/2-4/family/pepper_technical/video_pep.html

kochigami, 2016-06-06:
ご教示ありがとうございました．
カメラのことをあまり分かっていないので勉強します．

（以下，カラー画像と点群の重ね合わせのズレについて）
- CameraTop_optical_frame CameraDepth_optical_frame間のtfを-0.044から目分量で0.074に直した
```
<node pkg="tf" name="depth_to_top_camera_frame_publisher" type="static_transform_publisher" args="0.039 0.074 0.030 0.000 0.000 0.000 CameraTop_optical_frame CameraDepth_optical_frame 10"/>
```
- ros-naoqiのどこかでissueか何か書くところがあれば，後で書こうと思います．
（そもそも上のxtion単体のカラー画像がずれているようなので，pepperのtfの問題ではないかもしれません．）

変更前
[image]

変更後
[image]

k-okada, 2016-06-06:
2016-06-06 13:50 GMT+09:00 Kanae Kochigami <notifications@github.com>:

>
>    - CameraTop_optical_frame
>    CameraDepth_optical_frame間のtfを-0.044から目分量で0.074に直した
>
>
すばらしい．pepper_robot ? noaqi_driver に元コードがあるかな？そこにPR出しましょう．

--
◉ Kei Okada

kochigami, 2016-06-06:
[image]
[image]

k-okada, 2016-06-06:
＞ros-naoqiのどこかでissueか何か書くところがあれば，後で書こうと思います． （そもそも上のxtion単体のカラー画像がずれているようなので，pepperのtfの問題ではないかもしれません．）

https://github.com/ros-naoqi/pepper_robot/blob/master/pepper_description/urdf/pepper1.0_generated_urdf/pepper_sensors.xacro#L120
を変えればいいのかな．？

kochigami, 2016-06-07:
- コミュニティに質問しました

https://community.ald.softbankrobotics.com/en/forum/peppers-depth-image-compressed-publishing-10977

- 解像度とfpsを変えて ```getImageRemote```を使う

基本は320×240, 20fpsです． 
resolution:=0 (160×120), 1 (320×240), 2 (640×480) 
fps:=1, 5, 10, 20,30 で```getImageRemote```からpcdファイルの作成を試してみましたが，
サイズが小さくなっただけで，歪みは変わらない気がしました．
点の数：
resolution:=0, 1 => 各々160×120, 320×240
resolution:=2の時だけ点の数が4800になってしまった（また別の問題かと思います）

320×240　壁を横から
[image]

160×120　壁を横から
[image]

kochigami, 2016-06-07:
> https://github.com/ros-naoqi/pepper_robot/blob/master/pepper_description/urdf/pepper1.0_generated_urdf/pepper_sensors.xacro#L120
を変えればいいのかな．？

ご教示ありがとうございます．
ここで定義されているリンク同士のx, y, zの引き算と
tf_echoの結果が合わないので，理由が分かったらプルリクエストを出そうと思います．

（以下自分用のメモ）
Head -1-  CameraTop_frame -2- CameraTop_optical_frame 
Head -3- CameraDepth_frame -4- CameraDepth_optical_frame
1. xyz= 0.0868 0 0.1631
3. xyz= 0.05138 0.039 0.1194　（1と3の引き算でどうやって0.044, -0.035が出てくるかまだ分からない）
↓回転している向きは同じ
2. -1.57... 0 -1.57... 
4. -1.57... 0 -1.57...

```
rosrun tf tf_echo CameraTop_optical_frame CameraDepth_optical_frame 10
At time 1465299058.567
- Translation: [-0.039, 0.044, -0.035]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
            in RPY (radian) [0.000, -0.000, 0.000]
            in RPY (degree) [0.000, -0.000, 0.000]
```

kochigami, 2016-06-09:
> コミュニティに質問しました

try kRawDepthColorSpace instead of kDepthColorSpace => xtion単体のように平らにならなかったが，ボコボコは少なくなった
変更：
https://github.com/kochigami/naoqi_driver/blob/try-depth-raw/src/converters/camera.cpp#L129
```colorspace_((camera_source_!=AL::kDepthCamera)?AL::kRGBColorSpace:AL::kRawDepthColorSpace ),```
https://github.com/kochigami/naoqi_driver/blob/try-depth-raw/src/tools/alvisiondefinitions.h#L81
```const int kRawDepthColorSpace = 23;```

結果：
kDepthColorSpace (depth cloud in rviz)
[image]

kRawDepthColorSpace (depth cloud in rviz)
[image]

kRawDepthColorSpace  pcd file from ```getImageRemote``` 点の数は76800 (320*240)
[image]

カメラで今後やるべきことまとめ：
- tfを直しプルリクエストを送る
- kRawDepthColorSpaceで効果ありかもう一度確認する
- kRawDepthColorSpaceもオプションで選択できるようにプルリクエストを送る

kochigami, 2016-06-09:
> kRawDepthColorSpaceで効果ありかもう一度確認する

効果はある．
Rawの時点でボコボコしているので，きっとこの時点で圧縮されている．
しかしkDepthColorSpaceを使うと，上半分くらいで点群の層が前に飛び出していて，もっとボコボコになる．（おそらくさらに圧縮している）

kDepthColorSpace 壁の点群の層が二段に分かれている
[image]
[image]

kRawDepthColorSpace 層レベルのボコボコは見られない
[image]
[image]

kochigami, 2016-09-07:
時間が経ってしまったのですが、点群が歪んでしまう原因として

１．圧縮説
２．私の計算違う説
３．ペッパーの目の丸みで歪む説
があると思うのですが、

[image]
```
roslaunch jsk_pepper_startup jsk_pepper_startup.launch
rosrun rviz rviz -d path/to/pepper_bringup/config/pepper.rviz 
```
でデプスセンサと赤外線カメラ画像（同じデバイスから出てくるそうです）
https://github.com/ros-naoqi/naoqi_driver/blob/master/src/converters/camera.cpp#L149
を比べたら、赤外線カメラ画像は歪んでいないので、３はなさそうです。（理解が間違っていたらすみません）

・　```getRemoteImage``` というNAOqiの関数から得た点群が歪んでいる、
・　```kRawDepthColorSpace``` を使ってもやや歪んでいる
ので、1が濃厚という結論で終わっていますが、
2ももう一度確認します。

k-okada, 2016-09-09:
> converters/camera.cpp#L149

3はあるかもね．目のパーツ外せれば，一回外したい．

furushchev, 2016-09-13:
pepperにレーシックをしてみました。

[image]
[image]

手術前ではdepth画像の右上、左上に若干白くなってしまう部分があり、rvizでみるとその部分に何もないにも関わらずpepperの目から1.5mくらいのところに障害物があることになっていたのが、術後では治っているように思います。

[image]
[image]

ただやはりまだ平面を見た時の点群はゆがんでいるようです。

[image]

そこで、https://github.com/kochigami/naoqi_driver/commit/d8ce96404251adf236ebe14084037a7a1a053112
のようにしてRAWのdepth画像を使ってみると、以下のようにxtion単体と比較しても遜色ない点群が出るようになったと思います。

[image]

[image]
[image]

kochigami, 2016-09-14:
手術ありがとうございます。
私も確認してみました。

レーシック手術あり・なしの比較　（kRawDepthImageColorSpace使用）
点群の歪みがなくなっています。

レンズ摘出なし
[image]

レンズ摘出あり
[image]

kRawDepthImageColorSpaceとkDepthImageColorSpaceの比較　（レーシック手術なし）
> 手術前ではdepth画像の右上、左上に若干白くなってしまう部分があり、rvizでみるとその部分に何もないにも関わらずpepperの目から1.5mくらいのところに障害物があることになっていた

に関係しているように思います。

kDepthImageColorSpace
[image]

kRawDepthImageColorSpace
[image]

@furushchev さんのコメントと合わせると、
レンズを取ると、歪みが取れる＋見えてはいけないものが見えなくなる
kRawDepthImageColorSpaceを使っても、歪みが取れる＋見えてはいけないものが見えなくなる
２つを組み合わせるととても良いということが分かりました。

kochigami, 2016-10-16:
@furushchev 

すみません、
上のコメントでrtabmap_rosを使われていたとき、debファイルを使われていましたか。
それともソースから実行されましたか。

私はrosのindigoで、debの 0.11.8-0trusty-20161004-020811-0700 を使っていて、
RGB-D SLAM モードで試しているのですが、
地図が上のようにきれいにできないのと、自己位置推定にすぐ失敗するので困っています。

立ち上げるlaunchファイルはこのような感じでしたか。
もし覚えていらっしゃればご教示ください。ご迷惑をお掛けします。

(odom_frame_idを設定しないといけないのでしょうか。
そうであれば、ソースが必要で、今はビルドが通りません。)

```
<launch>
  <arg name="rviz" default="false" />
  <arg name="rtabmapviz" default="true" />
  <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
    <arg name="rtabmap_args" value="--delete_db_on_start" />
    <arg name="frame_id" value="base_link"/>
    <arg name="depth_topic" value="/pepper_robot/camera/depth_registered/sw_registered/image_rect_raw" />
    <arg name="rgb_topic" value="/pepper_robot/camera/front/image_raw" />
    <arg name="camera_info_topic" value="/pepper_robot/camera/front/camera_info" />
    <arg name="rviz" value="$(arg rviz)" />
    <arg name="rtabmapviz" value="$(arg rtabmapviz)" />
  </include>
</launch>
```

furushchev, 2016-10-16:
@kochigami debianからインストールしたと思います。 launchファイルもそれでいい気がします。 rtabmapvizに今までの履歴をリセットするボタンがあるので、それを最初に押してから始めると綺麗に取れると思います。
（あとちょっとでも激しく動くとダメでした。）

kochigami, 2016-10-17:
@furushchev 
ご教示ありがとうございます。動かし方を変えてみます。