jsk-ros-pkg/jsk_robot #621: [baxtereus] (send self :state :potentio-vector :wait-until-update t)が動かない

pazeshun, 2016-06-17:
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_baxter_robot/baxtereus/baxter-interface.l#L236 を、複数種類の`/robot/joint_states`に対応するため`(push (send self :state :potentio-vector :wait-until-update t) avs)`と書き換えてみたところ、以下のようなエラーが出てきた。
```
/opt/ros/indigo/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 cannot find method :to-nsec:to-nsec in (send ts :to-nsec)E:
```
変数を出力させてみたところ、
```
interrupt1.B1-irteusgl$ ,ts 
nil 
2.B1-irteusgl$ ,(assoc :stamp-list robot-state) 
(:stamp-list #<ros::time #X7d4c7e8 1466133685.976> #<ros::time #X7d4c7e8 1466133685.976> #<ros::time #X7d4c7e8 1466133685.976> #<ros::time #X7d4c7e8 1466133685.976> #<ros::time #X7d4c7e8 1466133685.976> #<ros::time #X7d4c7e8 1466133685.976> #<ros::time #X7d4c7e8 1466133685.976> #<ros::time #X7d4c7e8 1466133685.976> #<ros::time #X7d4c7e8 1466133685.976> #<ros::time #X7d4c7e8 1466133685.976> #<ros::time #X7d4c7e8 1466133685.976> #<ros::time #X7d4c7e8 1466133685.976> #<ros::time #X7d4c7e8 1466133685.976> #<ros::time #X7d4c7e8 1466133685.976> #<ros::time #X7d4c7e8 1466133685.976> #<ros::time #X7d4c7e8 1466133685.976> nil) 
```
のように、最後にnilが入ってしまっているのが原因ではないかと考えられる。

pazeshun, 2016-06-17:
`(send *ri* :state :potentio-vector :wait-until-update t)`単体で発生しました。
```
2.irteusgl$ (send *ri* :state :potentio-vector :wait-until-update t)
/opt/ros/indigo/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: cannot find method :to-nsec in (send ts :to-nsec)
3.E1-irteusgl$ ,ts
nil
4.E1-irteusgl$ ,robot_state
*unbound*
5.E1-irteusgl$ ,robot-state
((:stamp . #<ros::time #X77acfc0 1466136509.043>) (:stamp-list #<ros::time #X68ffca0 1466136509.037> #<ros::time #X68ffca0 1466136509.037> #<ros::time #X68ffca0 1466136509.037> #<ros::time #X68ffca0 1466136509.037> #<ros::time #X68ffca0 1466136509.037> #<ros::time #X68ffca0 1466136509.037> #<ros::time #X68ffca0 1466136509.037> #<ros::time #X68ffca0 1466136509.037> #<ros::time #X68ffca0 1466136509.037> #<ros::time #X68ffca0 1466136509.037> #<ros::time #X68ffca0 1466136509.037> #<ros::time #X68ffca0 1466136509.037> #<ros::time #X68ffca0 1466136509.037> #<ros::time #X68ffca0 1466136509.037> #<ros::time #X68ffca0 1466136509.037> #<ros::time #X68ffca0 1466136509.037> nil) (:effort . #f(0.0 0.0 -21.928 -1.24 1.368 10.068 1.036 -4.456 0.548 22.712 -2.632 -1.868 8.232 -0.116 -4.472 -0.54 0.0)) (:velocity . #f(0.0 -0.003142 -0.014923 0.006676 -0.027882 -0.003534 0.00589 0.001963 -0.016493 -0.003142 0.00589 -0.009817 0.009032 -0.006283 0.008247 0.003534 0.0)) (:position . #f(0.0 -0.049471 -1.64941 2.35044 1.7008 -0.041034 1.59572 0.164519 0.0 1.63906 2.35044 -1.70118 -0.043335 -1.59496 0.163752 0.0 0.0)) (:name "head_nod" "head_pan" "left_e0" "left_e1" "left_s0" "left_s1" "left_w0" "left_w1" "left_w2" "right_e0" "right_e1" "right_s0" "right_s1" "right_w0" "right_w1" "right_w2" "torso_t0"))
```

mmurooka, 2016-06-17:
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/40 で僕の入れたバグだと思うので見てみます．

mmurooka, 2016-06-17:
https://github.com/jsk-ros-pkg/jsk_robot/pull/622 で直ったと思います．

[メモ] メソッドの返ってくる時間の比較
```
9.irteusgl$ (dotimes (i 10) (bench (send *ri* :state :potentio-vector)))
;; time -> 0.002906[s]
;; time -> 0.000316[s]
;; time -> 0.000302[s]
;; time -> 0.0003[s]
;; time -> 0.000307[s]
;; time -> 0.000302[s]
;; time -> 0.000288[s]
;; time -> 0.000299[s]
;; time -> 0.00029[s]
;; time -> 0.000291[s]
nil
10.irteusgl$ (dotimes (i 10) (bench (send *ri* :state :potentio-vector :wait-until-update t)))
;; time -> 0.029206[s]
;; time -> 0.009861[s]
;; time -> 0.010032[s]
;; time -> 0.029897[s]
;; time -> 0.009919[s]
;; time -> 0.009816[s]
;; time -> 0.030272[s]
;; time -> 0.009831[s]
;; time -> 0.009757[s]
;; time -> 0.049429[s]
nil
```

closed 2016-06-20 by k-okada