jsk-ros-pkg/jsk_robot #654: [peppereus] pepper-interface.lでgazeboのペッパーの関節角を動かせるようにする

kochigami, 2016-09-08:
```send *pepper* :head :neck-p :joint-angle 10``` , ```send *ri* :angle-vector (send *pepper* :angle-vector)``` のようにしてgazeboでも動くと便利

- [ ] 立ち上げる ```*ri*```を実機モデルとシミュレータどちらにするか区別しているところを見つける
- [ ] ```send *ri* :angle-vector (send *pepper* :angle-vector)``` が具体的にどのような処理を行っているか見つける
- [ ] gazeboでは各jointのcontrollerに分かれているので、それを振り分けるメソッドを書く (元々の角度列を読みこんで、上書きするのを忘れない)
- [ ] wheels controllerを使うとペッパーがピョンピョンして転ぶが、```:go-pos``` をgazeboでどうやるか調べる

kochigami, 2016-09-09:
> 立ち上げる ```*ri*``` を実機モデルとシミュレータどちらにするか区別しているところを見つける

```
シミュレーションモードだと
send *ri* :robot-interface-simulation-callback
1
実機だと
send *ri* :robot-interface-simulation-callback
nil
```
gazeboが立ち上がっている時の区別が必要

> send ```*ri*``` :angle-vector (send ```*pepper*``` :angle-vector) が具体的にどのような処理を行っているか見つける

この中にgazebo modeの処理を書くようにするために、pepper-interface.lの中で```:angle-vector``` をオーバーライドした（正しくないと思う）

#655

kochigami, 2016-09-29:
> wheels controllerを使うとペッパーがピョンピョンして転ぶが、:go-pos をgazeboでどうやるか調べる

（planar move pluginを使ったら、```:go-velocity```は使えるようになったが、ペッパーが変な挙動をする。）

fetchやpr2はmove_baseを使っている。
pepperは```move_base_simple/goal``` をパブリッシュし、それを```moveTo```に渡してgo-posしている。
gazeboの立ち上げ時にmove_baseのノードを立ち上げればよいことは分かったが、move_baseのノードとpepperのgazeboのノードをどうつなげるかが分からない。

k-okada, 2016-09-30:
wheels controller
つかうとどういうtopicをうけとるようになあているかな？ cmd_velがでているとうれしい．であれば、
move_baseはmove_base_simpl/goalを受け取ってcmd_velを出すプログラムなのでつながる．

https://github.com/ros-naoqi/naoqi_bridge/pull/62
とかみているかな．これ中身は見ていないけど、実機でしか動かないプログラムになっているかな．

kochigami, 2016-09-30:
ご教示ありがとうございました。

wheel controllerのトピック
```
/pepper_dcm/Wheels_controller/command
/pepper_dcm/Wheels_controller/follow_joint_trajectory/cancel
/pepper_dcm/Wheels_controller/follow_joint_trajectory/feedback
/pepper_dcm/Wheels_controller/follow_joint_trajectory/goal
/pepper_dcm/Wheels_controller/follow_joint_trajectory/result
/pepper_dcm/Wheels_controller/follow_joint_trajectory/status
/pepper_dcm/Wheels_controller/state
```
で/cmd_velは出ていない状況です。

naoqi_navigation: localizationしながら充電ベースに行くプログラムが入っているようです。
Naoqi 2.5.2以上対応ということで、何も起きなかったので、アップグレードを待ってみます。
```
roslaunch naoqi_navigation navigation_full_rviz.launch
rostopic list
/clicked_point
/cmd_vel
/imu
/initialpose
/joint_states
/joint_stiffness
/localization_uncertainty
/localization_uncertainty_array
/move_base_simple/goal
/naoqi_exploration_map
/naoqi_exploration_map_updates
/naoqi_pod_publisher
/odom
/rosout
/rosout_agg
/tf
/tf_static
```
Naoqi2.4.3だと、
実機につないでも、（roslaunch jsk_pepper_startup jsk_pepper_startup.launch） ```/localization_uncertainty```, ```/naoqi_exploration_map``` は何も出てこない
rvizだけでも動きそう（？）

他のロボットはどうしているのかもう一度調べてみます。

k-okada, 2016-09-30:
なるほど、
/pepper_dcm/Wheels_controller/follow_joint_trajectory/goal
で受け取ったデータはどこに送られるのかな？直接gazeboにおくられる？
/pepper_dcm/Wheels_controller/command
はどういうデータ型かな．

--
◉ Kei Okada

kochigami, 2016-09-30:
```
rostopic info /pepper_dcm/Wheels_controller/follow_joint_trajectory/goal 
Type: control_msgs/FollowJointTrajectoryActionGoal

Publishers: None

Subscribers: 
 * /gazebo (http://kochigami-ThinkPad-T450:44690/)
```

```
rostopic type /pepper_dcm/Wheels_controller/command 
trajectory_msgs/JointTrajectory
```

このコントローラは他のコントローラ（例：```/pepper_dcm/RightArm_controller```）と一緒に定義されており、
https://github.com/ros-naoqi/pepper_virtual/blob/master/pepper_control/config/pepper_trajectory_control.yaml
にあります。

urdfを直しているこのプルリクエストが関係していると思いました。
https://github.com/ros-naoqi/pepper_robot/pull/28

勉強すること：
pepper_controlがどうやってコントローラを追加しているのか
urdf

k-okada, 2016-09-30:
> pepper_control/config/pepper_trajectory_control.yaml
> ros-naoqi/pepper_robot#28

お

+ <gazebo>
+ <plugin filename="libgazebo_ros_diff_drive.so" name="
differential_drive_controller">
+ <alwaysOn>true</alwaysOn>
+ <updateRate>100</updateRate>
+ <robotNamespace>/</robotNamespace>
+ <leftJoint>WheelFL</leftJoint>
+ <rightJoint>WheelFR</rightJoint>
+ <wheelSeparation>0.170</wheelSeparation>
+ <wheelDiameter>0.140</wheelDiameter>
+ <torque>36.9</torque>
+ <commandTopic>cmd_vel</commandTopic>
+ <odometryTopic>odom_diffdrive</odometryTopic>
+ <odometryFrame>odom</odometryFrame>
+ <robotBaseFrame>base_footprint</robotBaseFrame>
+ </plugin>
+ </gazebo>

は普通に車輪型ロボット用のgazebo pluginをついかしている
http://www.theconstructsim.com/how-to-build-a-differential-drive-simulation/

で上にcmd_velとみえるからcmd_velが受け付けるようになっていないかな？
としすると
http://wiki.ros.org/rqt_robot_steering

で動作確認できる

とすると普通に(ここが難しいけど)move_baseつかえば動く気がする

kochigami, 2016-10-07:
ご教示ありがとうございました。
結局、gazeboでmove_baseを使うと実機と挙動が変わると今は考えています。

naoqi_driverのgo-pos, go-velocityは以下のように独立しています。

| method | topic | naoqi API |
| --- | --- | --- |
| go-pos  | /move_base_simple/goal| move |
| go-velocity | /cmd_vel | moveToward |

gazeboでmove_baseで/move_base_simple/goal => (path planning) => /cmd_vel すると、
動き方が変わってしまうと思いました。

現在解決するべき問題は2つあります。
- ペッパーが/cmd_velのトピックを受け、傾きながら進む問題
- :go-pos, :go-velocityのgazebo実行時のメソッド定義

（自習として）
- move_baseの使い方（planningのyamlファイルなどを追加する）

kochigami, 2016-10-07:
> :go-pos, :go-velocityのgazebo実行時のメソッド定義

メモ
x, y, theta それぞれを現在の速度で割って、その時間分 ```:go-velocity``` することを考えている
現在の速度を取ってこれるように
 https://github.com/ros-naoqi/naoqi_bridge/pull/61
https://github.com/ros-naoqi/naoqi_bridge_msgs/pull/20

- naoqieusにメソッド追加  ```:get-move-config```,
- 上書き ```:go-pos```