jsk-ros-pkg/jsk_robot #664: [jsk_pepper_startup] want to add speech function when launch is killed

kochigami, 2016-09-24:
I'd like to add speech (or movement) function which can tell when ```jsk_pepper_startup.launch``` is killed.
I think it is very useful when I have to be near Pepper, not my PC in demo time.

However, I'm not sure how to do or google it. Could anyone help me?
Is there any command which can observe whether ros node is alive or not ?

furushchev, 2016-09-24:
my proposal is

- create node that uses signal hook to speech (e.g. rospy.on_shutdown )
- define the node as 'required' in launch file

this should work only when speech engine is available even without
startup.launch.
If not so, you may need to add code to speech on shutdown in speech node.

kochigami, 2016-09-25:
@furushchev  
Thank you very much for your comment. It really helped. 
I'll test my program tomorrow and send a pull-request for it.

(Memo for me: http://wiki.ros.org/rospy/Overview/Initialization%20and%20Shutdown)

closed 2016-09-25 by kochigami

k-okada, 2016-09-26:
> 20Shutdown)

I think you knowthe jsk_pepper_startup is started when pepper talks
"じょいくらんと" at
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/jsk_pepper_startup/nodes/joy-client.l#L84
and would like to notified when he stops the code.
I think easies way is to re-define `ros::roseus-sigint-handler` which is
called when roseus program receives C-c and speaks something
https://gist.github.com/k-okada/a8cfe74d77015f2c76b4f451edfbd633

Otherwise we'd add this feature to naoqi_driver.

kochigami, 2016-09-26:
@k-okada 

Thank you very much for your comment. Would you mind if I ask one more question? 
I'm not sure how to use ```ros::roseus-sigint-handler``` in your gist code.

I added ```ros::roseus-sigint-handler``` like
https://gist.github.com/kochigami/49b4132dbc47e318c319b5db8db20251

However, the result didn't seem to show executing ```ros::roseus-sigint-handler``` after pushing C-c.
```
[ INFO] [1474864440.431225535]: msg [hello world #i(1474864440 431146746)]
[ INFO] [1474864440.441172183]: msg [hello world #i(1474864440 441139942)]
[ INFO] [1474864440.451171301]: msg [hello world #i(1474864440 451131030)]
[ INFO] [1474864440.461170802]: msg [hello world #i(1474864440 461130000)]
^Cinterrupt1.B1-irteusgl$ 
```

```rostopic echo /chatter``` finishes
```
data: hello world #i(1474864440 441139942)
---
data: hello world #i(1474864440 451131030)
---
data: hello world #i(1474864440 461130000)
---
```

kochigami, 2016-09-26:
I couldn't find the correct way of using ```ros::roseus-sigint-handler```because of my low ability.
I have to continue studying it.

However, demo time is approaching. Therefore I made another pull-request just in case (https://github.com/jsk-ros-pkg/jsk_robot/pull/666). 
Once I learned how to use ```ros::roseus-sigint-handler```, I'll update it.

k-okada, 2016-09-26:
1: you do not need `(ros::roseus-sigint-handler ros::sig ros::code)` at the end of code, this code is 'signal handler', called whenever sigint occurs, and this does not have to call explictly from your main code
2: you have to create launch file and start you code, put your code under /tmp and run from roslaunch with --screen option, and hit C-c, you'll see the result
```
$ cat hoge.launch
<launch>
  <node pkg="roseus" type="roseus" name="talker"
        args="/tmp/talker.l" />
</launch>
$ roslaunch --screen hoge.launch
```
3: this is very complicating, but C-c on euslisp interpreter did not call 'signal handler' , it just calls https://github.com/jsk-ros-pkg/jsk_roseus/blob/1.5.3/roseus/roseus.cpp#L505
and `ros::roseus-sigint-handler` only enabled when you start lisp code from launch file https://github.com/jsk-ros-pkg/jsk_roseus/blob/1.5.3/roseus/euslisp/roseus.l#L30

kochigami, 2016-09-27:
Thank you very much for your help.
I could update the pull-request to use ```roseus-sigint-handler``` and then I'll study roseus.l and roseus.cpp.

@furushchev  @k-okada 
Thank you very much for your kind help again.