jsk-ros-pkg/jsk_robot #667: mognodbの使い方

furushchev, 2016-09-26:
1. 画像を入れる
2. イベントを入れる
3. 認識結果を入れる

furushchev, 2016-09-26:
sample code https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus_mongo/euslisp/mongo-client-sample.l

furushchev, 2016-09-26:
rosbagっぽいやつ
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_lifelog/mongodb_log.py

furushchev, 2016-09-26:
#### setup

- `rosparam set robot/name pr1012` # pr1040, baxter, pepper, etc...`
- `sudo apt-get install mongodb`
- `sudo service mongodb start`
- `roslaunch jsk_robot_startup mongodb_local.launch`

#### store message

```lisp
(ros::roseus "sample_node")
(require :mongo-client "package://roseus_mongo/euslisp/mongo-client.l")

;; store event
(mongo::insert (instance jsk_demo_common::FunctionEvent :init :name "pick") :meta '((:result . "success")))

;; store result
(setq *ac* (instance ros::simple-action-client :init "awesome_recognition" ActionMsg))
(send *ac* :send-goal goal)
(mongo::insert (send *ac* :get-result))

;; store image
(ros::subscribe "image" sensor_msgs::Image
  #'(lambda (m)
       (mongo::insert m :meta '((:memo . "my awesome image")))))
(ros::spin)
```

see more detail: [sample code](https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus_mongo/euslisp/mongo-client-sample.l)

#### store by topic

```bash
$ rosrun jsk_pr2_startup mongodb_log.py --help
Usage: mongodb_log.py [options] [TOPICs...]

Options:
  -h, --help            show this help message and exit
  --nodename-prefix=ROS_NODE_NAME
                        Prefix for worker node names
  --mongodb-host=HOST   Hostname of MongoDB
  --mongodb-port=PORT   Hostname of MongoDB
  --mongodb-name=NAME   Name of DB in which to store values
  --mongodb-collection=COLLECTION
                        Name of Collection in which to store values. All
                        topics are stored in the collection if used this
                        option, otherwise topic names are used as collections
  -a, --all-topics      Log all existing topics (still excludes /rosout,
                        /rosout_agg)
  -e, --regex           Log topics matching the follow regular expression
  --all-topics-interval=ALL_TOPICS_INTERVAL
                        Time in seconds between checks for new topics
  -x REGEX, --exclude=REGEX
                        Exclude topics matching REGEX, may be given multiple
                        times
  --no-specific         Disable specific loggers
```

```bash
$ rosrun jsk_pr2_startup mongodb_log.py /camera/rgb/image_rect_color
```

furushchev, 2016-09-26:
#### Viewer

comming soon...

furushchev, 2016-09-26:
Cc: @k-okada @knorth55

furushchev, 2016-09-28:
@knorth55 
認識結果は過去のJSKのプログラムでは`posedetection_msgs/ObjectDetection`というメッセージでやりとりされていて、これをロボットの位置などとともに蓄積するプログラムが、現在では`jsk_pr2_startup/jsk_pr2_lifelog/objectdetection_db.py`にあります。
PR2固有のプログラムでもないので、ドキュメントとともに`jsk_robot_startup/lifelog`に移そうと思っています。
https://github.com/jsk-ros-pkg/jsk_robot/pull/672
マージされるまでは上リンクを見てもらえるといいと思います。

knorth55, 2016-09-28:
@furushchev ありがとうございます。参考にさせていただきます

closed 2016-10-13 by furushchev