jsk-ros-pkg/jsk_robot #678: [jsk_pepper_startup] catkin bt しただけではeuslisp用のメッセージは作られない

kochigami, 2016-10-07:
メモ
（ビルドしたら自動でeuslispのメッセージも出来るはずと思い込んでいるので、間違っていたらすみません。
確かにCMakeLists.txtにそのような記載は見つけられませんでした。
時間のある時に自分で考えます。）
```
rm -fr ~/catkin_ws/build/naoqi_bridge_msgs/, peppereus, naoqieus
それぞれ　catkin bt (--force-cmake)
```
した状態で
```
roseus pepper-interface.l
 [ERROR] [1475816831.688800176]: Could not find roseus messages for naoqieus under (/home/kochigami/catkin_ws/devel/share/roseus/ros /opt/ros/indigo/\
share/roseus/ros)                                                                                                                                                        
try rosrun roseus generate-all-msg-srv.sh naoqieus
[ERROR] [1475816831.688890999]: Could not find nil/manifest.l                                                                                                            
try rosrun roseus generate-all-msg-srv.sh naoqieus
/opt/ros/indigo/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 no such package "NAO_INTERACTION_MSGS""NAO_INTERACTION_MSGS" in (apply #'ros::load-org-for-ros\
 ros::fullname args)E:
```

```
roslaunch jsk_pepper_startup jsk_pepper_startup.launch
[ERROR] [1475816496.383552205]: Could not find roseus messages for peppereus under (/home/kochigami/catkin_ws/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh peppereus
[ERROR] [1475816496.383633549]: Could not find nil/manifest.l
try rosrun roseus generate-all-msg-srv.sh peppereus
/opt/ros/indigo/share/euslisp/jskeus/eus/Linux64/bin/irteusgl roseus-error: unbound variable naoqi_bridge_msgs::jointangleswithspeed in (ros::advertise "/pepper_robot/pose/joint_angles" naoqi_bridge_msgs::jointangleswithspeed 1), exitting...
```

k-okada, 2016-10-07:
2016-10-07 14:35 GMT+09:00 Kanae Kochigami <notifications@github.com>:

> （ビルドしたら自動でeuslispのメッセージも出来るはずと思い込んでいるので、間違っていたらすみません。
>

できるはずです．
catkin b naoqieus
して、全てのログを貼って下さい．

--
◉ Kei Okada

kochigami, 2016-10-07:
同時に```rosrun roseus generate-all-msg-srv.sh --all```してしまっているので、正しくないのかもしれないのですが、

```
 rm -fr ~/catkin_ws/build/peppereus/
kochigami@kochigami-ThinkPad-T450:~/catkin_ws/src/jsk_robot/jsk_naoqi_robot/peppereus$ catkin b peppereus
==> Expanding alias 'b' from 'catkin b peppereus' to 'catkin build peppereus'
--------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/indigo
Workspace:                   /home/kochigami/catkin_ws
--------------------------------------------------------------
Source Space:       [exists] /home/kochigami/catkin_ws/src
Log Space:          [exists] /home/kochigami/catkin_ws/logs
Build Space:        [exists] /home/kochigami/catkin_ws/build
Devel Space:        [exists] /home/kochigami/catkin_ws/devel
Install Space:      [unused] /home/kochigami/catkin_ws/install
DESTDIR:            [unused] None
--------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------
[build] Found '68' packages in 0.0 seconds.                                    
[build] Package table is up to date.                                           
Starting  >>> naoqi_bridge_msgs                                                
Finished  <<< naoqi_bridge_msgs                 [ 8.9 seconds ]                
Starting  >>> pepper_description                                               
Finished  <<< pepper_description                [ 0.2 seconds ]                
Starting  >>> naoqi_driver                                                     
Finished  <<< naoqi_driver                      [ 2.0 seconds ]                
Starting  >>> naoqi_driver_py                                                  
Finished  <<< naoqi_driver_py                   [ 0.1 seconds ]                
Starting  >>> naoqi_pose                                                       
Finished  <<< naoqi_pose                        [ 0.1 seconds ]                
Starting  >>> naoqieus                                                         
_______________________________________________________________________________
Warnings   << naoqieus:symlink /home/kochigami/catkin_ws/logs/naoqieus/build.symlink.006.log
Warning: Cannot symlink from /home/kochigami/catkin_ws/devel/.private/naoqieus/share/roseus/ros/bond/manifest.l to existing file /home/kochigami/catkin_ws/devel/share/roseus/ros/bond/manifest.l
Warning: Source hash: 857d02b788f67d8d09f957858c6d7f21
Warning: Dest hash: e681470f584630e2ac3790f7ab285176
Warning: Cannot symlink from /home/kochigami/catkin_ws/devel/.private/naoqieus/share/roseus/ros/naoqi_bridge_msgs/manifest.l to existing file /home/kochigami/catkin_ws/devel/share/roseus/ros/naoqi_bridge_msgs/manifest.l
Warning: Source hash: 905d92bbcff555aafe3b70f868f3db80
Warning: Dest hash: d8546cae5720949f1944920953b27460
...............................................................................
Finished  <<< naoqieus                          [ 0.3 seconds ]                
Starting  >>> naoqi_sensors_py                                                 
Finished  <<< naoqi_sensors_py                  [ 0.3 seconds ]                
Starting  >>> pepper_sensors_py                                                
Finished  <<< pepper_sensors_py                 [ 0.2 seconds ]                
Starting  >>> pepper_bringup                                                   
Finished  <<< pepper_bringup                    [ 0.1 seconds ]                
Starting  >>> peppereus                                                        
Finished  <<< peppereus                         [ 4.2 seconds ]                
[build] Summary: All 10 packages succeeded!                                    
[build]   Ignored:   58 packages were skipped or are blacklisted.              
[build]   Warnings:  1 packages succeeded with warnings.                       
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 28.7 seconds total.                                           
kochigami@kochigami-ThinkPad-T450:~/catkin_ws/src/jsk_robot/jsk_naoqi_robot/peppereus$ roscd
[... 100 lines cut ...]
Warning: Source hash: 857d02b788f67d8d09f957858c6d7f21
Warning: Dest hash: e681470f584630e2ac3790f7ab285176
Warning: Cannot symlink from /home/kochigami/catkin_ws/devel/.private/naoqieus/share/roseus/ros/naoqi_bridge_msgs/manifest.l to existing file /home/kochigami/catkin_ws/devel/share/roseus/ros/naoqi_bridge_msgs/manifest.l
Warning: Source hash: 905d92bbcff555aafe3b70f868f3db80
Warning: Dest hash: d8546cae5720949f1944920953b27460
...............................................................................
Finished  <<< naoqieus                          [ 0.3 seconds ]                
Starting  >>> naoqi_sensors_py                                                 
[... cut ...]
[build] Summary: All 10 packages succeeded!                                    
[build]   Ignored:   58 packages were skipped or are blacklisted.              
[build]   Warnings:  1 packages succeeded with warnings.                       
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 28.7 seconds total. 
```

```rosrun roseus generate-all-msg-srv.sh peppereus```したらjsk_pepper_startup.launchのエラーは出なくなりました。
未来館のデモ準備で気もそぞろなので、後でゆっくり考えます。

k-okada, 2016-10-09:
コマンドはcatkin b naoqieus -v -i でした．
```
[naoqieus:check] -- [roseus.cmake] generate all roseus messages

[naoqieus:check] -- [roseus.cmake] naoqieus depends on
std_msgs;actionlib_msgs;bond;diagnostic_msgs;geometry_msgs;nav_msgs;rosgraph_msgs;roscpp;actionlib;sensor_msgs;tf;tf2_msgs;topic_tools;trajectory_msgs;nao_interaction_msgs;naoqi_bridge_msgs
[naoqieus:check] -- [roseus.cmake] naoqieus will compile
bond;diagnostic_msgs;nav_msgs;trajectory_msgs;nao_interaction_msgs;naoqi_bridge_msgs;naoqieus
```
となって、naoqeus will compile ... というところで、どのメッセージがコンパイルされているかわかるはずです．

kochigami, 2016-10-12:
ご教示ありがとうございました。
naoqieusではメッセージのコンパイルが確認できました。
```
roslaunch jsk_pepper_startup jsk_pepper_startup.launch 
/opt/ros/indigo/share/euslisp/jskeus/eus/Linux64/bin/irteusgl roseus-error: unbound variable naoqi_bridge_msgs::jointangleswithspeed in (ros::advertise "/pepper_robot/pose/joint_angles" naoqi_bridge_msgs::jointangleswithspeed 1), exitting...
```
が解決しました。
しかし、
```
roslaunch jsk_pepper_startup jsk_pepper_startup.launch 
[ERROR] [1476249943.482647677]: Could not find roseus messages for peppereus under (/home/kochigami/catkin_ws/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh peppereus
[ERROR] [1476249943.482943011]: Could not find nil/manifest.l
try rosrun roseus generate-all-msg-srv.sh peppereus
```
となってしまいました。
確かに~/catkin_ws/devel/share/roseus/ros/以下に　naoeus, peppereusのmanifest.lはありませんでした。
そこで、
https://github.com/jsk-ros-pkg/jsk_roseus/blob/2f877119e1469a954de18c0fe4d5ad83c200c622/roseus/cmake/roseus.cmake#L61
を見て、
#682 の変更を加えたら
```
[peppereus:check] -- [roseus.cmake] peppereus will compile bond;humanoid_nav_msgs;nodelet;nao_interaction_msgs;naoqi_bridge_msgs;naoqi_driver;peppereus
```
となり生成しました。

無駄なことをしていると思うので、何かアドバイスがありましたらお願いいたします。

closed 2016-10-12 by kochigami

k-okada, 2016-10-13:
なるほど．はい．これで正解です．すばらしい．
#682