jsk-ros-pkg/jsk_robot #703: ロボットの動きががたがたする

k-okada, 2016-11-02:
`send *ri* :angle-vector` でロボットに姿勢を送った時にがたがたする人は、もう一度fetcheusのバージョンを調べてみて下さい．
```
$ rosversion fetcheus
1.0.7
$ rosversion pr2eus_moveit
0.3.6
```
となっているのが正解です、もうひとつ:angle-vectorを送った時に以下のような表示になるか調べてみて下さい．
チェックポイントはscaled trajectory total time というのが3.0以上になっている．例えば`(send *ri* :angle-vector ... 5000)` としたら5.0秒になっているはず．
最後の行が`use [arm]`ではなく、`use [arm_with_torso]`になっている．
```
[ INFO] [1478085022.045371242, 403.854000000]: ;; Planned Trajectory Total Time   2.163 [sec]
[ INFO] [1478085022.048472056, 403.854000000]: ;; Scaled Trajectory Total Time   3.000 [sec]
[ INFO] [1478085022.049795165, 403.854000000]: ;; generated 10 points for 3 sec using [arm_with_torso] group
```

以上のチェックポイントが全てただしくて、まだがたがたしていたら教えて下さい．

@itohdak, @708yamaguchi , @Affonso-Gui, ＠YutoUchimi

k-okada, 2016-11-02:
あ、やっぱ、上の条件満たしていてもがたがたするかも．

k-okada, 2016-11-02:
hope https://github.com/jsk-ros-pkg/jsk_robot/pull/699/commits/40f0963b4c2dcac3bef74af99be1f841d2b8ef05 and https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/255 works, this will send joint_trajectory without velocity and acceleration, if this not work, we'll calculate velocities.

Affonso-Gui, 2016-11-04:
Tested the changes and work it out! Normal movements, correct timing and correct use of torso.

Thank you very much

708yamaguchi, 2016-11-04:
下記のようにgazebo内でfetchの腕を動かしましたが、目立ったガタつきはありませんでした。
```
8.B2-irteusgl$ (send *fetch* :rarm :inverse-kinematics (make-coords :pos #f(800 0 1000)))
#f(0.0 -7.59839 16.7784 -191.393 105.694 -175.805 87.7943 10.7701 0.060155 0.132435)
9.B2-irteusgl$ (send *ri* :angle-vector (send *fetch* :angle-vector) 5000)
[ INFO] [1478263374.360624626, 851.253000000]: ;; Planned Trajectory Total Time   3.062 [sec]
[ INFO] [1478263374.361121700, 851.253000000]: ;; Scaled Trajectory Total Time   5.000 [sec]
[ INFO] [1478263374.361148806, 851.253000000]: ;; generated 28 points for 5 sec using [arm_with_torso] group
(nil nil nil)
```