jsk-ros-pkg/jsk_robot #704: Could you please give me collaborator rights on this repo?

wkentaro, 2016-11-03:
like so in jsk_recognition.

This is because I'd like to place .rosinstall for pr2 on this repo with branch named `pr1012-rosinstall` and `pr1040-rosinstall`.

k-okada, 2016-11-03:
Changed permission settings, but please be very careful for using branches to support multiple robots. Basically we should have same settings for these two robots.

wkentaro, 2016-11-03:
I see. I will start with same configuration with a pr2-rosinstall branch.

wkentaro, 2016-11-03:
@chiwunau @h-kamada @k-okada @furushchev 

I created a branch: https://github.com/jsk-ros-pkg/jsk_robot/tree/pr2-rosinstall
and updated the workspace in applications@prXXXXs,

```
- ros - hydro - src/.rosinstall -> ../rosinstall/jsk_robot/pr2.rosinstall
              - rosinstall/jsk_robot/pr2.rosinstall
```

wkentaro, 2016-11-03:
Thank you. @k-okada

closed 2016-11-03 by wkentaro

jsk-members, 2016-11-03:
Please consider using
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_pr2_robot/jsk_pr2_startup/jsk_pr2.rosinstall,
instead of creating new branch

See
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_pr2_robot/README.md
for
our administration policy

wkentaro, 2016-11-03:
I see.
I think `wget` does not work when non-admin user changes the rosinstall. (actually there are many diffs in rosinstall in robots and github)
So now I think below is better:

```
- ros - hydro - src/.rosinstall -> ../rosinstall/jsk_robot/jsk_pr2_robot/jsk_pr2_startup/jsk_pr2.rosinstall
              - rosinstall/jsk_robot/jsk_pr2_robot/...
```

wkentaro, 2016-11-03:
Hmm. But if we create another branch, we can set the repository as a element of wstool.
And if we changes the rosinstall, there is diff in `wstool info` and we will be motivated to commit and push it.

yuki-asano, 2016-11-07:
私では無くて、@YoheiKakiuchi さんのようです。
DRC関係でしょうか。

---------- Forwarded message ----------
On Tue, Apr 21, 2015 at 2:09 AM, Yohei Kakiuchi <youhei@jsk.imi.i.u-tokyo.ac.jp> wrote:
jsk-members というgithubアカウントを作りました。

共通のissueなどの編集でこれを使っていく予定です。

YoheiKakiuchi, 2016-11-08:
すいません。
issueのチェックボックスをチェックする権限が、issueの作成者と管理者にしかなくて意欲がわかないと言う話だったので、作ってしまいました。